ros2/rclcpp #731: fixup parameters and sub-nodes

wjwwood, 2019-05-21:
This is a follow up from https://github.com/ros2/rclcpp/pull/495#issuecomment-485906580.

Currently how parameters and "sub-nodes" interact is not documented and is not consistent or tested. That needs to be documented, audited, and tested.

mikeferguson, 2020-06-07:
So, is this ticket basically tracking that sub nodes do not actually use the sub_namespace in the name of the parameter? For example:

```
auto subnode = parent->create_sub_node("controller_name");
std::string controller_type = subnode->declare_parameter<std::string>("type", "");
```

I would expect the parameter to be "controller_name/type", but instead "ros2 param list" shows only a parameter named "type"? Is this expected behavior? Is this a bug?

fujitatomoya, 2020-06-08:
I am not sure about this either, I'd like to know.

> I would expect the parameter to be "controller_name/type", but instead "ros2 param list" shows only a parameter named "type"?

currently, yes. only `type` is declared w/o namespace.

I think that this is bug, it should be extended with sub_namespace?

wjwwood, 2020-06-08:
Yeah @mikeferguson I consider it a bug, or at least I think it will be surprising to most users.

The issue (why I didn't just fix it on the spot) is that the way the node class is separated into parts complicates this. The subnode feature was added to `Node`, but not to one of the components, meaning it's hard to use from the `NodeParameters` class. This is just my recollection of the issue, but maybe it's easier to fix now, I don't know off hand without trying to do it.

jlack1987, 2021-09-16:
@wjwwood could you recommend an acceptable way of fixing this or point me in the right direction? I'm very interested in fixing this issue and am willing to work on it

jlack1987, 2021-09-21:
I'm willing to spend some time attempting to fix this issue, but i'd appreciate it if I could get a more involved devs input on a suggested fix/implementation. I can just run off and implement a fix, but I don't want to end up spending a lot of time on a fix and it not being an acceptable implementation

fujitatomoya, 2021-09-22:
just sharing the update. we had a quick chat on this in MW WG. we consider this is bug, all entities of sub-nodes should be extended and propagated with sub-nodes namespace. (topics, services and parameters) as @wjwwood mentioned https://github.com/ros2/rclcpp/issues/731#issue-446871224 these should be tested well.

jlack1987, 2021-10-06:
I'm curious if this could be solved by injecting the sub namespace into the parameter name in the call [here](https://github.com/ros2/rclcpp/blob/master/rclcpp/src/rclcpp/node.cpp#L329). The reason i'm curious if this is an adequate solution is based on how the [declare_parameters](https://docs.ros2.org/foxy/api/rclcpp/classrclcpp_1_1Node.html#a5ebda51ed6fa4ce8f2cddc9f77e9674e) call behaves in that it already takes a namespace and, in the case where the `namespace` arg is non-empty, creates parameters that are `namespace.name`.

So would it be valid to simply namespace parameters into the `sub_namespace_` which is already available and if it's empty it'll continue to behave as it does now for a non-subordinate node? I'm relatively new to ros2 so there may be fallout from this strategy that I'm currently unaware. Would appreciate it if I could get a devs input on this. Thanks!

jlack1987, 2021-10-06:
I have also found that if you declare parameters with a sub node `n` with,

```
std::map<std::string, double> parameters = { { "a", 1. }, { "b", 2. }, { "c", 3. } };
n->declare_parameters(n->get_sub_namespace(), parameters);
```

and then retrieve those params with,

```
n->get_parameter<double>(n->get_sub_namespace() + ".b", param_b);
rclcpp::Parameter param = n->get_parameter(n->get_sub_namespace() + ".b");
```
that the second call there does get the correct parameter, but the first call doesn't find it and returns what looks like some uninitialized number(`7.892878e-293` in one run for example). This is because the templated call prepends the parameter name with the sub node namespace, but the other two API calls for getting a parameter don't do that

jlack1987, 2022-01-03:
I had a PR for this [here](https://github.com/ros2/rclcpp/pull/1796), but the approach was ultimately flawed so I closed it; however, there is a good deal of discussion in there that someone may find useful so I just wanted to link to it here.

fujitatomoya, 2022-01-03:
I added this to next MW WG topic not to miss it.

wjwwood, 2022-01-12:
@wjwwood to leave an idea about how to do this without creating new `NodeParameter` instances, but still get namespacing for parameter methods of the node class.

matemat13, 2025-04-07:
Did this ever get resolved? It seems that namespaced parameters of subnodes are still broken (not behaving as expected).

nirwester, 2025-06-02:
Is this fully solved by https://github.com/ros2/rclcpp/pull/2822 ?

fujitatomoya, 2025-06-02:
@nirwester yeah i think so, thanks for letting me know.

closed 2025-06-02 by fujitatomoya