ros2/rclcpp #732: Can change use_sim_time parameter type and delete it with CLI

jacobperron, 2019-05-22:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Debian packages
- Version or commit hash:
  - 0.7.3
- DDS implementation:
  - Fast-RTPS and Connext (not tested with OpenSplice)
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

Run a demo node:

```
ros2 run demo_nodes_cpp listener 
```

Try setting the parameter to something other than a Boolean:

```
ros2 param set listener use_sim_time foobar
```

Also, try deleting the parameter:

```
ros2 param delete listener use_sim_time
```

#### Expected behavior

It's expected that the `use_sim_time` parameter cannot be set to anything other than a Boolean and that it cannot be deleted.

#### Actual behavior

`use_sim_time` can be set to something other than a Boolean and can be deleted.

#### Additional information

An error is logged when trying to set `use_sim_time` with a different type, but the set appears to succeed anyways.

After deleting the parameter, if we try to set it again we get an error stating that the parameter is not declared:

```
$ ros2 param set listener use_sim_time True    
Setting parameter failed: parameter 'use_sim_time' cannot be set because it was not declared
```

I have not confirmed if this is just an issue with `use_sim_time` or parameters in general.

jacobperron, 2019-05-22:
@wjwwood FYI

ivanpauno, 2020-11-04:
IIUC, this was fixed in https://github.com/ros2/rclcpp/pull/875.

closed 2020-11-04 by ivanpauno