ros2/rclcpp #733: rclcpp_components component_container doesn't start

clalancette, 2019-05-22:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Source
- Version or commit hash:
  - 0.7.3
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

```
ros2 run rclcpp_components component_container
```

In another terminal:
```
ros2 component list
ros2 component load /ComponentManager composition composition::Talker
```

#### Expected behavior
The `ros2 component list` shows that the component is running.  Also, components can be loaded into the container.

#### Actual behavior
`ros2 component list` returns empty, and trying to load a component with `load` results in:

```
Unable to find container node '/ComponentManager'
```

#### Additional information

This could be user error, since this is the first time I'm using the component stuff, but I was just following the tutorial at https://index.ros.org/doc/ros2/Tutorials/Composition/

jacobperron, 2019-05-22:
Possible duplicate (or at least related) https://github.com/ros2/ros2cli/issues/221

clalancette, 2019-05-22:
> Possible duplicate (or at least related) [ros2/ros2cli#221](https://github.com/ros2/ros2cli/issues/221)

Yeah, maybe.  I filed a new bug because it isn't working at all for me, so I wasn't sure that it was the same thing.

dirk-thomas, 2019-05-22:
This is not a duplicate. The node shows up in `ros2 node list` but not in `ros2 component list`. PR incoming...

closed 2019-05-22 by dirk-thomas