ros2/rclcpp #734: Cross-compilation

roderick-koehle, 2019-05-23:
## Bug report
- Operating System:
  - Android NDK r19
  - ANDROID_ABI=x86,armeabi-7a, arm64-v8a
  - ANDROID_TOOLCHAIN_NAME=arm-linux-androideabi-clang
  - ANDROID_NATIVE_API_LEVEL=android-28
- Installation type:
  - Source
- Version or commit hash:
  - master
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

Android cross compilation with latest platform API results in a compile error in the signal_handler of rclcpp.

#### Actual behavior
```
Starting >>> rclcpp
--- stderr: rclcpp
/home/EU/dekoehlr/work/ros2_core_ws/src/ros2/rclcpp/rclcpp/src/rclcpp/signal_handler.cpp:176:7: 
error: cannot initialize a variable of type 'int' with an rvalue of type 'char *'
   int error_status = strerror_r(errnum, buffer, buffer_length);
       ^              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [CMakeFiles/rclcpp.dir/src/rclcpp/signal_handler.cpp.o] Error 1
make[1]: *** [CMakeFiles/rclcpp.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< rclcpp     [ Exited with code 2 ]
```

#### Additional information
The cause is the use of the reentrant `strerror_r` reporting function, which has a inconsistent function signature in the XSI and GNU string.h core library.
The feature test macros (see https://linux.die.net/man/3/strerror_r) to distinguish between XSI and POSIX are not reliable on Android as the _GNU_SOURCE may or may not be defined depending on the API level. For a discussion of the use of `strerror_r` see http://austingroupbugs.net/view.php?id=655.

Possible solutions are substituting the ambiguous `strerror_r` by the (re-entrant) `strerror_l`, or use the prototype definition check provided by cmake, e.g.:
```cmake
# Query platform depended string functions
check_prototype_definition(strerror_r
  "int strerror_r(int errnum, char *strerrbuf, size_t buflen)"
  "0"
  "string.h"
  XSI_STRERROR_R)
check_prototype_definition(strerror_r
  "char *strerror_r(int errnum, char *buf, size_t buflen)"
  "NULL"
  "string.h"
  GNU_STRERROR_R)
if(XSI_STRERROR_R)
   add_compile_options(-DXSI_STRERROR_R)
endif()
if(GNU_STRERROR_R)
   add_compile_options(-DGNU_STRERROR_R)
endif()
```

dirk-thomas, 2019-05-23:
Please consider to provide a pull request to address this since you have the platform at hand to verify that the potential patch will resolve the problem.

roderick-koehle, 2019-05-25:
I submitted pull request #742, with a minimalist approach.

nuclearsandwich, 2019-07-18:
@roderick-koehle with #742 merged I am going to close this issue. If there are other issues with cross-compilation we can re-open this one or you can submit a new one as appropriate.

closed 2019-07-18 by nuclearsandwich