ros2/rclcpp #743: Problem of not enabling multi-layered namespace

LeonJung, 2019-05-27:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - src
- Version or commit hash:
  - 0.6.2
- DDS implementation:
  - Any
- Client library (if applicable):
  - rclcpp

I couldn't put multi-layered namespace with the one in ros2 launch and another one in rclcpp::Node

Theoretically, the system should put the namespace set in launch file over the namespace set in Node instance parameter (and also the node name after that). However, the namespace in launch file overwrites another one whether is set or not. And therefore, the namespace removes the parameter wrapped by another namespace.

I actually tested putting namespace as "namespace1/namespace2" in Node function, and it works but not showing as "namespace2" in "namespace1" but as "namespace1/namespace2". So I guessed if the multi-layered namespace is really enabled in ros2. 

Am I doing well or is there anyone having other way to wrap with launch namespace 

Thanks

nuclearsandwich, 2019-07-18:
Can you please share an example launch file and steps to reproduce the problem as you see it? Being able to reproduce it will help us determine whether the current behavior is expected or not.

ivanpauno, 2019-07-18:
@LeonJung I don't fully understand your description of the problem, but this is maybe related https://github.com/ros2/launch_ros/pull/42.

wjwwood, 2020-03-23:
I haven't heard from you in a bit (which is totally fine!), but because of that I'm going to close this. If you're still working on it, feel free to speak up and I'll open it back up, or if you don't have time right now but do in the future, speak up then and I can open it again at that point.

Thanks!

closed 2020-03-23 by wjwwood