ros2/rclcpp #746: use_sim_time error message is logged from multiple nodes

jacobperron, 2019-05-29:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04 and macOS
- Installation type:
  - source (Linux) and fat archives (macOS)
- Version or commit hash:
  - 0723a0a6fc2f46edcbeebf2933c7929822177d56
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

Run the talker C++ demo:
```
ros2 run demo_nodes_cpp talker
```

Run the listener C++ demo:
```
ros2 run demo_nodes_cpp listener
```

Set the `use_sim_time` parameter for the talker to something other than a bool:
```
ros2 param set talker use_sim_time foo
```

#### Expected behavior

We see an error (or warning) logged from the talker node regarding the type mismatch for the `use_sim_time` parameter. E.g.

```
[ERROR] [talker]: use_sim_time parameter set to something besides a bool
```

#### Actual behavior

We see the above error message, not only for the talker node, but also for the listener node. I.e. both the talker and the listener log the following error message:

```
[ERROR] [talker]: use_sim_time parameter set to something besides a bool
```

#### Additional information

We can see that the parameter is set on the talker node, but not on the listener node, which is correct. The bug is that we see the listener log an error when it is not expected.

hidmic, 2019-07-26:
So, it looks like `rclcpp::TimeSource` is listening to changes to `use_sim_time` system wide (see [here](https://github.com/ros2/rclcpp/blob/master/rclcpp/src/rclcpp/time_source.cpp#L221)). Judging by https://github.com/ros2/rclcpp/issues/609, this subscription serves no specific purpose and could/should be replaced by an `on_set_parameters` callback. Is that right @wjwwood ? If so, I'll do so.

ivanpauno, 2019-07-26:
> So, it looks like `rclcpp::TimeSource` is listening to changes to `use_sim_time` system wide (see [here](https://github.com/ros2/rclcpp/blob/master/rclcpp/src/rclcpp/time_source.cpp#L221)). Judging [...]

`on_set_parameter` is for rejecting/accepting a parameter.
It shouldn't be used for updating a "global" state.
See discussion here https://github.com/ros2/rclcpp/pull/772#discussion_r305062053.
#609 is asking for a local "on_parameter_change" callback, but we currently don't have that feature.

At the moment, using `on_parameter_event` is the only correct way of doing this.
The fully qualified node name is part of the `Parameter Event` [message](https://github.com/ros2/rcl_interfaces/blob/a7ea280423e64f38d910e318d73a55bc22e71d42/rcl_interfaces/msg/ParameterEvent.msg#L8).
It should be used for filter nodes out.

hidmic, 2019-07-26:
> It shouldn't be used for updating a "global" state.

Fair enough. 

> The fully qualified node name is part of the Parameter Event message.

Yes, I can work with that. Just checking if there's a better way to do so.

hidmic, 2019-07-29:
@jacobperron #799 should've addressed your issue.

closed 2019-07-29 by hidmic

jacobperron, 2019-07-29:
< @jacobperron #799 should've addressed your issue. 

Works for me :+1:

PR #799 (2019-07-29): Make TimeSource ignore use_sim_time events coming from other nodes
Fixes #746.