ros2/rclcpp #750: Add an option for ignoring parameter overrides

ivanpauno, 2019-06-03:
## Feature request

#### Feature description
Follow up of #730 (see https://github.com/ros2/rclcpp/issues/730#issuecomment-496676163). Currently, the user don't have a way of ignoring a parameter override.

My suggestion is to change:

```cpp
  RCLCPP_PUBLIC
  const rclcpp::ParameterValue &
  declare_parameter(
    const std::string & name,
    const rclcpp::ParameterValue & default_value = rclcpp::ParameterValue(),
    const rcl_interfaces::msg::ParameterDescriptor & parameter_descriptor =
    rcl_interfaces::msg::ParameterDescriptor());
```
to
```cpp
  RCLCPP_PUBLIC
  const rclcpp::ParameterValue &
  declare_parameter(
    const std::string & name,
    const rclcpp::ParameterValue & default_value = rclcpp::ParameterValue(),
    const rcl_interfaces::msg::ParameterDescriptor & parameter_descriptor =
    rcl_interfaces::msg::ParameterDescriptor()
    bool ignore_override = false,
  );
```
and maybe, also add:
```cpp
  RCLCPP_PUBLIC
  const rclcpp::ParameterValue &
  declare_parameter(
    const std::string & name,
    const rclcpp::ParameterValue & default_value = rclcpp::ParameterValue(),
    bool ignore_override = false,
  );
```

When undeclared parameters are allowed and `automatically_declare_parameter_overrides` is `false`, the first `set_parameter` call does not actually set the parameter but it takes the override value (see the second example [here](https://github.com/ros2/rclcpp/issues/730#issuecomment-495801806)).

I don't see much usage of that feature, and I think it's a little confusing. I suggest always ignoring the parameter override in that case. That last change would be an api break, but of something with not much use cases. I don't see much value on adding a `ignore_override` argument to `set_parameter` method.

Note: Maybe, we could also add a node option for ignoring all the overrides. But I don't see much usage of that.

#### Implementation considerations
Should be fairly easy to implement. An equivalent would be needed in rclpy.

ivanpauno, 2019-06-03:
@chapulina @wjwwood any opinions?

chapulina, 2019-06-03:
Having a way to ignore parameter overrides sounds helpful to me.

As for the implementation, have you considered passing `ignore_overrides` as part of [ParameterDescriptor](https://github.com/ros2/rcl_interfaces/blob/master/rcl_interfaces/msg/ParameterDescriptor.msg)?

> I suggest always ignoring the parameter override in that case

I feel that `set_parameter` should definitely set it. If ignoring the overrides is the way to do this I guess I'm onboard.

closed 2019-06-24 by ivanpauno