ros2/rclcpp #755: Listener stops working

Vaan5, 2019-06-10:
- Operating System:
Windows 10
- Installation type:
binaries
- Version or commit hash:
dashing
- DDS implementation:
Fast-RTPS
- Client library (if applicable):
rclcpp

#### Steps to reproduce issue
I have a modified listener/talker example, where I send around a double value.

listener_node.cpp
```
#include <memory>

#include "rclcpp/rclcpp.hpp"

#include "demo/listener.hpp"

int main(int argc, char ** argv)
{
    setvbuf(stdout, nullptr, _IONBF, BUFSIZ);

    printf("Initialize communications\n");
    rclcpp::init(argc, argv);
    printf("Initialize communications - done\n");

    printf("Create node\n");
    auto node = std::make_shared<demo::Listener>();
    printf("Create node - done\n");

    printf("Spin\n");
    rclcpp::spin(node);
    printf("Spin - done\n");

    printf("Shutdown\n");
    rclcpp::shutdown();
    printf("Shutdown - done\n");

    return 0;
}
```

listener.cpp
```
#include "demo/listener.hpp"

#include <functional>

namespace demo {
Listener::Listener() : Node("listener") {
    this->sub_ = this->create_subscription<std_msgs::msg::Float64>("comm_topic", 1000, std::bind(&Listener::callback, this, std::placeholders::_1));
}

void Listener::callback(std_msgs::msg::Float64::SharedPtr msg) {
    RCLCPP_INFO(this->get_logger(), "Received %f", msg->data);
}

Listener::~Listener() {

}
}
```

listener.hpp
```
#ifndef DEMO__LISTENER_HPP_
#define DEMO__LISTENER_HPP_

#include "rclcpp/rclcpp.hpp"

#include "std_msgs/msg/float64.hpp"

#include "demo/visibility_control.h"

namespace demo {

    class Listener : public rclcpp::Node {
    public:
        explicit Listener();

        virtual ~Listener();

        void callback(std_msgs::msg::Float64::SharedPtr msg);
    private:
        rclcpp::Subscription<std_msgs::msg::Float64>::SharedPtr sub_;
    };

}  // namespace demo

#endif  // DEMO__LISTENER_HPP_
```
publisher_node.cpp
```
#include <memory>
#include <thread>
#include <chrono>

#include "rclcpp/rclcpp.hpp"

#include "demo/publisher.hpp"

int main(int argc, char ** argv)
{
    setvbuf(stdout, nullptr, _IONBF, BUFSIZ);

    printf("Initialize communications\n");
    rclcpp::init(argc, argv);
    printf("Initialize communications - done\n");

    printf("Create node\n");
    auto node = std::make_shared<demo::Publisher>();
    printf("Create node - done\n");

    printf("Sleeping\n");
    std::this_thread::sleep_for(std::chrono::milliseconds(1000));
    printf("Sleeping - done\n");

    printf("Start\n");
    node->start();
    printf("Start - done\n");

    printf("Spin\n");
    rclcpp::spin(node);
    printf("Spin - done\n");

    printf("Shutdown\n");
    rclcpp::shutdown();
    printf("Shutdown - done\n");

    return 0;
}
```

publisher.cpp
```
#include "demo/publisher.hpp"

namespace demo {
Publisher::Publisher() : Node("publisher") {
    this->pub_ = this->create_publisher<std_msgs::msg::Float64>("comm_topic");
}

void Publisher::start() {
    std_msgs::msg::Float64 msg;
    //rclcpp::WallRate loop_rate(10);
    while (true) {
        msg.data = this->counter_;
        RCLCPP_INFO(this->get_logger(), "Publishing %f", this->counter_);
        this->pub_->publish(msg);
        this->counter_ += 0.01;
        //loop_rate.sleep();
    }
}

Publisher::~Publisher() {

}
}
```

publisher.hpp
```
#ifndef DEMO__PUBLISHER_HPP_
#define DEMO__PUBLISHER_HPP_

#include "rclcpp/rclcpp.hpp"

#include "std_msgs/msg/float64.hpp"

#include "demo/visibility_control.h"

namespace demo {

    class Publisher : public rclcpp::Node {
    public:
        explicit Publisher();

        virtual ~Publisher();

        void start();
    private:
        rclcpp::Publisher<std_msgs::msg::Float64>::SharedPtr pub_;
        rclcpp::TimerBase::SharedPtr timer_;

        double counter_;
    };

}  // namespace demo

#endif  // DEMO__PUBLISHER_HPP_

```

#### Expected behavior
The listener will receive either all messages or if he can't process them fast enough some will fall out of the queue.

#### Actual behavior
The publisher works as expected - just keeps publishing messages.
The listener **STOPS** after receiving some messages (random amount). The log message from the callback won't be displayed anymore.

#### Additional information
- Is this a known bug?
- If I use a smaller frequency (10Hz), the listener seems to work properly. With a higher frequency (1000Hz) or (100Hz) the same problem occurs.

ivanpauno, 2019-06-10:
Your publisher node is looping for ever in `Publisher::start`, so it never spins.
Check https://github.com/ros2/demos/blob/master/demo_nodes_cpp/src/topics/talker.cpp for wall timer usage.

Vaan5, 2019-06-10:
@ivanpauno, You are right, but that is still interesting. On my machine, I get the following output:

ros2 run demo publisher_node
```Initialize communications
Initialize communications - done
Create node
Create node - done
Sleeping
Sleeping - done
Start
[INFO] [publisher]: Publishing 0.000000
[INFO] [publisher]: Publishing 0.010000
[INFO] [publisher]: Publishing 0.020000
[INFO] [publisher]: Publishing 0.030000
[INFO] [publisher]: Publishing 0.040000
[INFO] [publisher]: Publishing 0.050000
[INFO] [publisher]: Publishing 0.060000
[INFO] [publisher]: Publishing 0.070000
[INFO] [publisher]: Publishing 0.080000
[INFO] [publisher]: Publishing 0.090000
[INFO] [publisher]: Publishing 0.100000
[INFO] [publisher]: Publishing 0.110000
[INFO] [publisher]: Publishing 0.120000
[INFO] [publisher]: Publishing 0.130000
[INFO] [publisher]: Publishing 0.140000
[INFO] [publisher]: Publishing 0.150000
[INFO] [publisher]: Publishing 0.160000
[INFO] [publisher]: Publishing 0.170000
[INFO] [publisher]: Publishing 0.180000
[INFO] [publisher]: Publishing 0.190000
[INFO] [publisher]: Publishing 0.200000
[INFO] [publisher]: Publishing 0.210000
```

ros2 run demo listener_node
```
Initialize communications
Initialize communications - done
Create node
Create node - done
Spin
[INFO] [listener]: Received 0.000000
[INFO] [listener]: Received 0.010000
[INFO] [listener]: Received 0.020000
[INFO] [listener]: Received 0.030000
[INFO] [listener]: Received 0.040000
[INFO] [listener]: Received 0.050000
[INFO] [listener]: Received 0.060000
[INFO] [listener]: Received 0.070000
[INFO] [listener]: Received 0.080000
[INFO] [listener]: Received 0.090000
[INFO] [listener]: Received 0.100000
[INFO] [listener]: Received 0.110000
[INFO] [listener]: Received 0.120000
[INFO] [listener]: Received 0.130000
[INFO] [listener]: Received 0.140000
[INFO] [listener]: Received 0.150000
[INFO] [listener]: Received 0.160000
[INFO] [listener]: Received 0.170000
[INFO] [listener]: Received 0.180000
[INFO] [listener]: Received 0.190000
[INFO] [listener]: Received 0.200000
[INFO] [listener]: Received 0.210000
[INFO] [listener]: Received 0.220000
[INFO] [listener]: Received 0.230000
[INFO] [listener]: Received 0.240000
[INFO] [listener]: Received 0.250000
[INFO] [listener]: Received 0.260000
```

[... 182 lines cut ...]
[INFO] [listener]: Received 0.040000
[INFO] [listener]: Received 0.050000
[INFO] [listener]: Received 0.060000
[INFO] [listener]: Received 0.070000
[INFO] [listener]: Received 0.080000
[INFO] [listener]: Received 0.090000
[INFO] [listener]: Received 0.100000

...

[INFO] [listener]: Received 428.880000
[INFO] [listener]: Received 428.890000
[INFO] [listener]: Received 428.900000
[INFO] [listener]: Received 428.910000
[INFO] [listener]: Received 428.920000
[INFO] [listener]: Received 428.930000
[INFO] [listener]: Received 428.940000
[INFO] [listener]: Received 428.950000
[INFO] [listener]: Received 428.960000
[INFO] [listener]: Received 428.970000
```

This behavior is really awkward in my opinion. The listener just stops. I would have expected that it either drops messages (if it can't process them fast enough).

**Question 2**:  Is this a bug (known)? Can we somehow get information from the listener that something like this happened? With smaller frequencies (10Hz), this problem didn't happen on my machine.

jacobperron, 2019-06-19:
> Question 1: Why does the callback of the listener get called? If my understanding is correct - spin is only required for the subscription (+ timers + services) part of the communication; the publishing is independent of that, right?

I don't think spinning is required for publishers. AFAIK, we only need to spin for entities where we're waiting for something to be executed (e.g. a callback).

> Question 2: Is this a bug (known)? Can we somehow get information from the listener that something like this happened? With smaller frequencies (10Hz), this problem didn't happen on my machine.

It sounds like a bug. I'm unable to reproduce on Linux (Ubuntu 18.04); all messages appear to make it to the subscriber. Perhaps it is related to Windows. I'll try next time I'm at a Windows machine.

Can you confirm if the same issue happens with `std_msgs::msg::String`? (i.e. just change the [update rate in the demo talker](https://github.com/ros2/demos/blob/df5ec3a477c683bbaf9b3a2ec064680a0d36829d/demo_nodes_cpp/src/topics/talker.cpp#L63)).

ivanpauno, 2019-06-19:
> It sounds like a bug. I'm unable to reproduce on Linux (Ubuntu 18.04); all messages appear to make it to the subscriber. Perhaps it is related to Windows. I'll try next time I'm at a Windows machine.

I have been able to reproduce it on Linux (after retrying a bunch of times). The worst case I have seen:
our demos with the timer changed to 100us, published up to message number 4700, listened up to message 4676.

> I don't think spinning is required for publishers. AFAIK, we only need to spin for entities where we're waiting for something to be executed (e.g. a callback).

I suppose that spinning isn't needed for publishers, but I'm not sure.

Vaan5, 2019-06-21:
The same problem happens with the demo talker/listener (so with String messages) if I change the rate.

vinnamkim, 2019-06-22:
This problem seems to be related to Fast-RTPS. Using OpenSplice for DDS implementation, I can't reproduce this problem unlike Fast-RTPS.

ivanpauno, 2019-06-24:
> This problem seems to be related to Fast-RTPS. Using OpenSplice for DDS implementation, I can't reproduce this problem unlike Fast-RTPS.

Good catch.
I have seen a difference of one message after `cntrl+c`-ing in `opensplice` (and I moved the log after the `publish` in the talker code). But in `fastrtps` the problem seems to be much worst.

clalancette, 2019-06-24:
@vinnamkim One thing that is happening in your example is that you are using a "reliable" QoS (that is the default if you don't specify anything).  In that mode, all of the messages must be acknowledged before they are dropped from the queue.  One interesting thing might be to try switching to "Best Effort", and seeing what the results are; in that case, the publisher will forget about the message as soon as it sends it.

The other thing is to ping the Fast-RTPS people to see what they have to say about.  @vinnamkim they will most likely ask for a wireshark capture, so can you please do that?

@richiprosima can you take a look at this and see if this is expected behavior or a bug?

closed 2025-05-17 by mjcarroll