ros2/rclcpp #761: fail to have const member function as callback

Karsten1987, 2019-06-12:
When trying to set the callback to a `const` class member function, the function traits are complaining and no suitable overload is found.

To reproduce:

Use the example here: https://github.com/ros2/examples/blob/master/rclcpp/minimal_subscriber/member_function.cpp
and declare the function `topic_callback` to `const`.
This produces the following traceback:

```
In file included from /Users/karsten/workspace/osrf/ros2_full/src/ros2/examples/rclcpp/minimal_subscriber/member_function.cpp:17:
In file included from /Users/karsten/workspace/osrf/ros2_full/install/include/rclcpp/rclcpp.hpp:144:
In file included from /Users/karsten/workspace/osrf/ros2_full/install/include/rclcpp/executors.hpp:21:
In file included from /Users/karsten/workspace/osrf/ros2_full/install/include/rclcpp/executors/multi_threaded_executor.hpp:24:
In file included from /Users/karsten/workspace/osrf/ros2_full/install/include/rclcpp/executor.hpp:32:
In file included from /Users/karsten/workspace/osrf/ros2_full/install/include/rclcpp/memory_strategies.hpp:18:
In file included from /Users/karsten/workspace/osrf/ros2_full/install/include/rclcpp/memory_strategy.hpp:24:
In file included from /Users/karsten/workspace/osrf/ros2_full/install/include/rclcpp/any_executable.hpp:20:
In file included from /Users/karsten/workspace/osrf/ros2_full/install/include/rclcpp/callback_group.hpp:23:
In file included from /Users/karsten/workspace/osrf/ros2_full/install/include/rclcpp/client.hpp:31:
/Users/karsten/workspace/osrf/ros2_full/install/include/rclcpp/function_traits.hpp:52:40: error: reference to overloaded function could not be resolved; did you mean to call it?
    typename function_traits<decltype( &FunctionT::operator())>::arguments>::type;
                                       ^~~~~~~~~~~~~~~~~~~~~~
/Users/karsten/workspace/osrf/ros2_full/install/include/rclcpp/subscription_traits.hpp:91:39: note: in instantiation of template class 'rclcpp::function_traits::function_traits<std::__1::__bind<void (MinimalSubscriber::*)(std::__1::shared_ptr<std_msgs::msg::String_<std::__1::allocator<void> > >) const, MinimalSubscriber *, const std::__1::placeholders::__ph<1> &> >' requested here
    typename rclcpp::function_traits::function_traits<CallbackT>::template argument_type<0>>
                                      ^
/Users/karsten/workspace/osrf/ros2_full/install/include/rclcpp/node.hpp:240:45: note: in instantiation of template class 'rclcpp::subscription_traits::has_message_type<std::__1::__bind<void (MinimalSubscriber::*)(std::__1::shared_ptr<std_msgs::msg::String_<std::__1::allocator<void> > >) const, MinimalSubscriber *, const std::__1::placeholders::__ph<1> &>, void, void, void>' requested here
      typename rclcpp::subscription_traits::has_message_type<CallbackT>::type, AllocatorT>>
                                            ^
/Users/karsten/workspace/osrf/ros2_full/install/include/rclcpp/node_impl.hpp:115:7: note: in instantiation of default argument for 'create_subscription<std_msgs::msg::String_<std::__1::allocator<void> >, std::__1::__bind<void (MinimalSubscriber::*)(std::__1::shared_ptr<std_msgs::msg::String_<std::__1::allocator<void> > >) const, MinimalSubscriber *, const std::__1::placeholders::__ph<1> &>, std::__1::allocator<void> >' required here
Node::create_subscription(
      ^~~~~~~~~~~~~~~~~~~~
/Users/karsten/workspace/osrf/ros2_full/src/ros2/examples/rclcpp/minimal_subscriber/member_function.cpp:27:27: note: while substituting deduced template arguments into function template 'create_subscription' [with MessageT = std_msgs::msg::String_<std::__1::allocator<void> >, CallbackT = std::__1::__bind<void (MinimalSubscriber::*)(std::__1::shared_ptr<std_msgs::msg::String_<std::__1::allocator<void> > >) const, MinimalSubscriber *, const std::__1::placeholders::__ph<1> &>, AllocatorT = (no value), SubscriptionT = (no value)]
    subscription_ = this->create_subscription<std_msgs::msg::String>(
                          ^
In file included from /Users/karsten/workspace/osrf/ros2_full/src/ros2/examples/rclcpp/minimal_subscriber/member_function.cpp:17:
In file included from /Users/karsten/workspace/osrf/ros2_full/install/include/rclcpp/rclcpp.hpp:144:
In file included from /Users/karsten/workspace/osrf/ros2_full/install/include/rclcpp/executors.hpp:21:
In file included from /Users/karsten/workspace/osrf/ros2_full/install/include/rclcpp/executors/multi_threaded_executor.hpp:24:
In file included from /Users/karsten/workspace/osrf/ros2_full/install/include/rclcpp/executor.hpp:32:
In file included from /Users/karsten/workspace/osrf/ros2_full/install/include/rclcpp/memory_strategies.hpp:18:
In file included from /Users/karsten/workspace/osrf/ros2_full/install/include/rclcpp/memory_strategy.hpp:24:
In file included from /Users/karsten/workspace/osrf/ros2_full/install/include/rclcpp/any_executable.hpp:20:
In file included from /Users/karsten/workspace/osrf/ros2_full/install/include/rclcpp/callback_group.hpp:23:
In file included from /Users/karsten/workspace/osrf/ros2_full/install/include/rclcpp/client.hpp:31:
/Users/karsten/workspace/osrf/ros2_full/install/include/rclcpp/function_traits.hpp:54:45: error: implicit instantiation of undefined template 'std::__1::tuple_size<int>'
  static constexpr std::size_t arity = std::tuple_size<arguments>::value;
                                            ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/__tuple:25:49: note: template is declared here
template <class _Tp> class _LIBCPP_TEMPLATE_VIS tuple_size;
```

As a temporary workaround, one can explicitly assign the output of `std::bind` to a `std::function` object.

@esteve Do you mind helping out here?

esteve, 2019-06-13:
@Karsten1987 sure, no problem. Is this happening on macOS only? What version of macOS?

esteve, 2019-06-14:
I just tested this locally and it also fails with Linux, I'm working on a patch.

esteve, 2019-06-14:
@Karsten1987 I've pushed #763 as a first draft to solve this issue

closed 2019-06-20 by Karsten1987