ros2/rclcpp #766: Cannot create subscription to member function with extra argument using std::bind

sloretz, 2019-06-20:
## Bug report

**Required Info:**

- Operating System:
  - Bionic
- Installation type:
  - Binaries
- Version or commit hash:
  - 0.7.6-1bionic.20190612.221051
- DDS implementation:
  -  Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

<details><summary>Build this node, uncommenting each attempt to create the callback one at a time</summary>

```c++
#include <rclcpp/rclcpp.hpp>

#include <std_msgs/msg/string.hpp>

#include <functional>

class BindBug : public rclcpp::Node
{
public:
  explicit BindBug(const rclcpp::NodeOptions & options)
  : rclcpp::Node("bind_bug", options)
  {
    // First callback
    ///*
    auto callback = std::bind(
      &BindBug::on_msg,
      this,
      1,
      std::placeholders::_1);
    //*/

    // Second callback
    /*
    std::function<void(StringMsg::ConstSharedPtr)> callback = std::bind(
      &BindBug::on_msg,
      this,
      2,
      std::placeholders::_1);
    */

    // third callback
    /*
    auto callback =
      [this](StringMsg::ConstSharedPtr msg) {
        on_msg(3, msg);
      };
    */

    // Fourth callback
    /*
    auto callback = std::bind(
      &BindBug::reversed_on_msg,
      this,
      std::placeholders::_1,
      4);
    */

    // Fifth callback
    /*
    std::function<void(StringMsg::ConstSharedPtr)> callback = std::bind(
[... 273 lines cut ...]
        ^~~
/opt/ros/dashing/include/rclcpp/any_subscription_callback.hpp:150:8: note:   template argument deduction/substitution failed:
/opt/ros/dashing/include/rclcpp/any_subscription_callback.hpp:148:17: error: no type named 'type' in 'struct std::enable_if<false, void>'
     >::type * = nullptr
                 ^~~~~~~
/opt/ros/dashing/include/rclcpp/any_subscription_callback.hpp:148:17: note: invalid template non-type parameter
In file included from /usr/include/c++/7/future:48:0,
                 from /opt/ros/dashing/include/rclcpp/executors.hpp:18,
                 from /opt/ros/dashing/include/rclcpp/rclcpp.hpp:144,
                 from /home/sloretz/workspaces/dashing/src/lifecycle_manager/src/bind_bug.cpp:1:
/usr/include/c++/7/bits/std_function.h:541:2: error: 'std::function<_Res(_ArgTypes ...)>::_Requires<std::function<_Res(_ArgTypes ...)>::_Callable<typename std::decay<_U1>::type>, std::function<_Res(_ArgTypes ...)>&> std::function<_Res(_ArgTypes ...)>::operator=(_Functor&&) [with _Functor = rclcpp::create_subscription_factory(CallbackT&&, const rclcpp::SubscriptionEventCallbacks&, typename rclcpp::message_memory_strategy::MessageMemoryStrategy<CallbackMessageT, Alloc>::SharedPtr, std::shared_ptr<PublisherT>) [with MessageT = std_msgs::msg::String_<std::allocator<void> >; CallbackT = std::_Bind<void (BindBug::*(BindBug*, std::_Placeholder<1>, int))(std::shared_ptr<const std_msgs::msg::String_<std::allocator<void> > >, long unsigned int)>&; Alloc = std::allocator<void>; CallbackMessageT = std_msgs::msg::String_<std::allocator<void> >; SubscriptionT = rclcpp::Subscription<std_msgs::msg::String_<std::allocator<void> > >; typename rclcpp::message_memory_strategy::MessageMemoryStrategy<CallbackMessageT, Alloc>::SharedPtr = std::shared_ptr<rclcpp::message_memory_strategy::MessageMemoryStrategy<std_msgs::msg::String_<std::allocator<void> >, std::allocator<void> > >]::<lambda(rclcpp::node_interfaces::NodeBaseInterface*, const string&, const rcl_subscription_options_t&)>; _Res = std::shared_ptr<rclcpp::SubscriptionBase>; _ArgTypes = {rclcpp::node_interfaces::NodeBaseInterface*, const std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&, const rcl_subscription_options_t&}; std::function<_Res(_ArgTypes ...)>::_Requires<std::function<_Res(_ArgTypes ...)>::_Callable<typename std::decay<_U1>::type>, std::function<_Res(_ArgTypes ...)>&> = std::function<std::shared_ptr<rclcpp::SubscriptionBase>(rclcpp::node_interfaces::NodeBaseInterface*, const std::__cxx11::basic_string<char>&, const rcl_subscription_options_t&)>&]', declared using local type 'rclcpp::create_subscription_factory(CallbackT&&, const rclcpp::SubscriptionEventCallbacks&, typename rclcpp::message_memory_strategy::MessageMemoryStrategy<CallbackMessageT, Alloc>::SharedPtr, std::shared_ptr<PublisherT>) [with MessageT = std_msgs::msg::String_<std::allocator<void> >; CallbackT = std::_Bind<void (BindBug::*(BindBug*, std::_Placeholder<1>, int))(std::shared_ptr<const std_msgs::msg::String_<std::allocator<void> > >, long unsigned int)>&; Alloc = std::allocator<void>; CallbackMessageT = std_msgs::msg::String_<std::allocator<void> >; SubscriptionT = rclcpp::Subscription<std_msgs::msg::String_<std::allocator<void> > >; typename rclcpp::message_memory_strategy::MessageMemoryStrategy<CallbackMessageT, Alloc>::SharedPtr = std::shared_ptr<rclcpp::message_memory_strategy::MessageMemoryStrategy<std_msgs::msg::String_<std::allocator<void> >, std::allocator<void> > >]::<lambda(rclcpp::node_interfaces::NodeBaseInterface*, const string&, const rcl_subscription_options_t&)>', is used but never defined [-fpermissive]
  operator=(_Functor&& __f)
  ^~~~~~~~
make[2]: *** [CMakeFiles/bind_bug.dir/src/bind_bug.cpp.o] Error 1
make[1]: *** [CMakeFiles/bind_bug.dir/all] Error 2
make: *** [all] Error 2
```
</details>

Fifth callback (Reversed argument order with bind assigned to std::function) works fine
```
[INFO] [bind_bug]: Callback with [5] message 'Hello world'
```

### Additional information

wjwwood, 2019-07-11:
I think this is the issue that may be related:

https://github.com/ros2/rclcpp/issues/583

And the ROS answer referenced from that issue:

https://answers.ros.org/question/308386/ros2-add-arguments-to-callback/

Karsten1987, 2019-09-10:
sam thing as #273 ?

dirk-thomas, 2019-09-10:
To summarize: I think the failing cases are not supported since we can't ensure at compile time that the signature using `std::bind` is correct. Therefore I will go ahead and close this ticket.

Please feel free to continue commenting on it and it can be reopened if necessary.

closed 2019-09-10 by dirk-thomas