ros2/rclcpp #773: Can't call spin_until_future_complete inside a callback executed by an executor

orduno, 2019-06-28:
## Feature request

#### Feature description

`spin_until_future_complete` does not work recursively:

https://github.com/ros2/rclcpp/blob/0ccac1e3bd0f4314e11c86202c0c3ade57aed8ed/rclcpp/include/rclcpp/executors.hpp#L76

Could someone provide some background why this is not allowed?

orduno, 2019-08-07:
@wjwwood Could you provide some background on this? What's the root of the problem? We might be able to help.

nuclearsandwich, 2019-08-08:
Re-ping @wjwwood :bowing_man:

wjwwood, 2019-08-21:
I'm going to have to assume some understanding of how the executors and callback groups work to keep this reasonably brief. There are a few explanations floating around the internet, but I think the best thing to do if you want to understand them better is to start with the two most commonly used and just trace where the code goes in their spin functions:

- https://github.com/ros2/rclcpp/blob/65188b021da5879ec95a9a4fc0ca93ff79cc4e5f/rclcpp/src/rclcpp/executors/single_threaded_executor.cpp#L26-L39
- https://github.com/ros2/rclcpp/blob/65188b021da5879ec95a9a4fc0ca93ff79cc4e5f/rclcpp/src/rclcpp/executors/multi_threaded_executor.cpp#L41-L108

The problem with recursive spinning is different in those two executors.

----

For the single threaded executor, the issue is actually recursion, the code looks something like this:

- spin()
  - wait for work
  - match work with user callback
  - take data and call user callback
    - in user callback, call spin()  <-- recursion happens here

This spin() method is not designed to be re-entrant.

Also, you can too easily run into a dead lock using the single threaded executor, because if you did something like:

```c++
void my_callback(const SomeMessageType & message) {
  // ...
  auto future = service->async_send_request(request);
  spin_until_future_complete(future);
  // ...
}
```

And if the service callback and the above subscription callback are in the same mutually exclusive callback group, then even though you're spinning, the service would never be handled.

----

For the multi threaded executor, the issue is thread safety because only one thread may wait for work at the same time, and while you're executing your callback another thread is likely waiting for work to be ready. This means you may not be able to get work in your thread. This isn't such an issue, but it would require the implementation to be restructured to allow for this.

However, it suffers from the same callback group deadlock as the single threaded executors.

Also while waiting for work you're consuming the thread you're in which could be used to execute work otherwise (especially in the same callback group).

----

The "right" solution to this, in my opinion, requires an async/await syntax in C++, which doesn't exist right now. You can look at `asyncio` in Python3 or the new async/wait syntax in rust for what I mean. But basically we need the callbacks to be coroutines which we suspend while waiting for the condition to be met (some future is completed) and after that return to the callback and finish it. Since this is a language feature we can't really rely on that, but it would be the ideal way.

So ignoring that as a solution, I'd have to really dig into the architecture of the executor in order to figure out what we need to do in order to properly support this.

tgroechel, 2023-07-27:
Throwing out another way to "get around" this using async services + `SharedFutureResponse`: https://github.com/ros2/rclcpp/pull/1709#discussion_r1276537731