ros2/rclcpp #779: Couldn't make/build rclcpp local using master branch

popeaaa, 2019-07-07:
## Bug report

**Required Info:**

- Operating System:
  -  Ubuntu 18.04
- Installation type:
  -  binaries
- Version or commit hash:
  - current branch:master
- DDS implementation:
  - Fast RTPS
- Client library (if applicable):
  - rclcpp
#### Steps to reproduce issue
git clone https://github.com/ros2/rclcpp.git
cd rclcpp
mkdir build
cd build
cmake ..
wait cmake finish his job...
then, use make
and it prints a ERROR

#### Expected behavior
I tried to build rclcpp on my machine.but it failed.
make and make install should work fine
but now it didn't
#### Actual behavior
here is the "make" command prints :
/home/rbq/workspace/ROS2/source_codes_git/rclcpp/rclcpp/src/rclcpp/subscription_base.cpp: In member function ‘rmw_qos_profile_t rclcpp::SubscriptionBase::get_actual_qos() const’:
/home/rbq/workspace/ROS2/source_codes_git/rclcpp/rclcpp/src/rclcpp/subscription_base.cpp:134:35: error: ‘rcl_subscription_get_actual_qos’ was not declared in this scope
   const rmw_qos_profile_t * qos = rcl_subscription_get_actual_qos(subscription_handle_.get());
                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/rbq/workspace/ROS2/source_codes_git/rclcpp/rclcpp/src/rclcpp/subscription_base.cpp:134:35: note: suggested alternative: ‘rcl_subscription_get_options’
   const rmw_qos_profile_t * qos = rcl_subscription_get_actual_qos(subscription_handle_.get());
                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                   rcl_subscription_get_options
CMakeFiles/rclcpp.dir/build.make:1123: recipe for target 'CMakeFiles/rclcpp.dir/src/rclcpp/subscription_base.cpp.o' failed
make[2]: *** [CMakeFiles/rclcpp.dir/src/rclcpp/subscription_base.cpp.o] Error 1
CMakeFiles/Makefile2:1424: recipe for target 'CMakeFiles/rclcpp.dir/all' failed
make[1]: *** [CMakeFiles/rclcpp.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

christophebedard, 2019-07-07:
If you're using all binaries except `rclcpp` @ `master`, it's not going to work because `*subscription_get_actual_qos()` was added only after the latest release. See ros2/rmw#177.

AFAIK you're going to have to build all of these from source:

* ros2/rmw
* ros2/rmw_implementation
* ros2/rcl
* ros2/rclcpp
* ros2/rmw_fastrtps

ivanpauno, 2019-07-08:
If you're building from source, you have to build the complete ROS2 workspace as detailed in the [install instructions](https://index.ros.org/doc/ros2/Installation/Dashing/Linux-Development-Setup/). Trying only to rebuild one package and using the other packages from binaries can easily end in this kind of problems.
As @christophebedard said, `colcon` is the recommended build tool.

I'm closing the issue.

closed 2019-07-08 by ivanpauno