ros2/rclcpp #780: MultiThreadedExecutor with wall timer hung

Mygao, 2019-07-09:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04.2 LTS
- Installation type:
  - binaries
- Version or commit hash:
  - Dashing 0.7.6
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

This can be reproduced by modifying ros2 composition examples source code: SingleThreadedExecutor->MultiThreadedExecutor (https://github.com/ros2/examples/blob/master/rclcpp/minimal_composition/src/composed.cpp)
```

#include <memory>
#include "minimal_composition/publisher_node.hpp"
#include "minimal_composition/subscriber_node.hpp"
#include "rclcpp/rclcpp.hpp"

int main(int argc, char * argv[])
{
  rclcpp::init(argc, argv);
  rclcpp::executors::MultiThreadedExecutor exec;
  rclcpp::NodeOptions options;
  auto publisher_node = std::make_shared<PublisherNode>(options);
  auto subscriber_node = std::make_shared<SubscriberNode>(options);
  exec.add_node(publisher_node);
  exec.add_node(subscriber_node);
  exec.spin();
  rclcpp::shutdown();
  return 0;
}

```

#### Expected behavior
Message output goes on and on with count increasing without hanging
#### Actual behavior
Message output hangs. For example, 
ros2 run examples_rclcpp_minimal_composition composition_composed 
[INFO] [publisher_node]: Publisher: 'Hello, world! 0'
[INFO] [subscriber_node]: Subscriber: 'Hello, world! 0'
[INFO] [publisher_node]: Publisher: 'Hello, world! 1'
[INFO] [subscriber_node]: Subscriber: 'Hello, world! 1'

Stops here.

#### Additional information
I tested on 2 systems: Desktop and Laptop with both Ubuntu 18.04.2 LTS. Same ROS2 Dashing setups.

----
## Feature request

#### Feature description

#### Implementation considerations

gavanderhoorn, 2019-07-09:
Context: [[ros2] multi threaded executor with single node always makes the wall-timer hung.](https://answers.ros.org/question/327935) on ROS Answers.

Related (according to OP): #702.

ivanpauno, 2019-07-09:
I couldn't reproduce the problem in master.

> * Installation type:
>   
>   * binaries

Did you use the binaries of the first release, or Dashing release patch 1 binaries?

> #### Steps to reproduce issue
> This can be reproduced by modifying ros2 composition examples source code: SingleThreadedExecutor->MultiThreadedExecutor (https://github.com/ros2/examples/blob/master/rclcpp/minimal_composition/src/composed.cpp)

Did you only download this repo (`examples`) and build it in a overlay workspace of the binary installation? Please, share details of how you built it.

Mygao, 2019-07-10:
The first release was installed with debian package installation (https://index.ros.org/doc/ros2/Installation/Dashing/Linux-Install-Debians/).
I cloned the examples repo and built it with colcon. I used the first release binaries of ROS2 as environment.
Now I tried with newer release of ROS2 Dashing with 'apt upgrade' command and same problem occurs.
I think I should try building ROS2 from source.

Mygao, 2019-07-11:
I tried with ROS2 built from source (latest-release) and the example above(running with MultiThreadedExecutor) worked fine.

ivanpauno, 2019-07-11:
> Now I tried with newer release of ROS2 Dashing with 'apt upgrade' command and same problem occurs.

I will also try from binaries, to double check.

> I tried with ROS2 built from source (latest-release) and the example above(running with MultiThreadedExecutor) worked fine.

Thanks for checking it! I will try to figure out what PR solved the problem, and add it to the patch release 2 list (if it's not already there).

ivanpauno, 2019-07-11:
> I will also try from binaries, to double check.

I confirm that it isn't working when installing from debians. I have not clue about what PR solved the problem.

nuclearsandwich, 2019-07-25:
> > I tried with ROS2 built from source (latest-release) and the example above(running with MultiThreadedExecutor) worked fine.

> Thanks for checking it! I will try to figure out what PR solved the problem, and add it to the patch release 2 list (if it's not already there).

@ivanpauno if they were building `release-latest` from source then it should be identical to the current dashing release and there wouldn't be any PRs that are not already part of the binaries.

@Mygao or @ivanpauno have either of you tried reproducing this issue with the Dashing binary archive? https://github.com/ros2/ros2/releases/tag/release-dashing-20190614

That might add an additional data point.

ivanpauno, 2019-07-25:
> @ivanpauno if they were building `release-latest` from source then it should be identical to the current dashing release and there wouldn't be any PRs that are not already part of the binaries.

I didn't try with `release-latest`. I just checked with `master` (from source), and it was working.

> @Mygao or @ivanpauno have either of you tried reproducing this issue with the Dashing binary archive? https://github.com/ros2/ros2/releases/tag/release-dashing-20190614

No, I only tried with debians. I haven't tried with the binary archive.

nuclearsandwich, 2019-07-25:
> I didn't try with `release-latest`. I just checked with `master` (from source), and it was working.
> 

Trying with release-latest from source to confirm @Mygao's findings seems prudent as well to avoid chasing shadows.

ivanpauno, 2019-07-25:
I tried from source using `release-latest`, and I couldn't reproduce the bug.

ivanpauno, 2019-09-13:
@Mygao could you check if this still happening after the last patch release?

eric1221bday, 2019-09-24:
@ivanpauno I actually just ran into this issue and I'm using the latest debian of dashing as far as I can tell

eric1221bday, 2019-09-24:
OK so I just found this #836 which is the PR that fixed this issue. So it looks like we should probably put this in the patch release

closed 2019-09-24 by ivanpauno

nuclearsandwich, 2019-12-05:
This fix was released in Dashing Patch 4 but the issue wasn't removed from the board so it got bumped forward. I'm taking it off now but the fix was released.