ros2/rclcpp #813: Crash in callback group pointer vector iterator

guillaumeautran, 2019-08-02:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 16.04
- Installation type:
  - From Source (Crystal P4) But Master branch has the same issue (from code inspection)
- Version or commit hash:
  - HEAD of Master for RCLCPP
- DDS implementation:
  - N/A
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
There is 2 separate issues here:
 1. A sever crash can occur in the RCLCPP code when a thread is iterating over a callback_group while another thread is creating a new subscription (or timer, service, etc). The iterator over the actual callback group pointer vector is done unprotected by a lock causing a potential concurrent access leading to a crash.
 2. A memory leak is also present in the callback group pointer vector storing weak reference to subscriptions. If the subscription gets destroyed, the weak reference becomes invalid, however, that weak reference object is never removed from the callback group pointer vector. This overtime leads to memory leaks when subscriptions are added / removed dynamically. 

Reproduction step is by using the multithreaded executor as well as separate regular posix threads to create new topic subscriptions. The actual code change is self explanatory.

dirk-thomas, 2019-08-02:
Thank you for the report.

> The actual code change is self explanatory.

Please consider to contribute a pull request.

guillaumeautran, 2019-08-02:
@dirk-thomas I was too slow :) PR attached.