ros2/rclcpp #827: cannot publish data

hyunseok-yang, 2019-08-20:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Binaries(Debian package)
- Version or commit hash:
  - Dashing latest
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
In my workspace, I usually run gazebo + driver(for simualtor) on **terminal 1**. And I launch the navigation2 package on terminal 2.

```

```

#### Expected behavior
No termination on **terminal 1**
(I've never seen like this error message before with crystal version..)

Do you guys think it's a DDS issue(for example Fast RTPS?)?

#### Actual behavior

Not always...... but I got this error message and process died occasionally on **terminal 1** when I shutdown the navigation2(on terminal 2) by Ctrl-C and re-launch the navigation2 (on terminal 2).

[launch.sh-1] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[launch.sh-1]   what():  failed to publish message: cannot publish data, at /tmp/binarydeb/ros-dashing-rmw-fastrtps-shared-cpp-0.7.5/src/rmw_publish.cpp:52, at /tmp/binarydeb/ros-dashing-rcl-0.7.6/src/rcl/publisher.c:257

[mcu_porter_sim-3] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[mcu_porter_sim-3]   what():  failed to publish message: cannot publish data, at /tmp/binarydeb/ros-dashing-rmw-fastrtps-shared-cpp-0.7.5/src/rmw_publish.cpp:52, at /tmp/binarydeb/ros-dashing-rcl-0.7.6/src/rcl/publisher.c:257

#### Additional information
I configured default NodeOptions like below.
` : Node("micom_driver", rclcpp::NodeOptions())`

I configured QoS on publisher like below.
```
  auto qos = rclcpp::QoS(rclcpp::QoSInitialization::from_rmw(rmw_qos_profile_sensor_data));
  pubOdometryMsg = this->create_publisher<nav_msgs::msg::Odometry>("/odom", qos);
```

----
## Feature request

#### Feature description

#### Implementation considerations

ivanpauno, 2019-08-20:
Similar to https://github.com/ros2/ros2/issues/714.
It seems to be an error in FastRTPS.

dirk-thomas, 2019-09-06:
Closing as duplicate of ros2/ros2#714.

closed 2019-09-06 by dirk-thomas