ros2/rclcpp #828: Support more than one argument in RCLCPP_COMPONENTS_REGISTER_NODE

xhuan28, 2019-08-21:
## Feature request

#### Feature description

It is mentioned in [register_node_macro.hpp](https://github.com/ros2/rclcpp/blob/master/rclcpp_components/include/rclcpp_components/register_node_macro.hpp#L27) that, 
>  Valid arguments for NodeClass shall:
    **Have a constructor that takes a single argument that is a `rclcpp::NodeOptions` instance.**

Considering the scenario that if we need to remap the node name and namespace in the compile time, there are 3 arguments(node name, namespace and node options) for the component class. So I am wondering if it is feasible to support more than a single argument, otherwise the node which has multiple arguments in constructor can not implemented as a component. How do you think it only support the argument of `rclcpp::NodeOptions`?

jacobperron, 2019-08-21:
The recommended  way for remapping node names is at run-time (for example, with the command line or in a launch file). 

For example, with the CLI tool:

    ros2 component load /ComponentManager foo Foo --node-name new_node_name --node-namespace my_namespace

I can't think of a use-case for registering the same component multiple times with different names. If there is a case where you need to provide different names at compile-time, outside of the component use-case, then I suggest you add multiple constructors: one constructor accepting a single NodeOptions argument and another constructor with the node name, namespace, and NodeOptions.

jacobperron, 2019-08-21:
I'm closing this for now, but can re-open if there's a need for this that I'm not seeing.

closed 2019-08-21 by jacobperron