ros2/rclcpp #846: Creating `rclcpp_action` clients and servers in a `Node` constructor

jdlangs, 2019-09-04:
## Feature request

Similar to the [issue encountered with SyncParametersClient](https://github.com/ros2/rclcpp/issues/599), `rclcpp_action::create_client` and `create_server` expect an `rclcpp::Node::SharedPtr` when called. This is a problem for any class that inherits from `rclcpp::Node` as it cannot use `shared_from_this()` in its constructor to obtain a `shared_ptr` to pass on.

This can be resolved if `rclcpp_action` provides overloads that take a raw pointer. Since the current implementations directly access the node interfaces instead of keeping the `shared_ptr` around, it looks like another brief overload is all that's needed.

jacobperron, 2019-09-10:
This is possible using the signature accepting node interfaces. See [this demo code](https://github.com/ros2/demos/blob/97651912ebad439a6bc62c34671cad6a5bfee785/action_tutorials/action_tutorials_cpp/src/fibonacci_action_client.cpp#L39-L44) for an example:

```c++
this->client_ptr_ = rclcpp_action::create_client<Fibonacci>(
  this->get_node_base_interface(),
  this->get_node_graph_interface(),
  this->get_node_logging_interface(),
  this->get_node_waitables_interface(),
  "fibonacci");
```

jdlangs, 2019-09-10:
Hi @jacobperron, I'm unsure how to interpret you closing this issue since it was a request for a specific enhancement. Are you saying that changing the API to enable the use of raw pointers is not desired / will not be accepted? I'll note that the rclcpp parameter clients also have constructors that take the node interfaces but provide raw pointer overloads as well.

Karsten1987, 2019-09-10:
I think a templated solution could work here as well.

```c++
template<typename MessageT, typename NodePointerT>
auto create_client(NodePointerT node, std::string name)
{
  // static assert that node indeed is a pointer type
  static_assert(rcpputils::is_pointer<declval(node)>::value);

  return rclcpp::create_client<MessageT>(
    node->get_node_base_interface(),
    node->get_node_graph_interface(),
    node->get_node_logging_interface(),
    node->get_node_waitables_interface(),
    name));
}
```

We applied similar things to [diagnostics](https://github.com/ros/diagnostics/blob/dashing/diagnostic_updater/include/diagnostic_updater/diagnostic_updater.hpp#L365-L382) to take a templated pointer to a node. That template can either be deducted as a raw pointer or shared pointer.

jacobperron, 2019-09-11:
@jdlangs Sorry, I misread the request as more a lack of a way to do something.
I believe we would accept a PR to add an additional API for convenience. I like @Karsten1987's suggestion for a templated solution.

jacobperron, 2019-11-21:
I believe this ticket has been addressed by #892.

closed 2019-11-21 by jacobperron