ros2/rclcpp #851: parameter_blackboard node parameter cannot be set because it was not declared

fujitatomoya, 2019-09-13:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu18.04
- Installation type:
  - Source
- Version or commit hash:
  - [master repo](https://github.com/ros2/ros2/blob/master/ros2.repos)
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp, demo_nodes_cpp

#### Steps to reproduce issue

Terminal-1:
```
root@6cc2c62d0dd0:~/ros2_colcon_ws# ros2 run demo_nodes_cpp parameter_blackboard
[INFO] [parameter_blackboard]: Parameter blackboard node named '/parameter_blackboard' ready, and serving '1' parameters already!
[WARN] [rclcpp]: Failed to set parameter: parameter 'foo' cannot be set because it was not declared
```

Terminal-2:
```
root@6cc2c62d0dd0:~/ros2_colcon_ws# ros2 param set /parameter_blackboard foo 1
Setting parameter failed: parameter 'foo' cannot be set because it was not declared
```

#### Expected behavior

parameter should be set in success.

#### Actual behavior

parameter does not get set cz parameter is undeclared.

#### Additional information

parameter_blackboard specifies NodeOption allow_undeclare_parameter, so it should be able to set parameter without error.

the following fix might be related,

https://github.com/ros2/rclcpp/commit/b214324bf268e6342c366e50c986844211597859

## Feature request

W.I.P

#### Feature description

N.A

#### Implementation considerations

N.A

ivanpauno, 2019-09-13:
I've been able to reproduce the issue

fujitatomoya, 2019-09-20:
https://github.com/ros2/demos/pull/389 can fix this problem.

fujitatomoya, 2019-10-02:
@ivanpauno 

this can be closed.

closed 2019-10-04 by ivanpauno