ros2/rclcpp #855: Parameters in LifecycleNodes

buschbapti, 2019-09-17:
Hi all,

I am trying to add parameters to a node following the demo at https://github.com/ros2/demos/blob/master/demo_nodes_cpp/src/parameters/set_and_get_parameters.cpp

Problem is, the node is based on the Lyfecycle interface and I suspect rclcpp::SyncParametersClient does not accept them yet. I have seen that this problem appears in multiple other places as the NodeInterface is not yet fully integrated (see 
 https://github.com/ros-perception/image_common/issues/108 for example)

For reference, the code I use is:

```
this->parameters_client_ = std::make_shared<rclcpp::SyncParametersClient>(this);
```

where ```this``` extends from ```rclcpp_lifecycle::LifecycleNode```. I have also tried with ```std::make_shared<rclcpp::SyncParametersClient>(this->shared_from_this());``` in vain. The error is listed as follow:

```
/opt/ros/eloquent/include/rclcpp/parameter_client.hpp:201:12: note:   no known conversion for argument 1 from ‘std::shared_ptr<rclcpp_lifecycle::LifecycleNode>’ to ‘rclcpp::Node::SharedPtr {aka std::shared_ptr<rclcpp::Node>}’
```

which makes me suggest it is the same problem as https://github.com/ros-perception/image_common/issues/108

Thanks in advance for your support.

buschbapti, 2020-02-20:
Hello,

Any news on this issue? I am still unable to declare the parameter server in Lifecycle nodes and that is becoming problematic as this is an important feature. One workaround would be to declare a service for each parameters but it is bit tedious.

Thanks in advance.

fujitatomoya, 2020-02-25:
i guess that is related to https://github.com/ros2/rclcpp/issues/898, the problem is that we have separated classes rclcpp::Node and rclcpp_lifecycle::LifercycleNode. but i believe that rclcpp_lifecycle::LifercycleNode is built on top of rclcpp::Node. so rclcpp_lifecycle::LifecycleNode Class inherits based on rclcpp::Node and featured with lifecycle interfaces.

friendsocket, 2020-02-28:
I noticed this problem and began to deal with it these days.

As @buschbapti metioned, we can extend the constructor **SyncParametersClient(...)** to accept only rclcpp::Node. Just like the way this [file](https://github.com/ros2/rclcpp/blob/master/rclcpp/include/rclcpp/executor.hpp) handled. It is easy to implement. 

If you want the **rclcpp_lifecycle::LifecycleNode** as parameter of constructor **SyncParametersClient(...)**, just to be consistent with the original constructor that **rclcpp::Node** takes as an argument. Need to do more work. 
Cause the file **parameter_client.hpp/cpp** is located in the **rclcpp** package, which was compiled before pacakge **rclcpp_lifecycle**.

Here are some thoughts. 
1. Copy the relevant files to the package **rclcpp_lifecycle** and modify the interface to support **rclcpp_lifecycle :: LifecycleNode** as parameter. The disadvantage is obvious, this will generate redundant code with the same logic.
2. We can put the common files that Node and LifecycleNode depend on in a new package. Both Node and LifecycleNode are built on this new packages. The build sequence of package will be **new package-> rclcpp_lifecycle-> rclcpp-> others**. Instead of **rclcpp -> others**.

Any thoughts on this ?

PR #1249 (2020-08-05): Make parameter clients use template constructors
Fixes #855 for both synchronous or asynchronous parameter clients.
closed 2020-08-05 by audrow