ros2/rclcpp #860: Parameters not set from the command-line using new syntax

jacobperron, 2019-09-20:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - source
- Version or commit hash:
  - 
- DDS implementation:
  - N/A
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

Create a simple node that echos a parameter:

```
#include <iostream>
#include <memory>

#include "rclcpp/rclcpp.hpp"

class FooNode : public rclcpp::Node
{
public:
  explicit FooNode(const rclcpp::NodeOptions options = rclcpp::NodeOptions())
  : Node("foo_node", options)
  { 
    auto param_value = this->declare_parameter("foo", "bar");
    std::cout << "param value: " << param_value << std::endl;
  }
};

int main(int argc, char** argv)
{
  rclcpp::init(argc, argv);

  auto node = std::make_shared<FooNode>();

  rclcpp::spin(node);

  return 0;
}
```

Run the node and try to set the parameter "foo":

```
./foo --ros-args -p foo:=jacob
```

#### Expected behavior

The node prints the value set from the command line:

```
param value: jacob
```

#### Actual behavior

The node prints the default value:

```
param value: bar
```

#### Additional information

Passing the parameter via a YAML file works fine, e.g

*params.yaml*
```
foo_node:
  ros__parameters:
    foo: "jacob"
```

and running

```
./foo --ros-args --params-file params.yaml
```

jacobperron, 2019-09-20:
@hidmic FYI

hidmic, 2019-09-20:
Alright, it looks like in between changes to `rcl` and how the command line looks like, the last round of changes was left forgotten in my local clones of `rclcpp` and `rclpy`. Which also means things are not being tested properly. My bad. I'll clean it up, add tests and send it.

jacobperron, 2019-09-20:
> I'll clean it up, add tests and send it.

:bowing_man:

jacobperron, 2019-10-03:
@hidmic Looks like this was fixed? Though I don't know by what changes exactly.

hidmic, 2019-10-03:
Yes it was! By https://github.com/ros2/rcl/pull/508 and https://github.com/ros2/rclcpp/pull/865. Closing.

closed 2019-10-03 by hidmic

hamaney, 2020-04-02:
I see this issue in rclpy. 
I don't know how to check the version/branch I am on, but I installed ros2 using 
`curl -s https://raw.githubusercontent.com/ros/rosdistro/master/ros.asc | sudo apt-key add -`
    `sudo sh -c 'echo "deb [arch=amd64,arm64] http://packages.ros.org/ros2/ubuntu `lsb_release -cs` main" > /etc/apt/sources.list.d/ros2-latest.list'`

jacobperron, 2020-04-02:
@hamaney If you installed ROS 2 with `apt`, then the major version should be name of the package(s) you install (e.g. ros-**dashing**-desktop). Furthermore, it should also appear in the path of the setup file you source (e.g. source /opt/ros/**dashing**/setup.bash). If you are still not sure which version of ROS 2 you have installed, you can try echoing the environment variable `ROS_DISTRO` (e.g. on Linux: `echo $ROS_DISTRO`).

As pointed out on ROS answers, this method for passing ROS parameters is only supported since ROS 2 Eloquent. So if you are using ROS 2 Dashing, it will not work. If you continue to have problems, please comment on ROS Answers (or create a new question) with details about your installation. This lets other users benefit from the answer. (GitHub tickets are not as visible)

jacobperron, 2020-04-04:
@hamaney No worries. Because ROS follows a federated model for packages, each package has a version number, but ROS distro as a whole just has a name (e.g. Dashing). If you want to know the particular version of a package, you can use the following `ros2pkg`:

 ```sh
# Get the version of rclcpp
ros2 pkg xml rclcpp --tag version
```