ros2/rclcpp #863: rclcpp_compnents: test_api consistently times out

mjcarroll, 2019-09-20:
#  Bug report

**Required Info:**

- Operating System:
  - Linux
- Installation type:
  - Source
- Version or commit hash:
  - Dashing
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

* Execute `test_component_manager_api`

#### Expected behavior

Test does not time out

#### Actual behavior

Test times out

#### Additional information

This was originally reported as a build_cop issue: https://github.com/ros2/build_cop/issues/184

It's recently become more flaky, causing issues with PR builders and CI jobs (https://github.com/ros2/rclcpp/pull/857) with an attempted fix here https://github.com/ros2/rclcpp/pull/862

hidmic, 2019-09-24:
So, it's reported for Fast-RTPS and I cannot reproduce it locally using any other `rmw` implementation i.e. RTI Connext and ADLink Opensplice work just fine. I suspect it has to do with the time it takes for Fast-RTPS to bring up / tear down node state, but I have to dig in further.

hidmic, 2019-09-25:
Alright, some debugging suggests it's a race. Running just two tests, the first one passes and the second gets as far as sending a service request. The service server never gets that request, spinning times out and the test ends up waiting for an `std::future` that never comes (no pun intended).

I went into Fast-RTPS, looking for global state that could explain an interaction between consecutively run tests but I couldn't find any. Maybe eProsima folks can shed some light.

hidmic, 2019-09-25:
@richiware maybe?

richiware, 2019-09-29:
We are backporting fixed issues on FastRTPS 1.9 to [1.8](https://github.com/eProsima/Fast-RTPS/issues/709). ROS2 Dashing is using FastRTPS 1.8.x releases. Can you test with this [branch](https://github.com/eProsima/Fast-RTPS/tree/1.8.x)? It will be next 1.8.2 release. In case it still fails, I will try to reproduce the problem and take a look.

hidmic, 2019-10-01:
Thanks for reaching out @richiware! But it seems to me you're suggesting that this is specific to Fast-RTPS 1.8.x and, by extension, to Dashing. This is also happening on Eloquent. I just noticed the issue was not clear in this regard -- it's updated now.

hidmic, 2019-10-07:
FYI no other flakes have happened after https://github.com/ros2/rclcpp/pull/876. We could close this issue BUT I think it'd be good to track the underlying race which we are simply working around now.

richiware, 2019-10-09:
Yesterday I was able to reproduce it. I will try to figure out what's going on.

richiware, 2019-10-09:
I was testing modifying the default Durability QoS for services. Testing with TRANSIENT_LOCAL instead of VOLATILE makes pass all tests.
Therefore the service's request is sent before services are discovered between them (at DDS level). When the discovered is complete, due to Durability QoS is VOLATILE, there is no request to be sent again.

richiware, 2019-10-09:
It is weird because looking in the test's source code I've found `wait_for_service()` is being used. I'm not familiar with ROS services API, but this function should wait until the discovery is complete between client and server.

richiware, 2019-10-09:
Sounds plausible. If the graph doesn't have time to be informed about the teardown of the service, maybe `wait_for_service()` is returning immediately, when it shouldn't.

dirk-thomas, 2019-10-09:
From a quick look at both tests in `rclcpp_components` I noticed that they only call `rclcpp::init`. How about adding the corresponding `rclcpp::shutdown` call?

If either of you was able to reproduce the test failure before please consider trying it again with the patch proposed in #885.

hidmic, 2019-10-09:
@dirk-thomas `SetUpTestCase()` and `TearDownTestCase()` get called once per _test case_, where _test case_ in `gtest` means all the tests associated with a `::testing::TestCase` fixture subclass (not the same as `SetUp()` and `TearDown()`). Thus, in a setting like the one we had before https://github.com/ros2/rclcpp/pull/876, it won't make a difference. We could `init()` and `shutdown()` before and after each test though.

I'm concerned about the underlying issue though. If 

> All of this seems to suggest that graph updates are driven solely by network discovery. In other words, there's no intraprocess optimization. Thus the graph is essentially out-of-date until a full discovery is performed.

is true, it breaks `wait_for_*()` methods' contract (or the expectation at least).

wjwwood, 2019-10-09:
I agree with @hidmic, I would be surprised if shutdown in `TearDownTestCase()` made a difference, but we can still try it out.

I think it's more likely to be a discovery race condition, no? Like services from previous runs are still around at the beginning of the next test (that would explain why sleeping between tests fixes it). Unless fast rtps does something to "flush" things out in shutdown, I don't think even init/shutdown would fix this, but I could be wrong.

hidmic, 2019-10-09:
> Like services from previous runs are still around at the beginning of the next test (that would explain why sleeping between tests fixes it)

That's what I thought, but after browsing Fast-RTPS code for a while I can't realize how's that possible. And if that's the case, I think it'd be really inconvenient if there's no way to establish a "reliable" link.

dirk-thomas, 2019-10-09:
> Like services from previous runs are still around at the beginning of the next test

That exact case would benefit from this patch. In the repeated CI builds we run a test N times. So when the test itself doesn't shutdown correctly and doesn't unadvertise itself it might affect the re-run of itself.

wjwwood, 2019-10-09:
> That exact case would benefit from this patch. In the repeated CI builds we run a test N times. So when the test itself doesn't shutdown correctly and doesn't unadvertise itself it might affect the re-run of itself.

But I don't think shutdown does anything additional that destroying the node doesn't do. I agree something needs to be done, but I don't think shutdown will do anything. But again we can certainly test it.

wjwwood, 2019-10-09:
In https://github.com/ros2/rclcpp/pull/885#issuecomment-540215504 @dirk-thomas said:

> Without a shutdown the node won't communicate to other entities that it is going down.

Destroying the node is all that is needed. Shutdown (as far as I know) does nothing additional to signal to other nodes that something no longer exists.