ros2/rclcpp #880: Launchfile for intra process communication and lifecylce nodes

buschbapti, 2019-10-04:
Hi all,

I have created 3 nodes that use intra process communication following the [available tutorial](https://index.ros.org//doc/ros2/Tutorials/Intra-Process-Communication/). I have created a single executable and added the 3 nodes in the executor and this works fine.

The 3 nodes I have create are lifecycle nodes, therefore if I use the `ros2 run` command I have afterwards to configure and activate each node manually. I wanted to use a launchfile to simplify this process but I can't figure out how to properly do it in this case. I have tried to launch them as normal nodes using:

```python
# Prepare the visualization node.
    visualizer_node = launch_ros.actions.LifecycleNode(
        node_name='visualizer',
        package='modulo_core', 
        node_executable="modulo_core_test_cartesian", 
        output='screen')

    # Prepare the robot node.
    robot_interface_node = launch_ros.actions.LifecycleNode(
        node_name='robot_interface',
        package='modulo_core', 
        node_executable="modulo_core_test_cartesian", 
        output='screen')

    # Prepare the motion generator node.
    motion_generator_node = launch_ros.actions.LifecycleNode(
        node_name='motion_generator',
        package='modulo_core',
        node_executable="modulo_core_test_cartesian", 
        output='screen')
```

but this will run `modulo_core_test_cartesian` 3 times resulting in 3 instances of each node conflicting with each other. I can't find of any other solution than to have to separate the nodes in different executables but then intra process communication is lost. 

Is what I am trying to achieve a feature that is not available yet?

Thanks in advance.

ivanpauno, 2019-10-04:
Currently nodes can be written as [components](https://index.ros.org//doc/ros2/Tutorials/Composition/), and components can be configured and loaded in a container. You can do all that in a launch file (see for example https://github.com/ros2/demos/blob/master/composition/launch/composition_demo.launch.py).

I don't remember how to set up intra-process communication when using components/containers in launch files (maybe @hidmic knows better).

Currently, if you manually compose nodes in one executable (as you're currently doing), there's no nice way to configure it in a launch file. There's a proposed refactor considering that, between other things, https://github.com/ros2/launch/issues/114.

buschbapti, 2019-10-04:
@ivanpauno Thanks for your answer. I will look more into compositions then. If we can enable intra process communication using this then it is definitely a better approach anyway as it allows to compose nodes on the fly without building an executable which is something I also want to achieve.

hidmic, 2019-10-04:
> I don't remember how to set up intra-process communication when using components/containers in launch files (maybe @hidmic knows better).

Pretty easily once https://github.com/ros2/rclcpp/pull/871 is in.

fujitatomoya, 2019-10-07:
@buschbapti 

i believe that you can do what you want as followings,

1. implement nodes with options.use_intra_process_comms(true)
2. compose these nodes as [compositions](https://index.ros.org//doc/ros2/Tutorials/Composition/).
3. ros2 launch your_package your_launch_file

if im missing something, let me know.

here are my examples, (not using lifecycle node)

https://github.com/fujitatomoya/demos/blob/topic-20191007-composition-intra-launch/composition/launch/composition_demo.launch.py
https://github.com/fujitatomoya/demos/commit/664e73db4e16fa7abf1bb2ea44ffb8e696b070e8

```
root@6cc2c62d0dd0:~/ros2_colcon_ws# ros2 launch composition composition_demo.launch.py 
[INFO] [launch]: All log files can be found below /root/.ros/log/2019-10-07-15-48-40-367537-6cc2c62d0dd0-28374
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container-1]: process started with pid [28385]
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/talker' in container '/my_container'
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/listener' in container '/my_container'
[component_container-1] [INFO] [my_container]: Load Library: /root/ros2_colcon_ws/install/composition/lib/libtalker_component.so
[component_container-1] [INFO] [my_container]: Found class: rclcpp_components::NodeFactoryTemplate<composition::Talker>
[component_container-1] [INFO] [my_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<composition::Talker>
[component_container-1] [INFO] [my_container]: Load Library: /root/ros2_colcon_ws/install/composition/lib/liblistener_component.so
[component_container-1] [INFO] [my_container]: Found class: rclcpp_components::NodeFactoryTemplate<composition::Listener>
[component_container-1] [INFO] [my_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<composition::Listener>
[component_container-1] Publishing: 'Hello World: 1' and address: 0x5558D8CF64E0
[component_container-1] I heard: [Hello World: 1] and address: 0x5558D8CF64E0
[component_container-1] Publishing: 'Hello World: 2' and address: 0x5558D8CF64E0
[component_container-1] I heard: [Hello World: 2] and address: 0x5558D8CF64E0
[component_container-1] Publishing: 'Hello World: 3' and address: 0x5558D8CF64E0
[component_container-1] I heard: [Hello World: 3] and address: 0x5558D8CF64E0
[component_container-1] Publishing: 'Hello World: 4' and address: 0x5558D8CF64E0
[component_container-1] I heard: [Hello World: 4] and address: 0x5558D8CF64E0
[component_container-1] Publishing: 'Hello World: 5' and address: 0x5558D8CF64E0
[component_container-1] I heard: [Hello World: 5] and address: 0x5558D8CF64E0
[component_container-1] Publishing: 'Hello World: 6' and address: 0x5558D8CF64E0
[component_container-1] I heard: [Hello World: 6] and address: 0x5558D8CF64E0
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[component_container-1] [INFO] [rclcpp]: signal_handler(signal_value=2)
[INFO] [component_container-1]: process has finished cleanly [pid 28385]
```

ivanpauno, 2019-10-07:
I think this issue can be closed.
Feel free to continue commenting or re-opening if you feel that's necessary.

closed 2019-10-07 by ivanpauno

MCP-TTC, 2020-03-04:
Hi,

I was able to run the composition from both command line and launch file with standard nodes developed as components.

But if I develop lifecycle nodes as components, I can't use composition from command line and launch file. I can compose these nodes only with a new custom node that instantiate the components from code.

Did anyone try to compose lifecycle nodes from command line and/or with a launch file?