ros2/rclcpp #908: Reliable publisher/subscriber-connection drops messages in high frequency scenarios

DensoADAS, 2019-10-28:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - deb packages or source
- Version or commit hash:
  - Eloquent or ros2 master branch (22. October 2019)
- DDS implementation:
  - FastRTPS, OpenSplice, RTI connext
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
Create a simple publisher/subscriber pair, both having reliable QoS settings. 
Publish a fixed number of messages on the publisher side with a high frequency (>1kHZ -> <1ms intervals (e.g. 1us)).

Example:
Talker:
```
class Talker : public rclcpp::Node
{
public:
    DEMO_NODES_CPP_PUBLIC
    explicit Talker(const rclcpp::NodeOptions& options)
        : Node("talker", options)
    {
        setvbuf(stdout, NULL, _IONBF, BUFSIZ);
        auto publish_message =
            [this]() -> void
            {
                auto msg = std::make_unique<std_msgs::msg::Int32>();
                msg->data = count_++;
                pub_->publish(std::move(msg));

                if (count_ > 100000)
                {
                    timer_->cancel();
                    RCLCPP_INFO(this->get_logger(), "Sent 100000 messages");
                }
            };

        rclcpp::QoS qos = rclcpp::QoS(rclcpp::KeepLast(1000)).reliable();
        pub_ = this->create_publisher<std_msgs::msg::Int32>("chatter", qos);

        RCLCPP_INFO(this->get_logger(), "Sending messages...");
        timer_ = this->create_wall_timer(1us, publish_message);
    }

private:
    int32_t count_ = 1;
    rclcpp::Publisher<std_msgs::msg::Int32>::SharedPtr pub_;
    rclcpp::TimerBase::SharedPtr timer_;
};
```

Listener:
```
class Listener : public rclcpp::Node
{
public:
    DEMO_NODES_CPP_PUBLIC
    explicit Listener(const rclcpp::NodeOptions& options)
        : Node("listener", options),
          m_messageID(0),
          m_numReceivedMsgs(0)
    {
        setvbuf(stdout, NULL, _IONBF, BUFSIZ);
        auto callback =
            [this](const std_msgs::msg::Int32::SharedPtr msg) -> void
            {
                ++m_numReceivedMsgs;
                int32_t msgDiff = msg->data - m_messageID;
                m_messageID = msg->data;
            };

        rclcpp::QoS qos = rclcpp::QoS(rclcpp::KeepLast(1000)).reliable();
        sub_ = create_subscription<std_msgs::msg::Int32>("chatter", qos, callback);
        RCLCPP_INFO(this->get_logger(), "Ready to receive messages");
    }

    ~Listener()
    {
        RCLCPP_INFO(this->get_logger(), "Received %d messages", m_numReceivedMsgs);
        RCLCPP_INFO(this->get_logger(), "Lost %d messages", 100000 - m_numReceivedMsgs);
    }

private:
    rclcpp::Subscription<std_msgs::msg::Int32>::SharedPtr sub_;
    int32_t m_messageID;
    int32_t m_numReceivedMsgs;
};
```
#### Expected behavior
The subscriber receives all messages in order or an exception is thrown (or some other error propagation method).

#### Actual behavior
Some messages are published but not received on the subscriber side without any notification to the user or thrown errors.
    Basically behaves similar to best_effort strategy.

#### Additional information

A full implementation example which reproduces the error can be found here:
https://github.com/DensoADAS/high_freq_pub_example.git
This is basically the Talker/Listener demo from the demo_nodes_cpp package with a different message type for easier counting and a higher pub rate.

This problem occurs with various middleware implementations (as well as with the eloquent binaries and the ros2 master) and usually gets worse the higher the publisher frequency is.
Also different QoS settings were tested (different depth/keepAll) but the problem still occurs (less frequently with keepAll but some messages are still dropped).
The problem can be observed already with normal publishing rates, e.g. intervals between published messages of about 1ms.

With the simple example above the following dropped message counts were observed:
```
Sent messages: 10000 -> repeating 5x:
 
Publisher interval: 1us:           
FastRTPS:       7541    7926    7192    8623    8255
OpenSplice:     4946    426     5181    3934    5253
RTI Connext:    8603    2755    1417    5355    5467 

Publisher interval: 1ms:
FastRTPS:       53      0       33      1474    61

Sent messages: 100000 -> repeating 5x:

Publisher interval: 1us   
FastRTPS:       96052   94911   96321   97322   95244
OpenSplice:     9369    7710    15950   51295   12835
RTI Connext:    45810   22624   41902   30460   21139
```

fujitatomoya, 2019-10-29:
@DensoADAS 

can you describe your platform detail? what hardware are you trying to use?

DensoADAS, 2019-10-29:
Multiple machines were tested, all got the issue:

1. OS: Ubuntu 18.04
    CPU: Intel Core i5-6500 @ 3.20GHz
    RAM: 24GB
2. OS: Ubuntu 18.04
    CPU: Intel Xeon E5-2680 v4 @ 2.40GHz
    RAM: 16GB
3. OS: Ubuntu 18.04 (virtual machine on Windows 10 host)
    CPU: Intel Core i7-6820HQ @ 2.70GHz
    RAM: 8GB

The software on all machines is up-to-date. The test was executed on all machines using the eloquent binaries and in the VM also with the latest ros2 master.

Tell me if you need more information.

claireyywang, 2019-11-15:
Thanks for reporting this issue! I was able to reproduce this behaviour by sending 10 msgs in frequencies ranging from 1us to 1s. The higher frequency messages were sent in, the more loss I observed. It seems like Opensplice has relatively the most reliable delivery among all middleware implementations, with Fastrtps being the worst. This also confirms the dropped msgs counts. I'm currently not sure what caused this problem but will look into it.

dirk-thomas, 2019-11-16:
I tried to reproduce the problem and the first thing I noticed is that the talker is starting to publish messages right away without waiting for the subscriber to be discovered and matched.

You can easily reproduce that by printing `msg->data` for the first received message in the listener - it is greater than 1 and the offset matches the missing message count exactly.

Adding a sleep in the talker after creating the publisher but before creating the timer (e.g. `std::this_thread::sleep_for(std::chrono::seconds(1));`) lets the discovery happen and match the two endpoints before the first message is published. With that change the listener receives all the message for me.

Please comment if you can reproduce this and if there is any problem remaining.

Tobi2001, 2019-11-16:
@dirk-thomas The sleep fixes the missing messages at the beginning but the general problem still exists. Messages are dropped (quite randomly as it seems) throughout the publishing process; you can check this by adding the commented out code in line 34 of the listener:
```
RCLCPP_ERROR(this->get_logger(), "Lost %d msgs, curr: %d", msgDiff - 1, msg->data);
```
With the sleep and 100000 sent messages I still get around 80000 dropped ones with FastRTPS and the eloquent binaries.

fujitatomoya, 2019-11-18:
@claireyywang @dirk-thomas @Tobi2001 

confirmed this problem with Fast-RTPS. the more frequency, the more losses happen.

claireyywang, 2019-11-19:
I looked into fastrtps and it is mentioned that [in high rate or large data scenarios, there are a couple of parameters need to be tuned to avoid network package drop](https://fast-rtps.docs.eprosima.com/en/v1.7.0/advanced.html#tuning). Based on the [documentation ](https://fast-rtps.docs.eprosima.com/en/latest/xmlprofiles.html#xml-profiles) and [presentation slides](https://roscon.ros.org/2017/presentations/ROSCon%202017%20ROS2%20Fine%20Tuning.pdf),  it is possible to tune those parameters using an `xml` file.

dirk-thomas, 2019-11-19:
> The sleep fixes the missing messages at the beginning but the general problem still exists.

I can't reproduce the problem. With the sleep in place (and the error message commented in) the listener still reports the following for me with no error messages:

```
[INFO] [listener]: Received 100000 messages
[INFO] [listener]: Lost 0 messages
```

claireyywang, 2019-11-19:
@DensoADAS  @Tobi2001 @fujitatomoya  

@dirk-thomas and I did more testing and found out that by default (without any args), `fastrtps` build type is `release`, which caused the ~70% msg drop. But if we build `debug` instead, which is what @dirk-thomas used, the performance will be much better: msg drop around 0-10%. 

Here are the steps we took:
``` bash
  cd ~/ros2_ws/build/ 
  rm -rf fastrtps  # delete fastrtps build folder from workspace
  cd ~/ros2_ws # go to the root of workspace
  colcon build --packages-select fastrtps --cmake-args -DCMAKE_BUILD_TYPE=Debug  # rebuild in debug type
```
Please let me know if you are able to get a much better performance after trying it.

dirk-thomas, 2019-11-19:
@richiprosima The behavior we are seeing here is pretty weird. When building FastRTPS with the build type `Release` it drops a very significant number of messages (~60%-70%). When building `Debug` it surprisingly behaves much better - it drops none (or sometimes up to 10%).

DensoADAS/high_freq_pub_example#1 contains the above example with the additional sleep to ensure the endpoints have matched.

I already tried to get rid of all conditional logic in FastRTPS (depending on `__DEBUG` and `NDEBUG`) but that didn't make any difference. Any insight would be more than welcome.

fujitatomoya, 2019-11-21:
@DensoADAS 

i confirmed that cyclonedds does not have any message drop.
i have tested more than 10 times with cyclonedds, can you confirm for double check?

```
[Subscriber]
root@6cc2c62d0dd0:~/ros2_colcon_ws# export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
root@6cc2c62d0dd0:~/ros2_colcon_ws# ros2 run high_freq_pub listener 
[INFO] [listener_high_freq]: Ready to receive messages
^C[INFO] [rclcpp]: signal_handler(signal_value=2)
[INFO] [listener_high_freq]: Received 100000 messages
[INFO] [listener_high_freq]: Lost 0 messages

[Publisher]
root@6cc2c62d0dd0:~/ros2_colcon_ws# export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp
root@6cc2c62d0dd0:~/ros2_colcon_ws# ros2 run high_freq_pub talker
[INFO] [talker_high_freq]: Sending messages...
[INFO] [talker_high_freq]: Sent 100000 messages
^C[INFO] [rclcpp]: signal_handler(signal_value=2)
```

@claireyywang @dirk-thomas 
it surely depends on dds implementation, need some help from each dds vendor.
do you want me to create issue for each rmw implementation such as rmw_fastrtps?

dirk-thomas, 2019-11-21:
@fujitatomoya please try to reproduce the behavior with the different build types of FastRTPS mentioned above before adding another dimension to the problem.

fujitatomoya, 2019-11-21:
@dirk-thomas 

> When building FastRTPS with the build type Release it drops a very significant number of messages (~60%-70%). When building Debug it surprisingly behaves much better - it drops none (or sometimes up to 10%).

confirmed almost same behavior.