ros2/rclcpp #910: Domain ID not changable on rclcpp::NodeOptions

rotu, 2019-10-30:
rclcpp::NodeOptions has no way to set the Domain ID on a per-node basis, and it's kludgy to get the domain ID `node_options->get_rcl_node_options()->domain_id`. This functionality is exposed on `rcl_node_options_t`.

wjwwood, 2019-11-15:
This makes sense to me, we'd appreciate anyone who has time to contribute this addition to the NodeOptions. We should consider doing it in `rclpy` at the same time, if it's not already there.

suab321321, 2019-11-17:
@wjwwood sir I would like to contribute

ivanpauno, 2019-11-18:
Note: When we finish with https://github.com/ros2/rmw/issues/183, one domain id per node will have no more sense, and it will be one domain id per context.

wjwwood, 2019-11-18:
That's a good point @ivanpauno. In that case maybe we should hold off on this until we have a little more clear direction on that set of changes.

@suab321321 it sounds like we'd like to hold off on this for now. Our intention is to follow through with ros2/rmw#183, in which case this issue will no longer make sense. There's a chance we change our mind, in which case this issue would make sense again, but that is unlikely in my opinion.

I added a note to the issue description of ros2/rmw#183 that we should follow up on this issue once we've committed to doing that work or not.

suab321321, 2019-11-18:
@wjwwood okay

fujitatomoya, 2019-12-17:
@wjwwood @ivanpauno 

kinda off topic from this thread, but the same goes to [rclpy](https://github.com/ros2/rclpy) ?

actually [rclpy](https://github.com/ros2/rclpy) does not have support for ROS_DOMAIN_ID (https://github.com/ros2/rclpy/issues/484). i think that ROS_DOMAIN_ID should be supported via [rcl](https://github.com/ros2/rcl) and frontend(rclxxx) can ask the domain id to rcl. (just to reduce redundancy of implementation.)

could you share your thoughts on this?

ivanpauno, 2019-12-17:
> actually rclpy does not have support for ROS_DOMAIN_ID (ros2/rclpy#484). i think that ROS_DOMAIN_ID should be supported via rcl and frontend(rclxxx) can ask the domain id to rcl. (just to reduce redundancy of implementation.)

The `ROS_DOMAIN_ID` logic is actually implemented in `rcl`, as you suggested.
If it's not working, it must be an error in how `rclpy` is wrapping `rcl` stuff.

fujitatomoya, 2019-12-18:
@ivanpauno 

> The ROS_DOMAIN_ID logic is actually implemented in rcl, as you suggested.

sorry my bad, i should have checked.

https://github.com/ros2/rcl/blob/024c3ee50d9b79a85665efe1c191cc1225026519/rcl/src/rcl/node.c#L262-L281

and the question is why rclcpp has the same code to get the ROS_DOMAIN_ID?

https://github.com/ros2/rclcpp/blob/0c66d0c725ad1f977e2b907bc0619d6ceb6de238/rclcpp/src/rclcpp/node_options.cpp#L325-L355

i believe that we could delete the redundant code in rclcpp?
if so, i will make the issue and PR against it.

ivanpauno, 2019-12-18:
> and the question is why rclcpp has the same code to get the ROS_DOMAIN_ID?

It's needed just for better introspection. Currently, the default node options have a domain id value of `SIZEMAX`. When that value is detected in the node creation, it's updated to the value indicated by the env variable. To be able to introspect correctly the domain id if you call `get_rcl_node_options` before creating a node, the value is also get from the environment in that case.

> i believe that we could delete the redundant code in rclcpp?
if so, i will make the issue and PR against it.

Yes, the redundant code can be avoided.
As part of https://github.com/ros2/rcl/pull/515, I'm creating an `rcl_domain_id` function, that allows you to get the default domain_id from the environment.
Code can easily be reused after that gets merged.

fujitatomoya, 2019-12-19:
@ivanpauno 

got it, i will get it done after your PR is merged.

thanks

fujitatomoya, 2020-06-04:
@ivanpauno 

i got a question, since Context is introduced and mapped to dds participant, I think that domain id also should be mapped to Context (same as security) not only for NodeOptions.

i am considering the following,

- expose domain_id to NodeOptions
  default `std::numeric_limits<size_t>::max()` and ROS_DOMAIN_ID variable is set via  get_domain_id_from_env(), but user can overwrite the domain_id if needed. actually rmw_fastrtps/cyclonedds does not use this domain_id via rmw_node_create, so this is for implementation that can support domain_id with node level.
- expose domain_id to InitOptions
  this is actually effective with dds implementation such as fastrtps or cyclonedds. default `std::numeric_limits<size_t>::max()` and ROS_DOMAIN_ID variable is set, and user can overwrite the domain_id if needed.

do you have any other thoughts?

ivanpauno, 2020-06-04:
We shouldn't expose nothing in `NodeOptions`, as that will not continue to be used.
Exposing it in `InitOptions` is ok, but I would prefer to have a cleaner API (i.e.: not exposing that size_t max means "default").
```cpp
class InitOptions
{
...
void use_default_domain_id();  // this will take the domain id from the env
void set_domain_id(size_t domain_id); 
size_t get_domain_id() const; 
...
}
```

fujitatomoya, 2020-06-07:
I do agree with you, so maybe it is better to clean anything related to ROS_DOMAIN_ID stuff from NodeOptions. and we also update docs (if exist) explicitly about domain id with Context for user.

fujitatomoya, 2020-06-07:
@suab321321 

sorry i missed it, are you up for this?

ivanpauno, 2020-06-08:
> I do agree with you, so maybe it is better to clean anything related to ROS_DOMAIN_ID stuff from NodeOptions. and we also update docs (if exist) explicitly about domain id with Context for user.

If you want to clean that up, it's welcomed too! The only reference too `domain_id` in node related API is in `rmw` IIRC.
Exposing the domain id here is not related with that change, so it can be done separately.

fujitatomoya, 2020-06-11:
@ivanpauno 

either me or someone else from our team can work on this, you can leave this on us. though, we would require your review for sure.

closed 2020-08-05 by ivanpauno

aprotyas, 2022-05-28:
Looks like the new API to support setting/getting domain IDs targeted post-Foxy distributions.

Does anyone involved in this thread know how to obtain the domain ID from a node in Foxy? The snippet below gives me a runtime error:

```cpp
size_t
get_domain_id_from_node(
  const rclcpp::Node & node)
{
  const rcl_init_options_t * rcl_init_options = 
    node.get_node_base_interface()->get_context()->get_init_options().get_rcl_init_options();

  std::size_t domain_id;
  // const_cast is safe because `rcl_init_options_get_domain_id` only reads the input structure
  rcl_ret_t ret =
    rcl_init_options_get_domain_id(
    const_cast<rcl_init_options_t *>(rcl_init_options),
    &domain_id);
  if (RCL_RET_OK != ret) {
    throw std::runtime_error("Failed to get domain ID from rcl_init_options");
  }
}
```

fujitatomoya, 2022-05-31:
https://github.com/ros2/rcl/pull/946 backport foxy has been merged, so it should be working. can we take a look at the stack trace via gdb?

aprotyas, 2022-05-31:
@fujitatomoya I expected it to work too, but maybe you would have some insights from the backtrace:

<details>
<summary>Output of gdb exec -ex "run" -ex "thread apply all bt" -ex "quit"</summary>

```gdb
$ gdb build/rclcpp_foxy_domain_id/rclcpp_foxy_domain_id -ex "run" -ex "thread apply all bt" -ex "quit"

GNU gdb (Ubuntu 9.2-0ubuntu1~20.04.1) 9.2
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from build/rclcpp_foxy_domain_id/rclcpp_foxy_domain_id...
Starting program: /home/aprotyas/dev/workspaces/ros2_playground_ws/build/rclcpp_foxy_domain_id/rclcpp_foxy_domain_id 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff5723700 (LWP 97916)]
[New Thread 0x7ffff4e99700 (LWP 97917)]
[New Thread 0x7ffff4698700 (LWP 97918)]
[New Thread 0x7ffff3e97700 (LWP 97919)]
[New Thread 0x7ffff3689700 (LWP 97920)]
[New Thread 0x7ffff2e88700 (LWP 97921)]
[New Thread 0x7ffff267a700 (LWP 97922)]
[New Thread 0x7ffff1d79700 (LWP 97923)]
[New Thread 0x7ffff1565700 (LWP 97924)]

Thread 1 "rclcpp_foxy_dom" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50

Thread 10 (Thread 0x7ffff1565700 (LWP 97924)):
#0  futex_wait_cancelable (private=<optimized out>, expected=0, futex_word=0x7fffe4000ba8) at ../sysdeps/nptl/futex-internal.h:183
#1  __pthread_cond_wait_common (abstime=0x0, clockid=0, mutex=0x7fffe4000bb0, cond=0x7fffe4000b80) at pthread_cond_wait.c:508
#2  __pthread_cond_wait (cond=0x7fffe4000b80, mutex=0x7fffe4000bb0) at pthread_cond_wait.c:647
#3  0x00007ffff7658e30 in std::condition_variable::wait(std::unique_lock<std::mutex>&) () from /lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff6f7567d in std::condition_variable::wait<rmw_fastrtps_shared_cpp::__rmw_wait(char const*, rmw_subscriptions_t*, rmw_guard_conditions_t*, rmw_services_t*, rmw_clients_t*, rmw_events_t*, rmw_wait_set_t*, const rmw_time_t*)::<lambda()> >(std::unique_lock<std::mutex> &, rmw_fastrtps_shared_cpp::<lambda()>) (this=0x7fffe4000b80, __lock=..., __p=...) at /usr/include/c++/9/condition_variable:101
#5  0x00007ffff6f75153 in rmw_fastrtps_shared_cpp::__rmw_wait (identifier=0x7ffff701bd00 "rmw_fastrtps_cpp", subscriptions=0x7ffff1564560, guard_conditions=0x7ffff1564570, services=0x0, clients=0x0, events=0x0, wait_set=0x7fffe4000b60, wait_timeout=0x0) at /home/aprotyas/dev/workspaces/ros2_foxy_ws/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/src/rmw_wait.cpp:173
#6  0x00007ffff6f33296 in node_listener (context=0x5555555da350) at /home/aprotyas/dev/workspaces/ros2_foxy_ws/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/src/listener_thread.cpp:136
#7  0x00007ffff6f34385 in std::__invoke_impl<void, void (*)(rmw_context_t*), rmw_context_t*> (__f=@0x5555556ba8e0: 0x7ffff6f33075 <node_listener(rmw_context_t*)>) at /usr/include/c++/9/bits/invoke.h:60
#8  0x00007ffff6f342d1 in std::__invoke<void (*)(rmw_context_t*), rmw_context_t*> (__fn=@0x5555556ba8e0: 0x7ffff6f33075 <node_listener(rmw_context_t*)>) at /usr/include/c++/9/bits/invoke.h:95
#9  0x00007ffff6f34221 in std::thread::_Invoker<std::tuple<void (*)(rmw_context_t*), rmw_context_t*> >::_M_invoke<0ul, 1ul> (this=0x5555556ba8d8) at /usr/include/c++/9/thread:244
#10 0x00007ffff6f341c3 in std::thread::_Invoker<std::tuple<void (*)(rmw_context_t*), rmw_context_t*> >::operator() (this=0x5555556ba8d8) at /usr/include/c++/9/thread:251
#11 0x00007ffff6f34194 in std::thread::_State_impl<std::thread::_Invoker<std::tuple<void (*)(rmw_context_t*), rmw_context_t*> > >::_M_run (this=0x5555556ba8d0) at /usr/include/c++/9/thread:195
#12 0x00007ffff765ede4 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#13 0x00007ffff7360609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#14 0x00007ffff749a133 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

Thread 9 (Thread 0x7ffff1d79700 (LWP 97923)):
#0  futex_wait_cancelable (private=<optimized out>, expected=0, futex_word=0x5555555dcd4c) at ../sysdeps/nptl/futex-internal.h:183
#1  __pthread_cond_wait_common (abstime=0x0, clockid=0, mutex=0x5555555dbc00, cond=0x5555555dcd20) at pthread_cond_wait.c:508
#2  __pthread_cond_wait (cond=0x5555555dcd20, mutex=0x5555555dbc00) at pthread_cond_wait.c:647
#3  0x00007ffff7658e30 in std::condition_variable::wait(std::unique_lock<std::mutex>&) () from /lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff64646c5 in std::_V2::condition_variable_any::wait<std::unique_lock<std::recursive_timed_mutex> > (this=0x5555555dcd20, __lock=...) at /usr/include/c++/9/condition_variable:273
#5  0x00007ffff64623d4 in eprosima::fastrtps::rtps::AsyncWriterThread::run (this=0x5555555dcc80) at /home/aprotyas/dev/workspaces/ros2_foxy_ws/src/eProsima/Fast-DDS/src/cpp/rtps/resources/AsyncWriterThread.cpp:135
[... cut ...]
#6  0x000055555555cda1 in get_domain_id_from_node (node=std::shared_ptr<class rclcpp::Node> (use count 1, weak count 1) = {...}) at /home/aprotyas/dev/workspaces/ros2_playground_ws/src/ros2_playground/rclcpp_foxy_domain_id/src/main.cpp:61
#7  0x000055555555ceef in main () at /home/aprotyas/dev/workspaces/ros2_playground_ws/src/ros2_playground/rclcpp_foxy_domain_id/src/main.cpp:70
A debugging session is active.

	Inferior 1 [process 97912] will be killed.

Quit anyway? (y or n) y
```

</details>

Since it's pretty long, you can access it here too: https://pastebin.ubuntu.com/p/bFjpD3Rsn3/

wjwwood, 2022-05-31:
@aprotyas you should check `rcl_get_error_string()` when an rcl function fails to see more info. The backtrace isn't useful because it's just showing us where you threw your own exception.

@aprotyas the latest version takes a pointer to const as the first argument:

https://github.com/ros2/rcl/blob/72cee8a9fbda40304ff4f8e7c5d12b6ce6410f8d/rcl/include/rcl/init_options.h#L152

vs foxy:

https://github.com/ros2/rcl/blob/9b0b151a49ed9013c78520e738658c948bd93116/rcl/include/rcl/init_options.h#L150

Maybe there are other backports missing.

aprotyas, 2022-05-31:
> @aprotyas you should check `rcl_get_error_string()` when an rcl function fails to see more info. The backtrace isn't useful because it's just showing us where you threw your own exception.

@wjwwood that's a good point, thanks! I figured out what happened from the `rcl_get_error_string()` and a Valgrind trace of my test program:

```
terminate called after throwing an instance of 'std::runtime_error'
  what():  init_options->impl argument is null, at /home/aprotyas/dev/workspaces/ros2_foxy_ws/src/ros2/rcl/rcl/src/rcl/init_options.c:149
Aborted (core dumped)
```

```
==98403== Memcheck, a memory error detector
==98403== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==98403== Using Valgrind-3.15.0 and LibVEX; rerun with -h for copyright info
==98403== Command: ./build/rclcpp_foxy_domain_id/rclcpp_foxy_domain_id
==98403== Parent PID: 1
==98403== 
==98403== Invalid read of size 8
==98403==    at 0x505C2CA: rcl_init_options_get_domain_id (init_options.c:149)
==98403==    by 0x110C4A: main (main.cpp:102)
==98403==  Address 0x5b1bc20 is 0 bytes inside a block of size 8 free'd
==98403==    at 0x483F1CF: operator delete(void*, unsigned long) (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==98403==    by 0x4DDB25D: std::default_delete<rcl_init_options_t>::operator()(rcl_init_options_t*) const (unique_ptr.h:81)
==98403==    by 0x4DDB13B: std::unique_ptr<rcl_init_options_t, std::default_delete<rcl_init_options_t> >::~unique_ptr() (unique_ptr.h:292)
==98403==    by 0x4DDACD5: rclcpp::InitOptions::~InitOptions() (init_options.cpp:78)
==98403==    by 0x110C22: main (main.cpp:98)
==98403==  Block was alloc'd at
==98403==    at 0x483DE63: operator new(unsigned long) (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==98403==    by 0x4DDA9E1: rclcpp::InitOptions::InitOptions(rcl_init_options_t const&) (init_options.cpp:34)
==98403==    by 0x4DDAB19: rclcpp::InitOptions::InitOptions(rclcpp::InitOptions const&) (init_options.cpp:44)
==98403==    by 0x4DA088C: rclcpp::Context::get_init_options() (context.cpp:275)
==98403==    by 0x110BFD: main (main.cpp:98)
```

The problem is this line:

```cpp
const rcl_init_options_t * rcl_init_options = 
    node.get_node_base_interface()->get_context()->get_init_options().get_rcl_init_options();
```

`get_init_options()` invokes an `InitOptions` copy constructor, which eventually calls:

https://github.com/ros2/rclcpp/blob/5c1dd19456f276f878b00dbea4e8fc1ac2230308/rclcpp/src/rclcpp/init_options.cpp#L33-L34

Since this constructor allocates new memory, when this line terminates, the `rclcpp::InitOptions` instance terminates and the memory allocated for the `struct rcl_init_options_t` is free'd.

Replacing the problematic line with a block that allows me to get domain ID before the InitOptions goes out of instance works properly:

```cpp
const auto init_options = node->get_node_options().context()->get_init_options();                                                                                               
const rcl_init_options_t * rcl_init_options_ptr = __init_options.get_rcl_init_options();

// get_domain_id  
```

---

The error makes sense to me, but it was a bit unexpected. Should this behavior be expected?