ros2/rclcpp #912: [Dashing] Inter-process comms force copies in intra-process comms

hidmic, 2019-11-05:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Binaries
- Version or commit hash:
  - 0.7.11
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
See  https://github.com/ros2/demos/issues/412.

#### Expected behavior
Inter-process communication should not result in copies during intra-process communications.

#### Actual behavior
Copies occur.

#### Additional information
Copies seem to happen because, when both inter-process and intra-process communications are in use, messages published as `std::unique_ptr` are promoted to `std::shared_ptr`:

https://github.com/ros2/rclcpp/blob/fadd923aa0e0ef9fff1b1ef7860ed577ac455ea6/rclcpp/include/rclcpp/publisher.hpp#L116-L119

which then invariably forces copies when intra-process subscriptions take an `std::unique_ptr` as well:

https://github.com/ros2/rclcpp/blob/fadd923aa0e0ef9fff1b1ef7860ed577ac455ea6/rclcpp/include/rclcpp/intra_process_manager.hpp#L353-L360

even when there's only one subscription:

https://github.com/ros2/rclcpp/blob/fadd923aa0e0ef9fff1b1ef7860ed577ac455ea6/rclcpp/include/rclcpp/mapped_ring_buffer.hpp#L187-L197

As of Eloquent, the new intra-process comms implementation solved this via [`do_intra_process_publish_and_return_shared()`](https://github.com/ros2/rclcpp/blob/2ed445647400bacbff6420cfb8ccc7bee86ee469/rclcpp/include/rclcpp/experimental/intra_process_manager.hpp#L244).

hidmic, 2019-11-05:
So, backporting an experimental implementation doesn't sound like a great idea to me, but to solve this in the best possible way we'd need to get closer to that implementation i.e. being able to decide whether to copy or not based on the nature of the subscriptions.

dirk-thomas, 2019-11-05:
On the Dashing branch you either have to wait for the inter process publish call to return (which adds significant latency) or copy the data if it is requested externally (see https://github.com/ros2/rclcpp/blob/fadd923aa0e0ef9fff1b1ef7860ed577ac455ea6/rclcpp/include/rclcpp/publisher.hpp#L106-L111). The former was the case some time ago but the extra latency was considered to be less desired. The latter is the state in Dashing. Without a significant refactor (aka what has happened for Eloquent) I don't think you can achieve both.

hidmic, 2019-11-05:
Right. There's another option though, which is to copy `std::unique_ptr`-referred messages for inter-process communication to use. That improves owning publisher, single owning intra-process subscriber performance (that of the intra-process demo), but it penalizes owning publishers, multiple not-owning intra-process subscribers performance. 

Alternatively, we can mark this as `won't fix`. Do we have a place to list well-known bugs?

dirk-thomas, 2019-11-05:
I don't mind which of the two options we choose.

> Do we have a place to list well-known bugs?

Yes, each release page has a known issues section.

hidmic, 2019-11-13:
Alright, leaving a `won't fix` here and adding this as a well-known issue in Dashing (see https://github.com/ros2/ros2_documentation/pull/416).

nuclearsandwich, 2019-11-20:
Since the issue is publicly listed and not going to be fixed. Can the issue itself be closed?

hidmic, 2019-11-21:
Now that I merged https://github.com/ros2/ros2_documentation/pull/416, yes, I think we can close.

closed 2019-11-21 by hidmic