ros2/rclcpp #913: Intra-process subscriber does not call get_actual_qos() while publisher does, causing gtest_intra_process to fail

malsbat, 2019-11-06:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - source
- Version or commit hash:
  - 2ed445647400bacbff6420cfb8ccc7bee86ee469
- DDS implementation:
  - DPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
Run gtest_intra_process__rmw_dps_cpp. 

#### Expected behavior
Test passes.

#### Actual behavior
Test fails.

#### Additional information
The test fails due to can_communicate returning false from the qos reliability check.  The publisher qos is the get_actual_qos from the the RMW layer while the subscriber qos is the one supplied by the application (not get_actual_qos).

ivanpauno, 2019-11-18:
@malsbat Can you open a PR proposing a fix?

Thanks!

ivanpauno, 2019-11-18:
I saw you have already opened #914

wjwwood, 2019-12-12:
Can this be closed?

ivanpauno, 2019-12-12:
> Can this be closed?

Yes, forgo about this. Closing!

closed 2019-12-12 by ivanpauno