ros2/rclcpp #927: Android compile error rcl_allocator_t

machinekoder, 2019-11-19:
## Bug report

**Required Info:**

- Operating System:
  Ubuntu Bionic 18.04
- Installation type:
  From source
- Version or commit hash:
  latest `dashing` branch
- DDS implementation:
  Fast-RTPS
- Client library (if applicable):
  rclcpp

#### Steps to reproduce issue
That's the hard part to describe since Android build/install is hardly documented.

However, I get this error when compiling a Qt application for Android with ROS dependencies. It already worked.

#### Expected behavior

No compile errors.

#### Actual behavior
Compilation fails at include of the `rclcpp.h`:
```
../../../bin/ros2_android/ros2_android_ws/install_isolated/rclcpp/include/rclcpp/node_options.hpp:331:58: error: cannot convert 'rcutils_allocator_t' to 'void* (*)(size_t, void*) {aka void* (*)(unsigned int, void*)}' in initialization
   rcl_allocator_t allocator_ {rcl_get_default_allocator()};
```
I hope the `rcl_get_default_allocator` error triggers something. I'm not exactly sure where to look for the problem.

Note that I have to use QMAKE for this application, so I had to add all the ROS2 libs and includes "by hand". It already compiled and worked until recently. I haven't changed the Android ROS2 Installation, but I remember updating the systems ROS2 `dashing` install. Not sure it has something to do with that.

ivanpauno, 2019-11-20:
I can't imagine what the problem is from the error message.
Having more information, and details on how to reproduce the problem would be great.

dirk-thomas, 2019-12-05:
@machinekoder Please provide reproducible steps otherwise there is not much we can do about this.

wjwwood, 2020-01-22:
I think we need a reproducible case or more context. Just looking at the error it seems like something is passing a function pointer where instead it should be a struct. Is it possible that you changed what `rcl_get_default_allocator()` returns as part of your effort to get it working on android?

I'm going to close this because we cannot reproduce or diagnose. Feel free to continue to comment and I can reopen if necessary.

closed 2020-01-22 by wjwwood