ros2/rclcpp #940: Sim time always zero

maxlein, 2019-12-09:
## Bug report  

My simulation time is always zero although `use_sim_time` is set.  
It doesn't matter if I gazebo is running ( with ros plugins ) or not, always zero.

I also couldn't find anything in examples that would start something like a clock publisher.  

I posted this question some time ago also on ros answers [here.](https://answers.ros.org/question/331532/ros2-using-sim-time-parameter-nodes-time-is-always-zero/)  

**Required Info:**

- Operating System:
  - Ubuntu 18.04, Kernel 5.0
- Installation type:
  - binaries
- Version or commit hash:
  - eloquent
- DDS implementation:
  - CycloneDDS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

```
#include <cstdio>
#include <rclcpp/logging.hpp>
#include <rclcpp/node.hpp>
#include <rclcpp/rclcpp.hpp>

using namespace std::chrono_literals;

int main(int argc, char ** argv)
{
   rclcpp::init(argc, argv);

   rclcpp::Node node("timeTest");

   rclcpp::Parameter simTime( "use_sim_time", rclcpp::ParameterValue( true ) );
   node.set_parameter( simTime );
   auto useSimTime = node.get_parameter( "use_sim_time" ).as_bool();

   auto periodic_timer = node.create_wall_timer(
           100ms,
           [&]() {
               RCLCPP_INFO(node.get_logger(), "time %.2f.%d useSimTime: %ld", node.now().seconds(), node.now().nanoseconds(), useSimTime );
               RCLCPP_INFO(node.get_logger(), "Clock Time is %ld", rclcpp::Clock(RCL_ROS_TIME).now().nanoseconds());

           });

   rclcpp::spin(node.get_node_base_interface());

   rclcpp::shutdown();
  return 0;
}
```

Output:  

```  
[INFO 1575900167.873207694] [timeTest]: time 0.00.0 useSimTime: 1
[INFO 1575900167.873348031] [timeTest]: Clock Time is 1575900167873347711
[INFO 1575900167.973085505] [timeTest]: time 0.00.0 useSimTime: 1
[INFO 1575900167.973201782] [timeTest]: Clock Time is 1575900167973201292
[INFO 1575900168.072304368] [timeTest]: time 0.00.0 useSimTime: 1
[INFO 1575900168.072391229] [timeTest]: Clock Time is 1575900168072390973
[INFO 1575900168.172527744] [timeTest]: time 0.00.0 useSimTime: 1
[INFO 1575900168.172618773] [timeTest]: Clock Time is 1575900168172618553
```

#### Expected behavior
Node returns simulation time.  

#### Actual behavior
Time is always zero.

hidmic, 2019-12-09:
@maxlein is anyone publishing on the `/clock` topic?

maxlein, 2019-12-09:
Good point, no one is publishing on /clock.  
Shouldn't gazebo do that when launching with ros plugin?  

I start gazebo like that:  

```  
ld.add_action(ExecuteProcess(
        cmd=['gzserver', '--verbose', '-s', 'libgazebo_ros_factory.so', world],
        output=os.environ['LOG_TARGET']
    ) )
```

maxlein, 2019-12-09:
Ok, i figured that out.  
Took it from a bad example.  
I somehow thought ros_factory derives from ros_init and is therefore not needed..
  
This is correct:  ( without the factory, I can't spawn a shuttle )

```  
start_gazebo_server_cmd = ExecuteProcess(
        condition=IfCondition(use_simulator),
        cmd=['gzserver', '-s', 'libgazebo_ros_init.so', '-s', 'libgazebo_ros_factory.so', world],
        cwd=[launch_dir], output='screen')
```

closed 2019-12-09 by maxlein