ros2/rclcpp #941: Node and Lifecycle node is_sim_time() method

SteveMacenski, 2019-12-09:
## Feature request

#### Feature description

- A method to get if the current time is sim time, ei `is_sim_time()`
- Would help as a `Node` and `LifecycleNode` level method rather than trying to mess around with lower level clocks. 

Is is a correct understanding, it would be correct to say that:

```
bool is_sim_time() {
  return get_clock()->ros_time_is_active();
}
```

Yes?

wjwwood, 2020-01-02:
I'm going to close this for now, because this is a single instance of a more common case of "why isn't there a method to do X on the Node class?" Which we plan to address in an API review of `rclcpp`.

The thinking here at Open Robotics right now is that we want to move towards a syntax more like:

```c++
get_clock(my_node).ros_time_is_active();
```

Or extending to this feature request:

```c++
is_sim_time(my_node);
```

where `my_node` maybe of type `rclcpp::Node` or it might be another type like `rclcpp_lifecycle::LifecycleNode`.

The reason for this being that keeping the `Node` and `LifecycleNode` interfaces in sync is not trivial and the obvious solution to that, i.e. inheritance, has been avoided for other reasons in favor of composition. By using template methods which take a "node like" we can reduce the repeated code and allow users to create new types of nodes in the future while reusing much of the existing convenience functions.

closed 2020-01-02 by wjwwood

SteveMacenski, 2020-01-04:
> I'm going to close this for now, because this is a single instance of a more common case of "why isn't there a method to do X on the Node class?" 

I primarily opened this to start that conversation.  I think this, above other things I've seen missing, would have an immediately high value and I was planning on implementing it myself if I got the OK from the maintainers' end that this is wanted. 

I think we agree that this is something that should be added to the API review, but I'll also do it in the intermediate time to get it out there. Up to you if you'd rather wait and have it be part of a larger discussion. An aside, I'd love to be involved in that conversation. I have many thoughts to the point I've started writing an `rclEZ` `rclcpp` wrapper to make application level code clearer to algorithm developers. 

`get_clock(my_node).ros_time_is_active();` I think that would make sense for an internal `rclcpp` level, but as an applications/algorithm ROS developer, I think that's a bit superfluous for the simple thing its trying to accomplish. We refer to simulation time in parameters and documentation as `use_sim_time` not `ros_time_is_active`, which I think is a bit unintuitive. Having the entry level API just be `is_sim_time(node)` and `node->is_sim_time()` would be much more concise and less confusing, to me at least.

wjwwood, 2020-01-22:
> Having the entry level API just be `is_sim_time(node)` and `node->is_sim_time()` would be much more concise and less confusing, to me at least.

I think that `is_sim_time(node)` is totally fine to have. `node->is_sim_time()` is not as desirable since you need to have a method on all kinds of `node`. Which isn't too bad for one function, but just doesn't scale. Imagine we added a `supports_sim_time()` (just as an example), in the first case it's just one free function that takes a template typed node, in the latter case you need to add a method to at least two additional classes (`Node` and `LifecycleNode`), and maybe more in the future.

> If you would rather me not implement this before an API review, then I understand to close the ticket if you can link me to where this was added as a discussion point for that review. If you'd rather me implement it and include it additionally to validate in the API review, I think it should be reopened as the point of discussion for implementation.

If you want to implement something now, that's fine, but I'd personally prefer to see a `use_sim_time()` free function rather than a new method on the `Node` class. We don't have an agenda for the API review yet, so I can't add it to that.

Do you think this needs to be reopened?

dirk-thomas, 2020-01-22:
> I'd personally prefer to see a `use_sim_time()` free function rather than a new method on the `Node` class.

The current idea to use free functions taking a node argument has certainly many advantages. I just want to mention that it also comes with a downside: reduced discoverability.

For methods you can easily type `node->` and completion will tell you what methods are there to be involved.

For free functions that is not the case. You can't easily find all functions which can take a specific `node`argument.

wjwwood, 2020-01-22:
That's true, but you can at least take the longer route in that case, e.g.:

```
node->get_X_interface()->some_task_related_to_X()
```

But it's true higher level functions aren't as easy to find.

SteveMacenski, 2020-01-22:
> I'd personally prefer to see a use_sim_time() free function rather than a new method on the Node class.

See, I prefer the opposite. On application level code using this API, I'm probably deriving my base class from `Node` or `LifecycleNode`. Having it be a class method would allow then the statement
```
this->is_sim_time();
``` 

Which is more clear than
```
rclcpp::is_sim_time(shared_from_this());
```

which is now really a bit convoluted IMO for what should be a straight forward operation. I understand from a maintainers standpoint having just 1 templated free method is easier, but I think for the main external facing API, the focus should be on user experience and readability. 

With that said, I don't necessarily agree that all these methods that could be free methods would be easier maintained that way. I could see a world where `LifecycleNodeInterface` and `NodeInterface` derives from a base `NodeishInterface` which implements some of these core things that care consistent across all `NodeIsh` implementations, for now representing lifecycle and regular nodes.

Anyhow, that's a little off topic. The ticket can stay closed, it doesn't sound like we have consensus if or how this should be implemented

ivanpauno, 2020-01-23:
>Which is more clear than
>```
>rclcpp::is_sim_time(shared_from_this());
>```

It will actually be:
```cpp
rclcpp::is_sim_time(this);
```
And I don't see much difference between the method style and this.

Just to have some context, why an inheritance based solution was discarded to mantain `Node` and `LifeCycleNode` api closer? I guess there are some old discussions about that.

> The current idea to use free functions taking a node argument has certainly many advantages. I just want to mention that it also comes with a downside: reduced discoverability.

Yeah, I agree that's a downside of the free function style.

PS: We have to push the c++ standard to accept the [unified call proposal](https://isocpp.org/blog/2016/02/a-bit-of-background-for-the-unified-call-proposal) :rofl:.