ros2/rclcpp #944: std::bind with new arguments in service callback

SteveMacenski, 2019-12-16:
## Bug report

**Required Info:**

- Operating System:
  -  Ubuntu 18.04
- Installation type:
  -  Source, master
- Version or commit hash:
  - 
- DDS implementation: Fast RTPS
  - 
- Client library (if applicable): rclcpp
  - 

#### Steps to reproduce issue

Hi,

Having an issue with binding service callbacks with multiple arguments. So when I do what's shown below, it compiles but crashes at runtime looking for some symbols. I thought "huh, that's funky" and found #273 that seems very similar but not quite the same. 

I thought maybe it was how I was creating the bind, so I tried an experiment doing then exact same thing but with a subscriber and `std_msgs::msg::String` with the same additional callback arguments, which worked without an issue. So I think this is an issue specific to services. 

It compiles find without warnings, but fails at runtime when it hits this line of the configure lifecycle step.

```
    std::function<void(const std::shared_ptr<rmw_request_id_t>,
      std_srvs::srv::SetBool::Request::SharedPtr,
      std_srvs::srv::SetBool::Response::SharedPtr)> toggle_srv_callback;

    toggle_srv_callback = std::bind(
      &SpatioTemporalVoxelLayer::BufferEnablerCallback, this,
      _1, _2, _3, _observation_buffers.back());
    std::string toggle_topic = source + "/toggle_enabled";
    auto server = rclcpp_node_->create_service<std_srvs::srv::SetBool>(
      toggle_topic, toggle_srv_callback);

void SpatioTemporalVoxelLayer::BufferEnablerCallback(
  const std::shared_ptr<rmw_request_id_t> request_header,
  const std::shared_ptr<std_srvs::srv::SetBool::Request> request,
  std::shared_ptr<std_srvs::srv::SetBool::Response> response,
  const std::shared_ptr<buffer::MeasurementBuffer> buffer) {}

```

#### Expected behavior

Create the service at runtime, rather than crash

#### Actual behavior

Crashes with error about couldn't find symbols

```
[controller_server-8] /home/s.macenski/Documents/nav/navigation2_ws/install/nav2_controller/lib/nav2_controller/controller_server: symbol lookup error: /home/s.macenski/Documents/nav/navigation2_ws/install/spatio_temporal_voxel_layer/lib/libspatio_temporal_voxel_layer_core.so: undefined symbol: _ZN22rosidl_typesupport_cpp31get_service_type_support_handleIN8std_srvs3srv7SetBoolEEEPK29rosidl_service_type_support_tv

```

#### Additional information

The code snippet that actually worked using a subscriber

```
    std::function<void(std_msgs::msg::String::SharedPtr)> toggle_srv_callback;
    toggle_srv_callback = std::bind(
      &SpatioTemporalVoxelLayer::BufferEnablerCallback, this,
      _1, _observation_buffers.back());
    std::string toggle_topic = source + "/toggle_enabled";
    auto server = rclcpp_node_->create_subscription<std_msgs::msg::String>(
     toggle_topic, rclcpp::SystemDefaultsQoS(), toggle_srv_callback);

  void SpatioTemporalVoxelLayer::BufferEnablerCallback(
   const std_msgs::msg::String::SharedPtr /*hi*/,
   const std::shared_ptr<buffer::MeasurementBuffer> /*buffer*/) {}
```

SteveMacenski, 2019-12-16:
And I found the issue.

Looks like I didn't include `find_package(std_srvs REQUIRED)`. I wouldn't have thought it'd compile without it given the snippet below, clearly called and included

```
#include "std_srvs/srv/set_bool.hpp"
...
    auto server = rclcpp_node_->create_service<std_srvs::srv::SetBool>(
      toggle_topic, toggle_srv_callback);
```

This ticket came up in research https://answers.ros.org/question/321261/ros2-crystal-member-service-linking-rosidl_typesupport_cpp/ to help me find that. 

Is there not a way to have it fail on compile for something like that? I'm not entirely clear why it didn't fail to compile.

closed 2019-12-16 by SteveMacenski

wjwwood, 2020-01-22:
I think it is because the type support is dynamically loaded (in the case that there is more than one rmw implementation) and that doesn't happen until runtime. See:

https://github.com/ros2/rosidl_typesupport/blob/1769ba822b64573a450b8fe6a716d92deba9256e/rosidl_typesupport_cpp/src/type_support_dispatch.hpp#L72

But that's only used if there are multiple rmw implementations to select from. If you build with just one, then it may have failed at compile time.