ros2/rclcpp #958: Links in rclcpp_lifecycle API documentation don't exist

SteveMacenski, 2019-12-30:
Hi,

In the index (https://index.ros.org/p/rclcpp_lifecycle/) if you click on API docs it gives you a 404 (http://docs.ros2.org/dashing/api/rclcpp_lifecycle/). However this works for rclcpp, actions, etc. I think this is likely because of a missing Doxyfile (https://github.com/ros2/rclcpp/blob/master/rclcpp_action/Doxyfile) that the other elements in rclcpp repo contain, unless there's something also in the backend stopping the documentation being hosted on that hyperlink.

Reporting from ROS Answers (https://answers.ros.org/question/340779/lifecyclenode-rclcpp_lifecycle-api-docs-returns-404/?comment=340806#post-id-340806)

jacobperron, 2020-01-03:
Currently, there's no automated doc jobs and we manually build documentation for a small subset of packages. I checked and rclcpp_lifcycle is not in the set of packages we're building docs for, so this explains the 404. This the case for many packages; for example the API link for [launch](https://index.ros.org/p/launch/) also results in a 404.

If there's a strong desire, I can add rclcpp_lifecycle to the list for next time someone manually generates docs (it's not a scalable solution obviously).

closed 2020-01-03 by jacobperron

SteveMacenski, 2020-01-03:
Ah, got it. Does the index webpage not know there’s no valid docs links? Having a bunch of hyperlinks 404 may result in alot of superfluous tickets. 

I would think lifecycle should be added to the manual list since its a heavily used interface

jacobperron, 2020-01-04:
> Does the index webpage not know there’s no valid docs links? Having a bunch of hyperlinks 404 may result in alot of superfluous tickets.

I'm not familiar with the implementation of ROS index. If there is no API documentation to reference, then the dead link should probably not be there. Though, I'm not sure how trivial this is considering API docs are hosted separately from the index. Consider opening an issue at https://github.com/ros-infrastructure/rosindex