ros2/rclcpp #961: std::make_shared calls are now [[nodiscard]] in MSVC and cause build warnings

brawner, 2020-01-08:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10
- Installation type:
  - Source install, MSVC 2019, 16,24
- Version or commit hash:
  - master
  
#### Issue
std::make_shared is now [annotated as [[nodiscard]]](https://github.com/microsoft/STL/blob/master/stl/inc/memory#L1583) on Windows MSVC 2019 and will cause [MSBuild warnings](https://citest.ros2.org/job/ci_windows/70/warnings43Result/file.-168900435/). I can't tell if that's part of the STL spec, or if this is something that MS has decided to do.

This was noticed on the citest farm instance I'm running in a docker container. It installs the most recent VS BuildTools which may be ahead of the ci.ros2.org version.

This affects test_node.cpp and test_parameter_client.cpp
https://github.com/ros2/rclcpp/blob/master/rclcpp/test/test_node.cpp
https://github.com/ros2/rclcpp/blob/master/rclcpp/test/test_parameter_client.cpp

If desired, I can create a corresponding PR. Wrapping the make shared call in an EXPECT_TRUE(make_shared() != nullptr) statement would probably resolve the issue.

dirk-thomas, 2020-01-08:
A pull request would be appreciated. I would suggest to simply assign the returned value to a local variable.

Karsten1987, 2020-01-08:
FWIW, you might have to `(void)` that local variable then in order to avoid compiler warnings `-Wunused`.

brawner, 2020-01-13:
@dirk-thomas PR opened for this issue

wjwwood, 2020-01-22:
Closing as #963 was merged.

closed 2020-01-22 by wjwwood