ros2/rclcpp #969: Allow node clock to be used in THROTTLE log macros

mschickler, 2020-01-23:
## Feature request

#### Feature description
Current implementation of THROTTLE logger macros require rclcpp::Clock object to be a local (due to the way the lambda is defined in the macro body).  This precludes the use of the node clock or a clock data member.

#### Implementation considerations
The lambda could be changed slightly to cache a reference to the clock in a local variable which would allow the THROTTLE macro to be called like this:
RCLCPP_INFO_THROTTLE(get_logger(), *get_clock(), ...)
instead of:
rclcpp::Clock clock;
RCLCPP_INFO_THROTTLE(get_logger(), clock, ...)

The exact change would be in the rclcpp/rclcpp/resource/logging.hpp.em file.

--- logging.hpp.em	2020-01-23 16:23:39.796235096 -0500
+++ logging.hpp.em.new	2020-01-23 16:24:20.440236803 -0500
@@ -129,9 +129,9 @@
       typename ::rclcpp::Logger>::value, \
       "First argument to logging macros must be an rclcpp::Logger"); \
 @[ if 'throttle' in feature_combination]@ \
-    auto get_time_point = [&clock](rcutils_time_point_value_t * time_point) -> rcutils_ret_t { \
+    auto get_time_point = [&c=clock](rcutils_time_point_value_t * time_point) -> rcutils_ret_t { \
       try { \
-        *time_point = clock.now().nanoseconds(); \
+        *time_point = c.now().nanoseconds(); \
       } catch (...) { \
         RCUTILS_SAFE_FWRITE_TO_STDERR( \
         "[rclcpp|logging.hpp] RCLCPP_@(severity)@(suffix) could not get current time stamp\n"); \

ivanpauno, 2020-01-24:
Sounds good, can you open a PR with the fix?
It would be good if it includes a new test case in [test_logging.cpp](https://github.com/ros2/rclcpp/blob/fdaf96f2171e70f3e013610aa44df2d7e9c866a3/rclcpp/test/test_logging.cpp#L181), just showing that it works when a temporal is passed.

mschickler, 2020-01-24:
Will do.

mschickler, 2020-01-24:
@ivanpauno I assume I need privileges in this repo to create a branch for the PR?  Is there a process for getting that level of access?

ivanpauno, 2020-01-24:
> @ivanpauno I assume I need privileges in this repo to create a branch for the PR? Is there a process for getting that level of access?

You just need to fork, and then you will be able to open a PR perfectly.

ivanpauno, 2020-02-18:
Fixed by #970, closing!

closed 2020-02-18 by ivanpauno