ros2/rclcpp #974: SIGABRT: realloc(): invalid next size at remove_null_handles

maxlein, 2020-01-30:
I get this crash when using newest released packages in eloquent.  

I have navigation nodes which run in a screen and amcl which runs in another screen.  
If I stop amcl, the global planner is complaining that global pose is missing and waiting till it's available again, which is ok.  
But when I start amcl again, planner crashes. (We were talking about this here: #938)

https://github.com/ros2/rclcpp/blob/34dae0a8c9ca0da8310ebc9a343d4d356aedcde4/rclcpp/src/rclcpp/executor.cpp#L482

But there is also another error showing up, in case the above one is not.  

<details><summary>stacktrace</summary>

```cpp  
#0  tcache_get (tc_idx=1) at malloc.c:2943
#1  __GI___libc_malloc (bytes=32) at malloc.c:3050
#2  0x00007f1ba4e14258 in operator new(unsigned long) () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007f1b645a530d in std::_Function_base::_Base_manager<std::_Bind<void (tf2_ros::MessageFilter<sensor_msgs::msg::LaserScan_<std::allocator<void> >, tf2_ros::Buffer>::*(tf2_ros::MessageFilter<sensor_msgs::msg::LaserScan_<std::allocator<void> >, tf2_ros::Buffer>*, std::_Placeholder<1>, unsigned long))(std::shared_future<geometry_msgs::msg::TransformStamped_<std::allocator<void> > > const&, unsigned long)> >::_M_clone(std::_Any_data&, std::_Any_data const&, std::integral_constant<bool, false>) (__source=..., __dest=...) at /usr/include/c++/7/bits/std_function.h:192
#4  std::_Function_base::_Base_manager<std::_Bind<void (tf2_ros::MessageFilter<sensor_msgs::msg::LaserScan_<std::allocator<void> >, tf2_ros::Buffer>::*(tf2_ros::MessageFilter<sensor_msgs::msg::LaserScan_<std::allocator<void> >, tf2_ros::Buffer>*, std::_Placeholder<1>, unsigned long))(std::shared_future<geometry_msgs::msg::TransformStamped_<std::allocator<void> > > const&, unsigned long)> >::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) (__dest=..., __source=..., __op=<optimized out>) at /usr/include/c++/7/bits/std_function.h:227
#5  0x00007f1ba12ce114 in std::function<void (std::shared_future<geometry_msgs::msg::TransformStamped_<std::allocator<void> > > const&)>::function(std::function<void (std::shared_future<geometry_msgs::msg::TransformStamped_<std::allocator<void> > > const&)> const&) (this=0x7f1b200ac138, __x=...) at /usr/include/c++/7/bits/std_function.h:677
#6  0x00007f1ba12cb324 in tf2_ros::Buffer::<lambda(tf2::TransformableRequestHandle, const string&, const string&, tf2::TimePoint, tf2::TransformableResult)>::<lambda> (this=0x7f1b200ac120) at ./src/buffer.cpp:205
#7  std::_Function_base::_Base_manager<tf2_ros::Buffer::waitForTransform(const string&, const string&, const TimePoint&, const Duration&, tf2_ros::TransformReadyCallback)::<lambda(tf2::TransformableRequestHandle, const string&, const string&, tf2::TimePoint, tf2::TransformableResult)> >::_M_clone (__source=..., __dest=...) at /usr/include/c++/7/bits/std_function.h:192
#8  std::_Function_base::_Base_manager<tf2_ros::Buffer::waitForTransform(const string&, const string&, const TimePoint&, const Duration&, tf2_ros::TransformReadyCallback)::<lambda(tf2::TransformableRequestHandle, const string&, const string&, tf2::TimePoint, tf2::TransformableResult)> >::_M_manager(std::_Any_data &, const std::_Any_data &, std::_Manager_operation) (__dest=..., __source=..., __op=<optimized out>)
    at /usr/include/c++/7/bits/std_function.h:227
#9  0x00007f1b9f8566c5 in std::function<void (unsigned long, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::chrono::time_point<std::chrono::_V2::system_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > >, tf2::TransformableResult)>::function(std::function<void (unsigned long, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::chrono::time_point<std::chrono::_V2::system_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > >, tf2::TransformableResult)> const&) (__x=..., this=0x7f1b43ffc8d8) at /usr/include/c++/7/bits/std_function.h:677
#10 std::pair<unsigned int, std::function<void (unsigned long, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::chrono::time_point<std::chrono::_V2::system_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > >, tf2::TransformableResult)> >::pair<unsigned int&, true>(unsigned int&, std::function<void (unsigned long, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::chrono::time_point<std::chrono::_V2::system_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > >, tf2::TransformableResult)> const&) (__y=..., __x=<optimized out>, this=0x7f1b43ffc8d0) at /usr/include/c++/7/bits/stl_pair.h:312
#11 std::make_pair<unsigned int&, std::function<void (unsigned long, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::chrono::time_point<std::chrono::_V2::system_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > >, tf2::TransformableResult)> const&>(unsigned int&, std::function<void (unsigned long, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::chrono::time_point<std::chrono::_V2::system_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > >, tf2::TransformableResult)> const&) (__x=<synthetic pointer>: <optimized out>, __y=...) at /usr/include/c++/7/bits/stl_pair.h:529
#12 tf2::BufferCore::addTransformableCallback(std::function<void (unsigned long, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::chrono::time_point<std::chrono::_V2::system_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > >, tf2::TransformableResult)> const&) (
    this=this@entry=0x55e4fa591b00, cb=...) at ./src/buffer_core.cpp:1287
#13 0x00007f1ba12cbb01 in tf2_ros::Buffer::waitForTransform(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::chrono::time_point<std::chrono::_V2::system_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > > const&, std::chrono::duration<long, std::ratio<1l, 1000000000l> > const&, std::function<void (std::shared_future<geometry_msgs::msg::TransformStamped_<std::allocator<void> > > const&)>) (this=this@entry=0x55e4fa591af0, target_frame="map", source_frame="scan_link", time=..., timeout=..., callback=...)
    at ./src/buffer.cpp:238
#14 0x00007f1b645bdd94 in tf2_ros::MessageFilter<sensor_msgs::msg::LaserScan_<std::allocator<void> >, tf2_ros::Buffer>::add (this=0x55e4f9c79550, evt=...) at /opt/ros/eloquent/include/tf2_ros/message_filter.h:356
#15 0x00007f1b645bfa8c in std::function<void (message_filters::MessageEvent<sensor_msgs::msg::LaserScan_<std::allocator<void> > const> const&)>::operator()(message_filters::MessageEvent<sensor_msgs::msg::LaserScan_<std::allocator<void> > const> const&) const (__args#0=..., this=0x55e4fa730ec8) at /usr/include/c++/7/bits/std_function.h:706
#16 message_filters::CallbackHelper1T<message_filters::MessageEvent<sensor_msgs::msg::LaserScan_<std::allocator<void> > const> const&, sensor_msgs::msg::LaserScan_<std::allocator<void> > >::call (this=0x55e4fa730ec0,
    event=..., nonconst_force_copy=<optimized out>) at /opt/ros/eloquent/include/message_filters/signal1.h:74
#17 0x00007f1b645bf840 in message_filters::Signal1<sensor_msgs::msg::LaserScan_<std::allocator<void> > >::call (event=..., this=0x55e4fa6c1c48) at /opt/ros/eloquent/include/message_filters/signal1.h:117
#18 message_filters::SimpleFilter<sensor_msgs::msg::LaserScan_<std::allocator<void> > >::signalMessage (event=..., this=0x55e4fa6c1c48) at /opt/ros/eloquent/include/message_filters/simple_filter.h:133
[... cut ...]
- DDS implementation:
  - rmw_fastrtps_cpp
- Client library (if applicable):
  - rclcpp

**ROS2 Doctor:**  

  PLATFORM INFORMATION
system           : Linux
platform info    : Linux-5.0.0-29-lowlatency-x86_64-with-Ubuntu-18.04-bionic
release          : 5.0.0-29-lowlatency
processor        : x86_64

  RMW MIDDLEWARE
middleware name    : rmw_fastrtps_cpp

  ROS 2 INFORMATION
distribution name      : eloquent
distribution type      : ros2
distribution status    : active
release platforms      : {'ubuntu': ['bionic']}

ivanpauno, 2020-02-20:
@hidmic I think this one and #938 are related to the problem you're having (specially #938), can you check?

dirk-thomas, 2020-03-05:
@hidmic Friendly ping.

hidmic, 2020-03-06:
> Then I changed rmw from fastrtps to cyclone and I can't reproduce these errors...

Only because of this, it may be https://github.com/eProsima/Fast-RTPS/pull/1013 which has made it into debs AFAIK. It could also be https://github.com/ros2/rclcpp/pull/999, which hasn't because it's not ABI compatible.
 
@maxlein would you mind upgrading your Eloquent installation and building latest `rclcpp` from source along with nav2 packages?

hidmic, 2020-12-01:
@maxlein now that Eloquent is EOL, and unless you're still experiencing issues, I'm inclined to close this ticket.

maxlein, 2020-12-01:
We are already on foxy and have no issues anymore.
So it's ok for me.

closed 2020-12-01 by hidmic