ros2/rclcpp #975: Add function to Client class to reduce service request boilerplate

Shivlocin, 2020-01-30:
## Feature request

#### Feature description

Currently I have to use a lot of boilerplate code to implement a service request in a node which is being spun by an executor.

Example:
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;Create srv request
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;populate request
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;call async_send_request (returns future)
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;call std::future wait_for (prevent blocking indefinitely)
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;assess return value (ready, timeout, etc)
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;perform work accordingly 

#### Implementation considerations

I propose adding a function to the client class which would take the request and a timeout value as arguments and return pair (bool or future status & the future) as appropriate.  

Proposed function would handle:
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;call async_send_request
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;call std::future wait_for(timeout)
&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;return wait_for result & the future

jacobperron, 2020-02-10:
I think you are describing a synchronous send request method, which sounds like a reasonable API to add. [`rclpy` has a synchronous method](https://github.com/ros2/rclpy/blob/822f19406de40f9e7df0c73aa4d52837a42bcfde/rclpy/rclpy/client.py#L67) as well.

Suggestion: instead of returning the future status and future, we can return the service response directly. In the event of a timeout we can raise an exception.

Shivlocin, 2020-02-10:
A synchronous request is exactly what I'm looking for, thank you.

jacobperron, 2020-02-10:
@Shivlocin If you would like to implement a synchronous method, I'm happy to review a pull request :)

Shivlocin, 2020-02-10:
I'll happily have a go at it. More to follow.

tonynajjar, 2023-11-13:
Should this be closed as a duplicate to https://github.com/ros2/rclcpp/issues/1533? The other PR has more information/debate

clalancette, 2023-11-13:
Yes, that's a good point.  I will close this one.

closed 2023-11-13 by clalancette