ros2/rclcpp #976: lifecycle_publisher: method naming

wieset, 2020-01-31:
Probably not the most important issue but I felt like inquiring about it :)

Is there a rationale behind naming the methods `on_activate()` and `on_deactivate()` with the "on" prefix in `LifecyclePublisherInterface`? 

To me, it feels wrong to call these methods directly. They feel like state transition methods that should be called internally when a state transition is requested. Just like in `LifecycleNodeInterface` I would expect to have methods `activate()` and `deactivate()`, which then would trigger the state transition.

This is probably loosely related to #654. A universal lifecycle interface for nodes and helper classes with automatic triggering of state transitions would be great!

Karsten1987, 2020-01-31:
If I understand you correctly, you'd wish for `rclcpp_lifecycle` to call these functions implicitly so that the user doesn't have to call this on its own?
e.g. https://github.com/ros2/demos/blob/master/lifecycle/src/lifecycle_talker.cpp#L146-L152

That might not be a trivial change as we have to introduce some book keeping in the lifecycle node to know which lifecycle publishers exist. Also I am wondering if that behavior is wished for in 100% of the cases. But I guess if not, users can always fall back to a regular publisher in that case.

---
 
Alternatively, we could also rename the methods on the lifecycle publisher interface from `on_activate` to `activate` to make it clear that this functions are to be called explicitly.

wieset, 2020-01-31:
Thanks for your answer! Your first point is the one raised in #654 and would of course be nice to have, but would need some development effort. For now, I was mainly thinking of your second point, renaming the methods to `activate()` / `deactivate()`. I think that would make their intent clearer - to be called by the user to activate / deactivate the publisher.