ros2/rclcpp #978: Setting certain extra-arguments to components has no effect

fabmene, 2020-02-03:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Binaries
- Version or commit hash:
  - Eloquent
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
Minimal component example:
``` 
#include "rclcpp/rclcpp.hpp"
class ClientNode : public rclcpp::Node
{
public:
  explicit MyNode(const rclcpp::NodeOptions& options){
};
```

```

#### Expected behavior

#### Actual behavior

#### Additional information

<!-- If you are reporting a bug delete everything below
     If you are requesting a feature deleted everything above this line -->
----
## Feature request

#### Feature description
<!-- Description in a few sentences what the feature consists of and what problem it will solve -->

#### Implementation considerations
<!-- Relevant information on how the feature could be implemented and pros and cons of the different solutions -->

ivanpauno, 2020-02-13:
If I'm not mistaken, it's supposed only to work with `use_intra_process_comms`.
Maybe @hidmic can clarify

hidmic, 2020-02-14:
Correct. But I don't see any fundamental reason why these other options should not be included. @wjwwood @mjcarroll thoughts? If everyone agrees, @fabmene a PR adding support for those options would be much appreciated.

ivanpauno, 2020-02-14:
I don't think that allowing configuration through the command line of `allow_undeclared_parameters` is a good idea.
Same about `use_global_arguments`.

hidmic, 2020-02-14:
What command line @ivanpauno? Extra arguments only appear in component loading service calls. Having said, I do have a hard time thinking of a use case for dynamically changing `allow_undeclared_parameters` on load.

ivanpauno, 2020-02-14:
> What command line @ivanpauno? Extra arguments only appear in component loading service calls. Having said, I do have a hard time thinking of a use case for dynamically changing `allow_undeclared_parameters` on load.

Ups, sorry. But the same comment applies to launch files.
If the original program wasn't planned to be used with `allow_undeclared_parameters=true` or `use_global_arguments=true`, passing them from a launch file can just generate problems. Specially the first one, the second one might have an use-case I'm not seeing.

fabmene, 2020-02-17:
Thank you for providing inside on this matter. To clarify, I do not have a use case for the aforementioned arguments either, but was merely experimenting with components and their limitations.

ivanpauno, 2020-02-17:
@hidmic Does any tutorial mentions `extra_arguments` and how it can be used. I think that mentioning it in a tutorial would clarify the situation.

wjwwood, 2020-03-03:
Thinking about this more, it's like we have two categories of settings for nodes:

- settings that do not significantly affect behavior of the node APIs
  - by "do not significantly affect behavior" I mean can be changed without violating any important assumptions the node developer is making, e.g. doesn't change blocking behavior or hard coded names of resources, etc...
  - for me this includes `use_intra_process`
- settings that affect behavior of the node APIs and cannot be a parameter
  - for me this includes `use_global_arguments` and `allow_undeclared_parameters`
- settings that affect behavior of the node APIs and can be a parameter
  - for me this includes things like "publisher_X_should_use_reliable" or something like that

For the second case, options that affect the behavior of the node and therefore the original developer should be aware of them, I'd normally say "use a parameter if you want to make it configurable", but in these cases like `use_global_arguments` and `allow_undeclared_parameters` happen before parameters are available or affect parameters.

Maybe the components interface in rclcpp should only accept a subset of `rclcpp::NodeOptions` (see: https://github.com/ros2/rclcpp/blob/f30329fbec79cb699f41a37cc27c0c475de3edc1/rclcpp_components/include/rclcpp_components/node_factory.hpp#L40-L42), limiting it to the first class of node options in my list above.

This limited set of node options would dictate what can be part of the "extra options" for each client library:

https://github.com/ros2/rcl_interfaces/blob/93cedce2dacd25fa4f2969d022ccbe3f2903e3fe/composition_interfaces/srv/LoadNode.srv#L19

ivanpauno, 2020-03-04:
> Maybe the components interface in rclcpp should only accept a subset of rclcpp::NodeOptions (see: ...), limiting it to the first class of node options in my list above.

Yes, I agree with that. That method could still take `rclcpp::NodeOptions` as a parameter, but arguments to modify its attributes should be exposed only for a subset of them.

Other arguments that I believe enter in the first category:
- [enable_rosout](https://github.com/ros2/rclcpp/blob/f30329fbec79cb699f41a37cc27c0c475de3edc1/rclcpp/include/rclcpp/node_options.hpp#L158).
- [start_parameter_services](https://github.com/ros2/rclcpp/blob/f30329fbec79cb699f41a37cc27c0c475de3edc1/rclcpp/include/rclcpp/node_options.hpp#L207).
- [start_parameter_event_publisher](https://github.com/ros2/rclcpp/blob/f30329fbec79cb699f41a37cc27c0c475de3edc1/rclcpp/include/rclcpp/node_options.hpp#L223).

The three of them can usually be configured by the user without the need to change how the node was implemented (in some corner cases, maybe not, but I think they should still be exposed).

We could add them to be configurable when loading a component, like here: https://github.com/ros2/rclcpp/blob/f30329fbec79cb699f41a37cc27c0c475de3edc1/rclcpp_components/src/component_manager.cpp#L170.

We could also add some ros arguments to make them configurable in manually composed executables.

hidmic, 2020-03-04:
> Maybe the components interface in rclcpp should only accept a subset of `rclcpp::NodeOptions`

Hmm but then that'd apply to dynamic loading only -- statically constructing a node would still allow full access to all options. I think that if we establish that some options are not to be modified by client code, then maybe those should not be accessible in `rclcpp::NodeOptions` to begin with. And of course, when writing a component one may still enable client code to change these private options in a variety of non-standard, application-specific ways. 

---

I'm inclined to think that putting restrictions doesn't do us any good, even more so if restrictions are partial. All options that @ivanpauno mentioned make sense to be exposed, same for `use_intra_process`, same for `use_global_arguments` and even `allow_undeclared_parameters` if e.g. a user ever wants to extend a composable node through inheritance. IMHO having a standard way to pass these common node attributes will always be better than each component rolling out their own.

wjwwood, 2020-03-04:
I wasn't suggesting that we change `NodeOptions` or change nodes to take something other than `NodeOptions` in their constructor, instead I was saying that we should have a new options struct that is a subset (most likely) of `NodeOptions` that is passed to the "node factory":

https://github.com/ros2/rclcpp/blob/f30329fbec79cb699f41a37cc27c0c475de3edc1/rclcpp_components/include/rclcpp_components/node_factory.hpp#L40-L42

Something like this:

```c++
class ComposableNodeOptions;
class NodeOptions
{
public:
  explicit NodeOptions(const ComposableNodeOptions &);
};

class NodeFactory
{
public:
  virtual
  NodeInstanceWrapper
  create_node_instance(const ComposableNodeOptions & options) = 0;
};

class Node
{
public:
  Node(..., const NodeOptions &, ...);
};

// ...

class MyNode : public Node
{
public:
  explicit MyNode(const NodeOptions & no) : Node("my_node", ..., no, ...) {...};
};

class MyNodeFactory : public NodeFactory
{
public:
  virtual
  NodeInstanceWrapper
  create_node_instance(const ComposableNodeOptions & cno)
  {
    return NodeInstanceWrapper(std::make_shared<MyNode>(NodeOptions(cno)));
  }
};
```

In this way your custom node (`MyNode` above) can still be fully configured with `NodeOptions`, but when being loaded dynamically using this package (`rclcpp_components`) it would be restricted to only the `ComposableNodeOptions`.

hidmic, 2020-03-06:
> I wasn't suggesting that we change NodeOptions or change nodes to take something other than NodeOptions in their constructor, instead I was saying that we should have a new options struct that is a subset (most likely) of NodeOptions that is passed to the "node factory":

Oh yeah, I got that. I still stand by what I said above. We're restricting options because we think they are not meant to be changed by a user, but we only do so for the dynamic composition case. Code composing `MyNode` manually could still affect all those options you'd rather not. 

So instead of introducing an asymmetry, I'd be fine with proper documentation. People will always find ways of shooting themselves in the foot if they are willing to. You can make a mess in `rclpy` too if you start playing with object attributes you shouldn't be playing with.

wjwwood, 2020-03-06:
Ok, well I don't feel strongly about it. Documentation is fine with me too.

hidmic, 2020-10-21:
Circling back! It seems we were discussing whether to restrict options for composable nodes only OR just allow everything and document potential side-effects. I was and still am inclined towards the latter. Patch is small if you guys @wjwwood @ivanpauno agree.

wjwwood, 2020-10-21:
Documentation is ok for me.

gezp, 2021-09-14:
Hi, what's the status of this issue right now?

i'm trying to implement dynamically composed bringup for Nav2, and Nav2 has a large param file for multiple nodes (also contain some internal node's parameters), when `use_global_arguments` is false (https://github.com/ros2/rclcpp/blob/master/rclcpp_components/src/component_manager.cpp#L151),  the internal nodes (e.g. `local_costmap`, `global_costmap`) couldn't load parameters from container's param file. i couldn't solve this problem by other ways.

@wjwwood , in your opinion, `use_global_arguments`  cannot be a parameter, how about adding a parameter for container, so that users could set `use_global_arguments` of all component nodes true?  or do you have another idea to load parameters for  internal nodes of component node? could you give me some suggestions ? thank you!

SteveMacenski, 2021-09-15:
> I don't think that allowing configuration through the command line of allow_undeclared_parameters is a good idea. Same about use_global_arguments.

> Specially the first one [allow_undeclared_parameters], the second one [use_global_arguments] might have an use-case I'm not seeing.

I agree (somewhat) with `allow_undeclared_parameters`. My personal thoughts is that parameter should be entirely removed since users have gotten a chance to get used to the new params workflows in ROS 2 -- no need to keep that around. 

 The `use_global_arguments` does have a legitimate use case. For composition nodes that have internally another node object, we need to be able to pass in the parameters to that secondary internal node. An example is Nav2, where each server has a single node, but the servers that contain costmaps have another independent node for the costmaps. This is because a costmap is a sufficiently complex unique entity in its own right that needs control over its flow of sensor data and parameters separately from the server that uses it. However, it is not appropriate to compose the costmap as a separate node since it needs to be statically tied to the task servers for ultra-fast access by the planning and control algorithms regardless of composition strategies employed. 

I'm way less interested / concerned by dynamically being able to flip it on and off, but it needs to be able to be set on launch of the container so that it can be used on the odd situation it is required. It seems completely arbitrary that all node types can use it, but simply the container of components restricts it. I can definitely appreciate the point that it is probably not a good idea in **general** to do this, but the point of ROS 2 is to meet the needs of commercial users, and there are many non-general times where things that might generally be considered a bad idea need to be done for specific situations. I believe this is one of them.

 It would be a shame if Nav2 had to create its own, separate implementation of a component container simply to be able to get parameters to our nodes. This will almost certainly come back up again in the future.

fujitatomoya, 2021-09-17:
> Circling back! It seems we were discussing whether to restrict options for composable nodes only OR just allow everything and document potential side-effects. I was and still am inclined towards the latter.

I am with @hidmic with this.

> Code composing MyNode manually could still affect all those options you'd rather not.

if the custom node implementation wasn't planned to be used with some arguments, probably node implementation needs to be responsible for that?

gezp, 2021-09-18:
>  the internal nodes (e.g. local_costmap, global_costmap) couldn't load parameters from container's param file. 

sorry, the internal nodes (e.g. `local_costmap`, `global_costmap`) create their own `rclcpp::NodeOptions` (with default `use_global_arguments(true)`) , so they could load parameters from container node's argument `--params-file`. only component Node can't load parameters from container node.

SteveMacenski, 2021-09-27:
Still feels like a niche we should have resolved

fujitatomoya, 2024-12-13:
closing in favor of https://github.com/ros2/rclcpp/pull/2685

closed 2024-12-13 by fujitatomoya