ros2/rclcpp #983: Cross-compile for mips/musl: XSI-compliant strerror_r

mauropasse, 2020-02-05:
## Bug report

**Required Info:**

- Operating System:
  - 
- Installation type:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

The issue is that in rclcpp/src/rclcpp/[signal_handler.cpp](https://github.com/ros2/rclcpp/blob/master/rclcpp/src/rclcpp/signal_handler.cpp#L168) there's an #if directive on __safe_strerror:
```
RCLCPP_LOCAL                                                                    
void                                                                            
__safe_strerror(int errnum, char * buffer, size_t buffer_length)                
{                                                                               
#if defined(_WIN32)                                                             
  ...                              
#elif defined(_GNU_SOURCE) && (!defined(ANDROID) || __ANDROID_API__ >= 23)
  /* GNU-specific */                                                            
  char * msg = strerror_r(errnum, buffer, buffer_length);      // THE ONE IT'S USED                  
   ...                                                                           
#else                                                                           
  /* XSI-compliant */                                                           
  int error_status = strerror_r(errnum, buffer, buffer_length);    // THE ONE I NEED            
  ...                                                                          
#endif                                                                          
} 
```
My system should use the XSI-compliant version of error_status, which returns an int.
But when compiling the GNU-specific is used because _GNU_SOURCE gets defined in rcutils, ament_lint, fast-RTPS, etc.

My fix is adding this to the #if directive, as I'm using musl:
```
 #elif defined(_GNU_SOURCE) && (!defined(ANDROID) || __ANDROID_API__ >= 23) && (defined(__UCLIBC__) || defined(__GLIBC__))
```
Do you think you can add that guard?
Thanks

ivanpauno, 2020-02-13:
@mauropasse Doesn't musl provide a macro, something like `__GLIBC__`?
It would be good to add another `ifdef` check, with that exception, instead of checking for `__UCLIBC__` and `__GLIBC__`.

Can you open a PR with a fix?

mauropasse, 2020-02-13:
Hi @ivanpauno , theres no __MUSL__ . [This link](https://wiki.musl-libc.org/faq.html) answers: Why is there no __MUSL__ macro?
Checking not __UCLIBC__ and not __GLIBC__ is what I've seen for this cases

ivanpauno, 2020-02-13:
Sounds reasonable

clalancette, 2024-01-19:
We actually removed this `strerror_r` in #989, in favor of using rcutils instead.  So I'll close this out.

closed 2024-01-19 by clalancette