ros2/rclcpp #985: expand_topic_or_service_name should use get_effective_namespace()

fmrico, 2020-02-08:
## Feature request

#### Feature description

`sub_node` concept is not being correctly used in some packages that use the function `expand_topic_or_service_name()`. They use with `get_namespace()` to create publishers, instead of `get_effective_namespace()`. Probably many people do not know the existence of this method or the concept of subnode. For example, [Here](https://github.com/ros-perception/image_common/blob/42cd97418b74ca1683dcd40c7f48a711302ac3f1/image_transport/src/camera_publisher.cpp#L89). If this node is a sub_node (created with `create_sub_node()`, then the effect of the subnode does not take effect. For:

```
    color_subnode_ = create_sub_node("rgb");
    image_transport::ImageTransport color_it(color_subnode_);
    pub_color_ = color_it.advertiseCamera("image", 1);
```

The effect is:

```
$ ros2 topic list
/camera_info
/image
/image/compressed
/image/compressedDepth
/image/theora
```

Instead of adding "/rgb" in front of each topic name. 

Maybe `expand_topic_or_service_name()` should receive the rclcpp::Node, and internally, it could call `get_effective_namespace()`.

I can work on it, but I would like to know if you think it's the right solution.

#### Implementation considerations

wjwwood, 2020-02-24:
> Here. If this node is a sub_node (created with create_sub_node(), then the effect of the subnode does not take effect.

Yeah, that looks like it is just doing the wrong thing in that case.

Unfortunately, I see the problem that these things (like image transport) take the `NodeBaseInterface` and `NodeTopicsInterface` rather than a `rclcpp::Node`, and that's a problem because the subnode mechanism exists completely within a `rclcpp::Node`. We need some way to capture this and provide it to interfaces like this when needed... I do not, for example, think the right solution is to make the image transport only take a `rclcpp::Node` object.

Quchaosheng, 2026-10-21:
The current implementation makes the sub-node namespace a property of the `rclcpp::Node` wrapper, not of the interfaces passed to consumers. The sub-node copy constructor shares `node_base_`, `node_topics_`, and the other interface objects with its parent, and computes `sub_namespace_`/`effective_namespace_` only on the `Node` instance (`rclcpp/src/rclcpp/node.cpp:259-275`). `Node` itself prefixes relative names before delegating (`rclcpp/include/rclcpp/node_impl.hpp:55-99`), while `NodeTopics::create_publisher` passes the raw name and shared base to the factory (`rclcpp/src/rclcpp/node_interfaces/node_topics.cpp:43-50`). Consequently, a consumer holding only `NodeBaseInterface` or `NodeTopicsInterface` has no way to recover the effective namespace.

This is the interface-layer problem described in [wjwwood's comment](https://github.com/ros2/rclcpp/issues/985#issuecomment-590574292), not merely a bad `expand_topic_or_service_name` call in `image_transport`. Changing that utility to take `rclcpp::Node` would fix one consumer and make the layering worse. It would still leave lifecycle nodes, custom node-likes, and the existing interface-based APIs inconsistent.

### Proposed interface

Add a small, non-abusive name-resolution facet:

```cpp
class NodeNameContextInterface
{
public:
  virtual ~NodeNameContextInterface() = default;

  virtual const std::string & get_effective_namespace() const = 0;
  virtual std::string resolve_topic_name(
    const std::string & name, bool only_expand = false) const = 0;
  virtual std::string resolve_service_name(
    const std::string & name, bool only_expand = false) const = 0;
};
```

`Node` should implement this with the same sub-namespace prefixing and remapping rules it already uses for `create_publisher`, `create_subscription`, and `create_service`. The context should be per sub-node; the physical `NodeBaseInterface` should continue to represent the one underlying `rcl_node_t` and should not change the meaning of `get_namespace()`. Conflating the physical and effective namespaces would break graph identity, fully-qualified node names, and code that intentionally asks for the physical namespace.

The context should be captured by the interface aggregate, not by requiring every consumer to know that a sub-node exists. `NodeInterfaces` currently just takes a tuple of interface `shared_ptr`s (`rclcpp/include/rclcpp/node_interfaces/node_interfaces.hpp:43-65`, and initialization at `detail/node_interfaces_helpers.hpp:135-150`). Add a `NodeNameContextInterface::SharedPtr` member to `NodeInterfaces`, populated from the originating node-like when available. The existing explicit constructor from loose interface pointers cannot infer the sub-namespace; it should accept an explicit context argument or use a documented root-context fallback.

For compatibility with consumers that still take individual interfaces:

- Keep the existing raw `NodeBaseInterface`/`NodeTopicsInterface` overloads, but document them as root-context APIs when no context is supplied.
- Add context-aware overloads for `create_publisher`, `create_subscription`, generic publishers/subscriptions, services/clients, and the action create functions. Prefer a single `NodeInterfaces<...>` or `NodeNameContextInterface` parameter over adding another unrelated raw pointer.
- Have `Node::get_node_*_interface()` for a sub-node return a small sub-node-aware view or carry the context separately. A `NodeTopics` decorator can prefix/resolve names before forwarding to the shared root interface without changing the root interface's identity. A decorator is preferable to adding another virtual to `NodeBaseInterface`, which would be an ABI break and still cannot distinguish two sub-node wrappers that share one `NodeBase`.
- Add `NodeNameContextInterface` to the aggregate requirements in `image_transport`'s rolling `RequiredInterfaces`, rather than reverting `image_transport` to `rclcpp::Node`.

This gives three compatibility levels:

1. Node-like users automatically get sub-node behavior through `NodeInterfaces`.
2. Existing interface-only users continue to compile, with documented root behavior unless they pass a context.
3. Rolling-only APIs can migrate to the context-aware overloads and eventually deprecate the ambiguous interface-only overloads.

### Affected call sites

The main paths to update are:

- `rclcpp/include/rclcpp/create_publisher.hpp:55-71`: resolve QoS parameter names and create the publisher through the context.
- `rclcpp/include/rclcpp/create_subscription.hpp:64-81`: the same path for subscriptions and QoS overrides.
- `rclcpp/src/rclcpp/node_interfaces/node_topics.cpp:43-50` and `73-90`: create-view naming; `resolve_topic_name` at `141-145`.
- `rclcpp/src/rclcpp/node_interfaces/node_services.cpp:78-82`: service resolution.
- `rclcpp/src/rclcpp/node_interfaces/node_graph.cpp:456-460`, `478-482`, `500-504`, `522-526`, `670-674`, and `785-789`: graph queries currently expand from the physical RCL namespace.
- `rclcpp_action/src/graph.cpp:77-81` and `122-126`, plus the raw-interface overloads in `rclcpp_action/include/rclcpp_action/create_server.hpp:58-68` and the corresponding client functions.
- `rclcpp/include/rclcpp/node_impl.hpp:73`, `94`, `145`, `161`, `179`, `196`, and `214`: keep these as the `Node` convenience methods, but delegate to the same context rather than duplicating prefix rules.
[... cut ...]
5. Keep the old overloads for one compatibility cycle, then deprecate only the ones whose behavior cannot be made correct from their arguments.

Regression tests should cover relative names under nested sub-nodes, absolute names, `~` private substitutions, remaps applied after the sub-namespace prefix, a consumer receiving `NodeInterfaces` from a sub-node, and a consumer using the explicit root-only overload. The important negative test is that a loose `NodeBaseInterface` cannot magically infer a calling sub-node's namespace; the API should require an explicit context instead of silently doing the wrong thing.