ros2/rclcpp #997: LifecycleNode: Take shutdown transition when exiting

wieset, 2020-02-18:
## Feature request

#### Feature description

When a lifecycle node is shut down using CTRL-C, it would be helpful to automatically call the `shutdown` transition. I would actually consider this "expected behavior".

Currently, the user needs to take care of calling the `shutdown` transition before exiting, for example through installing their own signal handler. Otherwise, the node is destructed from whichever state it's currently in, potentially leaving the system in an undefined or unsafe state.

#### Implementation considerations

If the node is currently in another transitioning state, the shutdown request needs to be deferred. If the node is already finalized, the request should be ignored.

claireyywang, 2020-02-27:
@wieset Would you like to propose a PR? We'd be happy to review it.

Rezenders, 2024-03-09:
Any updates on this?
I tired the solution on: https://answers.ros.org/question/343744/ros2-lifecycle-node-how-to-shut-down-cleanly/
But it doesn't work

fujitatomoya, 2024-03-14:
@Rezenders it would be appreciated if you can try https://github.com/ros2/rclcpp/pull/2450

fujitatomoya, 2024-04-06:
it has been 4 years, but finally we can close this. https://github.com/ros2/rclcpp/pull/2450 is merged.

closed 2024-04-06 by fujitatomoya

fujitatomoya, 2024-06-12:
Just a follow-up comment.

at this moment, that is a responsibility for user application to manage the state control including shutdown, see more information for https://github.com/ros2/rclcpp/issues/2520#issuecomment-2159092031.