ros2/rclcpp #1008: test_multi_threaded_executor timer_over_take is flaky due to timer jitter

brawner, 2020-03-01:
This seems to be a reoccurring flaky test, as a test introduced in #383 has been relaxed a couple of times (#501, and #907), but the test may still fail on different platforms. The measured period between timer callbacks is supposed to be 0.1s, but the linux test result below shows that the period could be less than half of that. While the test is not meant to measure timer jitter, the results suggest that the timer jitter cannot be reliably bounded.

See:
https://ci.ros2.org/view/nightly/job/nightly_linux_repeated/1780/testReport/(root)/projectroot/test_multi_threaded_executor/
https://ci.ros2.org/view/nightly/job/nightly_osx_repeated/1866/testReport/(root)/projectroot/test_multi_threaded_executor/
https://ci.ros2.org/view/nightly/job/nightly_windows-container_repeated/6/testReport/(root)/projectroot/test_multi_threaded_executor/

chapulina, 2021-02-22:
I think this may be a duplicate of #751.

This test is still flaky one year later. Here's the failure 

```
[ RUN      ] TestMultiThreadedExecutor.timer_over_take
/Users/osrf/jenkins-agent/workspace/nightly_osx_debug/ws/src/ros2/rclcpp/rclcpp/test/rclcpp/executors/test_multi_threaded_executor.cpp:91: Failure
Expected: (diff) > (PERIOD - TOLERANCE), actual: 0.732793 vs 0.75
[  FAILED  ] TestMultiThreadedExecutor.timer_over_take (4201 ms)
```

On #751, there was a suggestion to "remove the upper bound condition on the test", which I think may refer exactly to the offending expectation, and @clalancette agreed at the time (2 years ago!) that it was a good idea. 

---

https://github.com/osrf/buildfarmer/issues/161

clalancette, 2021-02-22:
Actually, I don't think we have the upper bound anymore.  I'm going to close out #751.

When I've looked into this in the past, it does look like a real bug in the multithreaded implementation.  That is, we are setting up a timer to expire once a second.  We should reasonably expect that timer to execute *no more than* once a second.  However, these failures show that sometimes callbacks happen much earlier than that, even down to 0.5 seconds.  So something is obviously wrong there, but it is a hard one to debug.

ivanpauno, 2021-04-12:
There was some investigation done [here](https://github.com/ros2/rclcpp/pull/1624#issuecomment-816069407).
https://github.com/ros2/rclcpp/pull/1628 will (hopefully) fix the issue.

PR #1628 (2021-04-13): Use a different implementation of mutex two priorities
Fixes #1008 

The naive implementation added in https://github.com/ros2/rclcpp/pull/1516 doesn't work as well as I expected.
Even though only one "low priority thread" can be waiting in the "data mutex", it's still pretty likely to find the following behavior:

- low priority thread releases data mutex and then the "barrier" mutex
- another low priority thread releases takes the barrier mutex and then also takes the "data" mutex, even when there was a "high priority" thread waiting.

i.e.: the "high priority" take isn't always as "high pri [... cut]
closed 2021-04-13 by ivanpauno