ros2/rclcpp #1009: declare_parameter<double>() gives ParameterTypeException, expected [double] got integer

ddengster, 2020-03-02:
- Operating System:
  - Eloquent
- Installation type:
  - Binaries
- Version or commit hash:
  - The same version as when you do apt-get eloquent packages

#### Steps to reproduce issue
Compile and run:
``` 
rclcpp::Node* node = new rclcpp::Node("asd");
//the following loc throws an error
node->declare_parameter<double>("state_publish_rate", 50.0);

//another similar error
node->get_parameter("state_publish_rate", 50.0).get_value();

```

#### Expected behavior

Double types should work.

#### Actual behavior
For both lines you get a variant of the following
```
terminate called after throwing an instance of 'rclcpp::ParameterTypeException'
what():  expected [double] got [integer]
```

#### Additional information
There is a workaround for the declare_parameter line:

```
lifecycle_node_->declare_parameter("state_publish_rate", rclcpp::ParameterValue(50.0));
```

but I cant find the other except for manually casting to double.

Feature #979 might be interested in this.

clalancette, 2020-03-02:
```
rclcpp::Node* node = new rclcpp::Node("asd");
//the following loc throws an error
node->declare_parameter<double>("state_publish_rate", 50.0);
```

I have this idiom in dozens of Dashing/Eloquent packages, so I'm fairly confident it works (one example is https://github.com/ros-drivers/phidgets_drivers/blob/e87cc88829160f074ac378e7f85be4df75282be1/phidgets_gyroscope/src/gyroscope_ros_i.cpp#L65; there are many others in that repository).  My guess is that the YAML file you are using to override the parameters is specifying that parameter as an integer, not a double, and hence the error.  There is an [open issue](https://github.com/ros2/rclcpp/issues/979) about relaxing this requirement, but in the meantime you should be able to change the YAML file override to add a `.0` to the end.

ddengster, 2020-03-03:
Yup my yaml was incorrect, thanks for pointing it out.

closed 2020-03-03 by ddengster