ros2/rclcpp #1010: [rclcpp_components] Make component manager public

Karsten1987, 2020-03-02:
TL;DR: We've discussed about having the ros2_control's controller manager inherit from the `rclcpp_components` component manager as we would like to have similar functionality.

More info here: https://github.com/ros-controls/roadmap/pull/7

In order to inherit though, the component manager header has to be modified to be a public header with `virtual` functions. Is there a reason why this is currently not the case?

fyi @mjcarroll

mjcarroll, 2020-03-02:
No, there isn't a strong reason other than it was not necessary to support a public API for it at the time.

Since no one is currently using this as an interface, I'm not quite sure what the ABI implications are, but it would seem reasonable to do for Foxy?

Karsten1987, 2020-03-02:
yeah, Foxy would be great so we can work with Foxy for the ros control release.

mjcarroll, 2020-03-02:
Are there additions to the API that will also be required in order to support the concept of a "strongly-typed component"? Or have you not quite made it far enough to determine yet?

Karsten1987, 2020-03-02:
I haven't really thought that through in all detail, but essentially we'd like to restrict the type of nodes the `ControllerManager` can load. 

From what I understand, the `NodeFactory` is already templated in a way that can be used to restrict to load components different from `rclcpp::Node`. 
In terms of API change, I believe it would be great if we could e.g. things like this https://github.com/ros2/rclcpp/blob/master/rclcpp_components/src/component_manager.cpp#L65-L66 turn into a function argument with default value. That way, we don't necessarily have to override that particular function.

mjcarroll, 2020-03-19:
Ping @Karsten1987 , have you determined if there is anything else you need API-wise?  Otherwise I will proceed with just exposing it as part of the public API and we can go from there.

Karsten1987, 2020-03-19:
I think we should be okay with the changes mentioned in my comment above. The essential requirements we would have to make it work for ros-control is to be able to subclass from the component manager and modify its logic to restrict loading class instances other than `rclcpp::Node`.

PR #1065 (2020-04-16): Make Component Manager public
closes #1010 

The first commit https://github.com/ros2/rclcpp/pull/1065/commits/d81d494ba6af5a8f8dc8c41a6311619eebb5ffa1 basically just turns the `component manager` into public API.

The second commit https://github.com/ros2/rclcpp/pull/1065/commits/156ff16a3722ce671bb655f341f5c32347e41a81 is enhancing the CMake macros for registering nodes with a new option `RESOURCE_INDEX` to specify under which index the file should be written.

With respect to the upcoming work in ros-control (https://github.com/ros-controls/roadmap/pull/7) that enables to not duplicate the  [... cut]
closed 2020-04-16 by Karsten1987