ros2/rclcpp #1039: [rclcpp] Progress Toward Quality Level 1

brawner, 2020-03-30:
This issue tracks the progression of `rclcpp` to Quality Level 1 and a 1.0 version level. It follows the outline described in [REP 2004](https://github.com/ros-infrastructure/rep/blob/c2ba03bac23b8f313e47820833fad000dbbe058b/rep-2004.rst). 

* [x] Version Policy
    - Follows ROS Core Quality Declaration
* [ ] Version >= 1.0.0
* [x] Change Control Process
    - Follows ROS Core
* [ ] PRs automatically tested on Jenkins

Documentation
* [ ] Per-feature documentation. 
    - Many features described in different places, but I couldn't find a single unified location.
* [ ] Per-item documentation in public API. 
    - Doxygen documentation is generated
    - Many API components are undocumented
* [x] Declared set of licenses
* [x] Copyright statement in each source file
* [ ] Quality Declaration document

Testing:
* [x] System tests, which cover "features"
* [x] Tests covering public API
* [ ] Code coverage (95%)
    - 78%, see https://ci.ros2.org/job/ci_linux_coverage/37/cobertura/
* [ ] Performance tests
* [x] Performance test policy: ROS Core
* [x] Package code style: ROS Core
* [x] Uses ament_lint_common

Dependencies:
* [ ] Runtime "ROS" dependencies are level 1
    - builtin_interfaces
    - rcl
    - rcl_interfaces
    - rcl_yaml_param_parser
    - rcpputils
    - rcutils
    - rmw
    - rosgraph_msgs
    - rosidl_generator_cpp
    - rosidl_typesupport_c
    - rosidl_typesupport_cpp
    - tracetools

Platform Support
* [x] Supports all tier 1 platforms

jacobperron, 2020-05-12:
We now have a 1.0.0 release: https://github.com/ros/rosdistro/pull/24844

ahcorde, 2020-06-25:
Bump QD to level 3 and fixed some links https://github.com/ros2/rclcpp/pull/1158

chapulina, 2020-11-26:
Level 1 achieved :100: https://github.com/ros2/rclcpp/pull/1477

closed 2020-11-26 by chapulina