ros2/rclcpp #1046: ROS2 Listener from demo_nodes_cpp does not work

rda112, 2020-04-01:
Operating System:
Windows 10
Installation type:
 ROS 2 Eloquent Elusor binaries
Client library (if applicable):
rclcpp

done all the steps as per the documentation.

ros2 run demo_nodes_cpp talker -> works fine

opened new terminal -> sourced the terminal
ros2 run demo_nodes_cpp listener

listener doesnot receive any data.

Could anyone help me with this?

#### Steps to reproduce issue

```

```

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

clalancette, 2020-04-01:
Please ask questions like this on https://answers.ros.org.  That way it is all in one place and is easily searchable in the future.

closed 2020-04-01 by clalancette

mattharmer, 2021-12-30:
Had same issue, but only on Windows running Ubuntu VirtualBox. All other install were fine. Check multicast is running, 2 Terminals (https://docs.ros.org/en/rolling/How-To-Guides/Installation-Troubleshooting.html)
ros2 multicast receive
ros2 multicast send
If this works all should be good I think..