ros2/rclcpp #1058: SIGINT not causing node to shut down - ros2 foxy dev build

cmorganBE, 2020-04-10:
## Bug report

**Required Info:**

- Operating System:
Ubuntu 20.04

- Installation type:
ros2 foxy source install from two days ago (I can provide repo revs if desired or a .repos file)

- Version or commit hash:
[my_ros2.repos.txt](https://github.com/ros2/rclcpp/files/4461892/my_ros2.repos.txt)

- DDS implementation:
Default, looks like fastrtps

- Client library (if applicable):
rclcpp

#### Steps to reproduce issue

Note that this code isn't actually publishing anything although it does create publishers. This is what I'm testing but the plan was to have a service to update values and this would expose them as a simulator of our actual system. It isn't finished or complete but imo its correct enough that it should shut down with SIGINT (ctrl-c) right?

```
#include <chrono>
#include <string>
#include <memory>
#include <filesystem>

#include "sensor_msgs/msg/temperature.hpp"
#include "rclcpp/rclcpp.hpp"
#include "rclcpp/clock.hpp"
#include "sensor_msgs/msg/fluid_pressure.hpp"
#include "mcu_sim_interfaces/srv/set_simulated_values.hpp"

using namespace std::chrono_literals;

class McuSimPublisher : public rclcpp::Node
{
public:
  McuSimPublisher()
  : Node("mcu")
  {
    clock = std::make_shared<rclcpp::Clock>(RCL_ROS_TIME);
    const auto qos_history_depth = 10;
    publisher_ambient_temperature_ =
      this->create_publisher<sensor_msgs::msg::Temperature>(
      std::string(
        get_name()) + "/sensor/ambient_temperature", qos_history_depth);
    publisher_internal_temperature_ =
      this->create_publisher<sensor_msgs::msg::Temperature>(
      std::string(
        get_name()) + "/sensor/internal_temperature", qos_history_depth);
    publisher_pressure_ =
      this->create_publisher<sensor_msgs::msg::FluidPressure>(
      std::string(
        get_name()) + "/sensor/pressure", qos_history_depth);
  }

  void set_internal_c(float c)
  {
    internal_temperature_.header.stamp = clock->now();
    internal_temperature_.temperature = c;
  }

  void set_ambient_c(float c)
  {
    ambient_temperature_.header.stamp = clock->now();
    ambient_temperature_.temperature = c;
  }

  void set_pressure_pascal(float p)
  {
    pressure_.header.stamp = clock->now();
    pressure_.fluid_pressure = p;
  }

  void publish_values()
  {
    publisher_ambient_temperature_->publish(ambient_temperature_);
    publisher_internal_temperature_->publish(internal_temperature_);
    publisher_pressure_->publish(pressure_);
  }

private:
  rclcpp::Clock::SharedPtr clock;

  sensor_msgs::msg::Temperature internal_temperature_;
  sensor_msgs::msg::Temperature ambient_temperature_;
  sensor_msgs::msg::FluidPressure pressure_;

  rclcpp::TimerBase::SharedPtr timer_;
  rclcpp::Publisher<sensor_msgs::msg::Temperature>::SharedPtr publisher_ambient_temperature_;
  rclcpp::Publisher<sensor_msgs::msg::Temperature>::SharedPtr publisher_internal_temperature_;
  rclcpp::Publisher<sensor_msgs::msg::FluidPressure>::SharedPtr publisher_pressure_;
};

std::shared_ptr<McuSimPublisher> publisher;

int main(int argc, char * argv[])
{
  rclcpp::init(argc, argv);

  publisher = std::make_shared<McuSimPublisher>();

  auto appname = std::filesystem::path(argv[0]).filename().string();

  std::cout << appname << " starting" << std::endl;
  rclcpp::spin(publisher);

  rclcpp::shutdown();
  std::cout << appname << " shutting down" << std::endl;
  return 0;
}
```

#### Expected behavior
ctrl-c should shut things down, and it does almost all of the time but not always

#### Actual behavior
Sometimes ctrl-c results in the message being displayed but the process doesn't shut down:
```
$ ros2 run mcu_sim mcu_sim_cpp
mcu_sim_cpp starting
^C[INFO] [1586527339.266956564] [rclcpp]: signal_handler(signal_value=2)
```

#### Additional information

gdb backtrace:

```
(gdb) bt
#0  futex_wait_cancelable (private=<optimized out>, expected=0, futex_word=0x5559a6008d38) at ../sysdeps/nptl/futex-internal.h:183
#1  __pthread_cond_wait_common (abstime=0x0, clockid=0, mutex=0x5559a6008d40, cond=0x5559a6008d10) at pthread_cond_wait.c:508
#2  __pthread_cond_wait (cond=0x5559a6008d10, mutex=0x5559a6008d40) at pthread_cond_wait.c:638
#3  0x00007fce91a9ee20 in std::condition_variable::wait(std::unique_lock<std::mutex>&) () from /lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007fce9117629f in void std::condition_variable::wait<rmw_fastrtps_shared_cpp::__rmw_wait(rmw_subscriptions_t*, rmw_guard_conditions_t*, rmw_services_t*, rmw_clients_t*, rmw_events_t*, rmw_wait_set_t*, rmw_time_t const*)::{lambda()#1}>(std::unique_lock<std::mutex>&, rmw_fastrtps_shared_cpp::__rmw_wait(rmw_subscriptions_t*, rmw_guard_conditions_t*, rmw_services_t*, rmw_clients_t*, rmw_events_t*, rmw_wait_set_t*, rmw_time_t const*)::{lambda()#1}) ()
   from /home/cmorgan/projects/ros2_foxy/install/rmw_fastrtps_shared_cpp/lib/librmw_fastrtps_shared_cpp.so
#5  0x00007fce91175d7f in rmw_fastrtps_shared_cpp::__rmw_wait(rmw_subscriptions_t*, rmw_guard_conditions_t*, rmw_services_t*, rmw_clients_t*, rmw_events_t*, rmw_wait_set_t*, rmw_time_t const*) () from /home/cmorgan/projects/ros2_foxy/install/rmw_fastrtps_shared_cpp/lib/librmw_fastrtps_shared_cpp.so
#6  0x00007fce9124ce0c in rmw_wait () from /home/cmorgan/projects/ros2_foxy/install/rmw_fastrtps_cpp/lib/librmw_fastrtps_cpp.so
#7  0x00007fce91783967 in rmw_wait () from /home/cmorgan/projects/ros2_foxy/install/rmw_implementation/lib/librmw_implementation.so
#8  0x00007fce91c23220 in rcl_wait () from /home/cmorgan/projects/ros2_foxy/install/rcl/lib/librcl.so
#9  0x00007fce920ae5aa in rclcpp::executor::Executor::wait_for_work(std::chrono::duration<long, std::ratio<1l, 1000000000l> >) ()
   from /home/cmorgan/projects/ros2_foxy/install/rclcpp/lib/librclcpp.so
#10 0x00007fce920af0e6 in rclcpp::executor::Executor::get_next_executable(rclcpp::executor::AnyExecutable&, std::chrono::duration<long, std::ratio<1l, 1000000000l> >) () from /home/cmorgan/projects/ros2_foxy/install/rclcpp/lib/librclcpp.so
#11 0x00007fce920b85b1 in rclcpp::executors::SingleThreadedExecutor::spin() () from /home/cmorgan/projects/ros2_foxy/install/rclcpp/lib/librclcpp.so
#12 0x00007fce920b2712 in rclcpp::spin(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>) ()
   from /home/cmorgan/projects/ros2_foxy/install/rclcpp/lib/librclcpp.so
#13 0x00007fce920b2844 in rclcpp::spin(std::shared_ptr<rclcpp::Node>) () from /home/cmorgan/projects/ros2_foxy/install/rclcpp/lib/librclcpp.so
#14 0x00005559a4974690 in main ()
(gdb) 
```

fujitatomoya, 2020-04-14:
@cmorganBE 

can you share your whole project for this package to reproduce issue?

cmorganBE, 2020-04-14:
Hi @fujitatomoya. I appreciate that. Pushed an example repo here, https://github.com/cmorganBE/ros2_sigint_hang that reproduces the issue with a script that makes it super easy to occur. I'm not sure this is entirely what you are looking for but please let me know if there is anything else necessary and I'll work to provide it.

Barry-Xu-2018, 2020-04-15:
@cmorganBE 

After testing your test example in docker with ubuntu:focal and latest ros2.repos, I can't reproduce the error you described.

My test log as follows,
```
$ i=1;while true; do echo ${i};./cause_hang.sh;i=$((i+1)); done
1
Starting test run
Sleeping for 10 seconds for the process to start
[WARN] [1586937879.338070279] [mcu]: This rmw implementation does not support ON_OFFERED_INCOMPATIBLE_QOS events, you will not be notified when Publishers offer an incompatible QoS profile to Subscriptions on the same topic.
[WARN] [1586937879.338385727] [mcu]: This rmw implementation does not support ON_REQUESTED_INCOMPATIBLE_QOS events, you will not be notified when Subscriptions request an incompatible QoS profile from Publishers on the same topic.
[WARN] [1586937879.338725135] [mcu]: This rmw implementation does not support ON_OFFERED_INCOMPATIBLE_QOS events, you will not be notified when Publishers offer an incompatible QoS profile to Subscriptions on the same topic.
[WARN] [1586937879.338938723] [mcu]: This rmw implementation does not support ON_OFFERED_INCOMPATIBLE_QOS events, you will not be notified when Publishers offer an incompatible QoS profile to Subscriptions on the same topic.
sigint-hang-cpp starting
sending -INT to all sigint-hang-cpp
[INFO] [1586937889.118124330] [rclcpp]: signal_handler(signal_value=2)
Sleeping for 2 seconds for shutdown
sigint-hang-cpp shutting down
Stopped
test_run passed
<omitted data>
180
Starting test run
Sleeping for 10 seconds for the process to start
[WARN] [1586937879.338070279] [mcu]: This rmw implementation does not support ON_OFFERED_INCOMPATIBLE_QOS events, you will not be notified when Publishers offer an incompatible QoS profile to Subscriptions on the same topic.
[WARN] [1586937879.338385727] [mcu]: This rmw implementation does not support ON_REQUESTED_INCOMPATIBLE_QOS events, you will not be notified when Subscriptions request an incompatible QoS profile from Publishers on the same topic.
[WARN] [1586937879.338725135] [mcu]: This rmw implementation does not support ON_OFFERED_INCOMPATIBLE_QOS events, you will not be notified when Publishers offer an incompatible QoS profile to Subscriptions on the same topic.
[WARN] [1586937879.338938723] [mcu]: This rmw implementation does not support ON_OFFERED_INCOMPATIBLE_QOS events, you will not be notified when Publishers offer an incompatible QoS profile to Subscriptions on the same topic.
sigint-hang-cpp starting
sending -INT to all sigint-hang-cpp
[INFO] [1586937889.118124330] [rclcpp]: signal_handler(signal_value=2)
Sleeping for 2 seconds for shutdown
sigint-hang-cpp shutting down
Stopped
test_run passed
<omitted data>
```

> Version or commit hash:
> my_ros2.repos.txt

Many repositories use master branch. So I cannot make sure I use the same version as your environmetn.

Could you execute below command in your environment to get exact version for each repository.

```
$ vcs export src --exact > my_ros2.repos
```

Beside,  how frequently does the error happen in your environment? (1/100 or 1/1000)

cmorganBE, 2020-04-15:
@Barry-Xu-2018 it was occurring perhaps 20-50% of the time for me.

I updated repositories and rebuilt foxy and now it doesn't occur for me... 

Foolishly on my part I didn't take the exact capture of repos before performing the update. I do have the terminal output for the update and looked at a number of the repos that were updated but didn't spot anything in the ones I thought would be the reasons for the fix, like rclcpp.

Thoughts? I guess I should close this one out for now and keep the test repo around in case it repeats itself. It would be more satisfying to know why its working correctly now around 10+ test runs but unless you or someone else spots this and has an idea the next best thing would be to roll back the repo versions from the log and retest which sounds like a pain.

fujitatomoya, 2020-04-16:
with https://github.com/ros2/ros2/commit/69bee87fb4ada4003c31f68bfd4c6c9018cde10f, I cannot reproduce the issue either.

@cmorganBE 

> I guess I should close this one out for now

i guess so, either me or @Barry-Xu-2018, we are not maintainer here, so it is up to you.

Barry-Xu-2018, 2020-04-16:
@cmorganBE 

> it was occurring perhaps 20-50% of the time for me.
>
> I updated repositories and rebuilt foxy and now it doesn't occur for me...

Got it.  
As fujitatomoya' said, you can decide if this issue should be closed.  
If find the same issue again, we can continue to investigate it :)

cmorganBE, 2020-04-16:
@Barry-Xu-2018, @fujitatomoya agreed. Thank you for responding and I'll keep this test case around for later. Take care!

closed 2020-04-16 by cmorganBE