ros2/rclcpp #1071: Dashing patch 6 / Ubuntu 18.04 Latching issue

muawijhe, 2020-04-17:
## Bug report

**Required Info:**

- Operating System: Ubuntu 18.04
- Installation type: Binary Installation (standard repo)
- DDS implementation: (default)
- Client library (if applicable): rclcpp

Let us consider the following piece of code:
`
// Create a QoS with 1 element of history
rclcpp::QoS oLatching = rclcpp::QoS(rclcpp::KeepLast(1));

// Enable latching accordingly to guide
oLatching = oLatching.transient_local();

// Create publisher
create_publisher<std_msgs::msg::Bool>("/test/bool", oLatching);

// publish the data...
// ....
// ...
`
accordingly to the guide, this should be sufficient to latch the
data of the topic /test/bool. However, when 

`ros2 topic echo /test/bool`

is used nothing is shown. Could you please tell me what is going
on?

Could you provide a valuable C++ example that actually works
both with regular topic echo and with rqt_gui for example?

Thanks a lot for your support and the good work you are doing
with ROS2 releases.

V.

closed 2020-04-30 by hidmic

muawijhe, 2020-04-30:
do you know why ubuntu release does not support your command?

usage: ros2 [-h] Call `ros2 <command> -h` for more detailed usage. ...
ros2: error: unrecognized arguments: --qos-durability

johnchars, 2023-04-04:
Dashing version seems not support `--qos-durability` option