ros2/rclcpp #1098: Implement a generic way to change per-node logging levels

clalancette, 2020-04-29:
## Feature request

#### Feature description

Most of the times when a developer wants to do debugging, one of the first things they do is to turn on debug logging.  One way to do that in ros2 is to run a command and increase the logging level, something like:

```
ros2 run demo_nodes_cpp talker --ros-args --log-level DEBUG
```

The problem with this is that it enables debugging on *all* components, not just the node.  While that *may* be what the user wants, in general I think they typically just want to debug their own node to start with.  We should provide a generic way to change the log level of a particular node, not of everything.

#### Implementation considerations
There's some possibility that this already exists in the ROS 2 ecosystem.  If it does, I don't think it is documented anywhere.  So either we need to implement this and document it, or something needs to show me how to currently do it so I can document it.

fujitatomoya, 2020-05-19:
@clalancette 

> There's some possibility that this already exists in the ROS 2 ecosystem.

I do not think this is supported on current implementation. rcl logging init/fini is exclusive in process space, so log setting is global scope.

iuhilnehc-ynos, 2020-05-28:
@clalancette @hidmic @fujitatomoya 

I'd like to implement this enhancement. The "--log-level" will be supported as following,

    a. use multiple items "--log-level", such as  
        
        --log-level debug --log-level talker1=error --log-level talker2=warning

    b. use one item, such as
        
        --log-level debug,talker1=error,talker2=warning

Could you help me review this [PR](https://github.com/ros2/rcl/pull/664)

clalancette, 2020-07-15:
This was actually solved with https://github.com/ros2/rcl/pull/664, so closing.

closed 2020-07-15 by clalancette