ros2/rclcpp #1118: User cannot setup qos in rclcpp_action create_client

jianming1481, 2020-05-18:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Binaries

- Version or commit hash:
  - N/A

- DDS implementation:
  - FastRTPS
- Client library (if applicable):
  - rclcpp_action
----
## Feature request

#### Feature description

In `rclpy`, user can set qos both on server and client. In `rclcpp_action` `create_server()`, user also can specify which qos they want. But in `rclcpp_action` `create_client()` function, there's no this options.  So I was confused if there's some mechanism in rclcpp_action that automatically set server and client in same qos. If so, please ignore this report. Thanks.
#### Implementation considerations

fujitatomoya, 2020-05-18:
@jianming1481 

> if there's some mechanism in rclcpp_action that automatically set server and client in same qos.

No i do not think so,
as you mentioned, rclcpp_action client will use default qos profile and rcl takes over this option to create endpoints. I think that this should be consistent.

hidmic, 2020-05-21:
@jianming1481 I don't see any fundamental limitation -- it seems `create_client()` simply doesn't an `rcl_action_client_options_t` argument and `Client` constructors use the default one. Would you be willing to contribute a patch? This will affect API, so bear in mind that it won't be merged immediately as we're in the midst of releasing Foxy.

fujitatomoya, 2020-05-22:
@hidmic @jianming1481 

could you check https://github.com/ros2/rclcpp/pull/1126?

closed 2020-06-16 by jacobperron