ros2/rclcpp #1121: lock-order-inversion (potential deadlock) in rclcpp::Client::wait_for_service, detected by ThreadSanitizer

yuriy-universal-ivanov, 2020-05-20:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Binaries, with apt-get
- Version or commit hash:
  - eloquent, from `deb http://packages.ros.org/ros2/ubuntu bionic main`
- DDS implementation:
  - Default (Fast-RTPS)
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
Reproduced when using `rclcpp::executors::MultiThreadedExecutor::spin()` in one thread and a `rclcpp::Client` waiting for a service in our own worker thread.

#### Expected behavior
Mutexes are always locked in the same order in rclcpp to avoid deadlocks.

#### Actual behavior
Some mutexes are locked in different order:
1) When `rclcpp::graph_listener::GraphListener::run_loop()` -> `rclcpp::node_interfaces::NodeGraph::notify_graph_change()`
2) When `rclcpp::node_interfaces::NodeGraph::get_graph_event()` -> undetected by TSan call in rclcpp: `<null> <null> (librclcpp.so+0x9fb1c)`

#### Additional information
In the worker thread:
```C++
      constexpr std::chrono::milliseconds single_spin_limit{100};

      while (!stop && rclcpp::ok() && !rclcpp_client->wait_for_service(single_spin_limit))
      {
      }
```
where `rclcpp_client` is `std::shared_ptr<rclcpp::Client<our service type>>` and `stop` is an `std::atomic<bool>`.

ThreadSanitizer detects a lock order inversion, which is a real threat of a deadlock:
```
WARNING: ThreadSanitizer: lock-order-inversion (potential deadlock) (pid=9849)
  Cycle in lock order graph: M470198622835870888 (0x000000000000) => M471324660181503496 (0x000000000000) => M470198622835870888

  Mutex M471324660181503496 acquired here while holding mutex M470198622835870888 in thread T159:
    #0 pthread_mutex_lock <null> (libtsan.so.0+0x3fadb)
    #1 <null> <null> (librclcpp.so+0x9fb1c)
    #2 our_project::FutureResponse<our_project::BinarySerializable<void> >::Impl::request(std::shared_ptr<rclcpp::Client<component_wrapper_ros2_interfaces::srv::BinaryBuffers> >, std::vector<unsigned char, std::allocator<unsigned char> >) /our_project/components/component-wrapper/include/./component-wrapper/future_response.h:124 (OurProjectComponentWrapper_test+0x5d82e)
    #3 our_project::FutureResponse<our_project::BinarySerializable<void> >::Impl::Impl(std::shared_ptr<rclcpp::Client<component_wrapper_ros2_interfaces::srv::BinaryBuffers> >, std::vector<unsigned char, std::allocator<unsigned char> >)::{lambda()#1}::operator()() const /our_project/components/component-wrapper/include/./component-wrapper/future_response.h:106 (OurProjectComponentWrapper_test+0x5b31c)
<...>

  Mutex M470198622835870888 previously acquired by the same thread here:
    #0 pthread_mutex_lock <null> (libtsan.so.0+0x3fadb)
    #1 rclcpp::node_interfaces::NodeGraph::get_graph_event() <null> (librclcpp.so+0xc1d38)
    #2 our_project::FutureResponse<our_project::BinarySerializable<void> >::Impl::request(std::shared_ptr<rclcpp::Client<component_wrapper_ros2_interfaces::srv::BinaryBuffers> >, std::vector<unsigned char, std::allocator<unsigned char> >) /our_project/components/component-wrapper/include/./component-wrapper/future_response.h:124 (OurProjectComponentWrapper_test+0x5d82e)
<...>

  Mutex M470198622835870888 acquired here while holding mutex M471324660181503496 in thread T161:
    #0 pthread_mutex_lock <null> (libtsan.so.0+0x3fadb)
    #1 rclcpp::node_interfaces::NodeGraph::notify_graph_change() <null> (librclcpp.so+0xc067a)

  Mutex M471324660181503496 previously acquired by the same thread here:
    #0 pthread_mutex_lock <null> (libtsan.so.0+0x3fadb)
    #1 rclcpp::graph_listener::GraphListener::run_loop() <null> (librclcpp.so+0x9ffa3)
```
future_response.h:124 contains `!rclcpp_client->wait_for_service(single_spin_limit)`

ivanpauno, 2020-05-20:
@yuriy-universal-ivanov can you share a complete example with reproduction steps (including how to get the tsan issue)?

From the available information, it's not obvious what's the problem.

yuriy-universal-ivanov, 2020-05-21:
Sure.

[rclcpp_1121_repro.zip](https://github.com/ros2/rclcpp/files/4663410/rclcpp_1121_repro.zip)

The archive contains two files: `rclcpp_1121_repro.cpp` and `CMakeLists.txt` to build it with CMake (it enables TSan). Target platform to test at - Linux (I tested at Ubuntu 18.04 with ROS2 eloquent). ROS2 should be sourced prior to invocation of CMake build so it succeeds. TSan from GCC 7+ or Clang 6+ is required to work correctly in modern Linux kernels.

When built (Debug configuration is recommended for human readable symbols in TSan report), it produces binary executable `rclcpp_1121_repro`. Start it. On finishing, TSan reports from 1 to 3 warnings. Among them there is always the one this bug is about (lock-order-inversion), the other 2 depends on whether you're lucky or not: they are race conditions on destroying rclcpp service and client without stopping multithreaded executors, and it's a separate problem, which may have been already fixed. I checked the history of changes of appropriate code in ROS2, so I'm waiting for Foxy to be released to check it - if it's not fixed there still, I'll also report another bug. This one is only about the lock-order-inversion.

ivanpauno, 2020-05-22:
> I checked the history of changes of appropriate code in ROS2, so I'm waiting for Foxy to be released to check it - if it's not fixed there still, I'll also report another bug

You should already be able to test with Foxy, see https://discourse.ros.org/t/ros-2-foxy-fitzroy-call-for-testing-and-package-releases/13998.

yuriy-universal-ivanov, 2020-05-22:
Thanks @ivanpauno, it's unfortunately only available in Ubuntu 20 repos, but I'll check it soon.

Anyway, the deadlock thing doesn't seem to be affected by those changes, so I believe it has to be investigated (you now have repro code to test).
Please let me know if there is anything else I could help regarding the reported issue.

fujitatomoya, 2020-05-25:
@ivanpauno 

using https://github.com/ros2/ros2/commit/4eece71bc42f9b28e1c21633d270d908bb508ca7, source build with Ubuntu20.04, I can reproduce the same warning message from TSan. reproduction procedure is following,

```
> cd colcon_ws
> cd src
> git clone git@github.com:fujitatomoya/ros2_test_prover.git
> cd ..
> colcon build --symlink-install --packages-select ros2_test_prover
> source install/local_setup.bash
> for i in {1..100}; do ros2 run ros2_test_prover rclcpp_1121_repro; sleep 3; done
```

entire log is attached, [rclcpp-issue-1121.log](https://github.com/ros2/rclcpp/files/4677468/rclcpp-issue-1121.log)

I did try a few hundreds times, but cannot actually make deadlock happen. it is warning that indicates potential lock inversion. (also data race can be reproducible as reported.)

fujitatomoya, 2020-05-27:
@ivanpauno 

Thread Sanitizer Analysis (potential lock inversion)

- Thread-18: (created by user request thread)

```
  Mutex M292025100016882824 acquired here while holding mutex M290617587694362920 in thread T18:
    #0 pthread_mutex_lock <null> (libtsan.so.0+0x5271c)
    #1 __gthread_mutex_lock /usr/include/x86_64-linux-gnu/c++/9/bits/gthr-default.h:749 (rclcpp_1121_repro+0x35fef)
    #2 std::mutex::lock() /usr/include/c++/9/bits/std_mutex.h:100 (rclcpp_1121_repro+0x364b4)
    #3 rclcpp::graph_listener::acquire_nodes_lock_(std::mutex*, std::mutex*, rcl_guard_condition_t*) <null> (librclcpp.so+0x560149)
    #4 operator() /root/docker_ws/ros2_colcon/src/fujitatomoya/ros2_test_prover/src/rclcpp_1121_repro.cpp:60 (rclcpp_1121_repro+0x20c03)
```

M292025100016882824 is GraphListener::node_graph_interfaces_mutex_, M290617587694362920 is NodeGraph::graph_mutex_.

this thread is created on https://github.com/fujitatomoya/ros2_test_prover/blob/master/src/rclcpp_1121_repro.cpp#L60,

this is 1st call from this node, so call in GraphListener::add_node, https://github.com/ros2/rclcpp/blob/87bb9f9758dce239d37256ae91d58ce6325f405b/rclcpp/src/rclcpp/graph_listener.cpp#L298

https://github.com/ros2/rclcpp/blob/87bb9f9758dce239d37256ae91d58ce6325f405b/rclcpp/src/rclcpp/graph_listener.cpp#L251 but this actually will be unlocked since using std::adopt_lock on https://github.com/ros2/rclcpp/blob/87bb9f9758dce239d37256ae91d58ce6325f405b/rclcpp/src/rclcpp/graph_listener.cpp#L302-L303

- Thread-21: (created by Thread-18)

```
  Mutex M290617587694362920 acquired here while holding mutex M292025100016882824 in thread T21:
    #0 pthread_mutex_lock <null> (libtsan.so.0+0x5271c)
    #1 __gthread_mutex_lock /usr/include/x86_64-linux-gnu/c++/9/bits/gthr-default.h:749 (rclcpp_1121_repro+0x35fef)
    #2 std::mutex::lock() /usr/include/c++/9/bits/std_mutex.h:100 (rclcpp_1121_repro+0x364b4)
    #3 std::lock_guard<std::mutex>::lock_guard(std::mutex&) /usr/include/c++/9/bits/std_mutex.h:159 (rclcpp_1121_repro+0x39082)
    #4 rclcpp::node_interfaces::NodeGraph::notify_graph_change() <null> (librclcpp.so+0x59ac5c)
```

this thread is create on  https://github.com/ros2/rclcpp/blob/87bb9f9758dce239d37256ae91d58ce6325f405b/rclcpp/src/rclcpp/graph_listener.cpp#L104

M292025100016882824 is GraphListener::node_graph_interfaces_mutex_, M290617587694362920 is NodeGraph::graph_mutex_.

https://github.com/ros2/rclcpp/blob/87bb9f9758dce239d37256ae91d58ce6325f405b/rclcpp/src/rclcpp/graph_listener.cpp#L146, then https://github.com/ros2/rclcpp/blob/87bb9f9758dce239d37256ae91d58ce6325f405b/rclcpp/src/rclcpp/node_interfaces/node_graph.cpp#L326

it seems,

- Thread-18: graph_mutex_ -> node_graph_interfaces_mutex_
- Thread-21: node_graph_interfaces_mutex_ -> graph_mutex_

```
WARNING: ThreadSanitizer: lock-order-inversion (potential deadlock) (pid=6962)
Cycle in lock order graph: M290617587694362920 (0x000000000000) => M292025100016882824 (0x000000000000) => M290617587694362920
                           NodeGraph::graph_mutex_ => GraphListener::node_graph_interfaces_mutex_ => NodeGraph::graph_mutex_
```

fujitatomoya, 2020-05-27:
There is another problem detected by Thread Sanitizer about `WARNING: ThreadSanitizer: data race (pid=12656)`. Callstack is related to eprosima::fastdds::rtps, data race condition between udp socket recv and close. so I create issue https://github.com/ros2/rmw_fastrtps/issues/391.

ivanpauno, 2020-05-27:
@fujitatomoya can you contribute with backports to Dashing/Eloquent?

Thanks for #1135 fix!!!

fujitatomoya, 2020-05-28:
@ivanpauno 

> can you contribute with backports to Dashing/Eloquent?

working on that in today (Japan Local Time)

will get back to you with PR.

fujitatomoya, 2020-05-28:
@ivanpauno 

could you help to merge the followings?

- https://github.com/ros2/rclcpp/pull/1138
- https://github.com/ros2/rclcpp/pull/1137

ivanpauno, 2020-06-01:
@fujitatomoya Thanks for the fix and backports!!

@yuriy-universal-ivanov A backport to Eloquent has been merged, the fix will be available on debians in next patch release.

closed 2020-06-01 by ivanpauno

yuriy-universal-ivanov, 2020-06-02:
@fujitatomoya @ivanpauno 
Awesome, thank you both a lot!